MarlinFirmware/Marlin #4544: NEED HELP....

boelle, 2016-08-06:
I have the SAV MK1 and would like to be able to use marlin

Sadly the hardware files i have is for IDE 1.0.6 and so i'm out of the game

The maker have dropped marlin support and now supports repetier as they still support 1.0.6

The way i used to do things was install teensyduino and then overwrite their board files with one from the maker of my controller.

In 1.6.8 that way does not work anymore and i cant figure how to do it. I have tried to use printrboards hardware files but that did not work either.

Are there anyone with the knowledge that can figure it? If not i will be forced to use repetier also

Roxy-3D, 2016-08-06:
I can't help with your question.   But as you know I have a PrintRboard.   I've been fighting the same problem as you and I can't update Marlin on that printer.   I gave up.   I bought a RAMPS v1.4 on eBay for  $4.01 and a Mega-2560 clone board for $5.88   I spent another $5.00 for the Step-Sticks.    

It seems to me PrintRbot should keep the files up to date so their hardware can be used by Arduino, but they don't seem to care.   For $15 you can make the problem go away.

boelle, 2016-08-06:
I just wonder how repetier is able to keep their things in order and still be able to run at 1.0.6 and marlin cant.... 

and 15$ does not quite cut it... the cheapest i can see is about 60£

epatel, 2016-08-06:
@boelle $18? http://www.ebay.com/itm/Ramps-kit-Mega-2560-R3-RAMPS-1-4-Controller-4pcs-A4988-Stepper-Driver-Module-/282035420142

boelle, 2016-08-06:
Item location:
深圳, 广东省, China

add 35% impor tax and processing fee

boelle, 2016-08-06:
so 120 DKK

with import tax and fee 282

add to that that the quality is doubtfull and if so the trouble with getting your money back

boelle, 2016-08-06:
oh well.... i think its cheaper and more safe in the long run to skip marlin for good... i hoped that a fix came up

it would be a shame to trash perfectly good electronics just to run some different firmware

berryelectronics, 2016-08-06:
@boelle where do you come from? atleast in germany there arent any import taxes for item < 23€

epatel, 2016-08-06:
@berryelectronics @boelle Yup, I am in Sweden and I order a lot from eBay/China and I can't remember anything adding tax (from USA, that happens mostly). It takes a while, often about 30 days.

thinkyhead, 2016-08-07:
Re-opening, since we include `pins_SAV_MKI.h` and should work out the method to build for it and add notes to the pins file on the subject.

boelle, 2016-08-07:
well all i need is a set of hardware files for it

as with ide 1.0.6 on 1.6.8 i start by installing teensyduino but then i'm stuck as the structure of hardware files are now very different...

would a link to the old hardware files do any good?

boelle, 2016-08-07:
@epatel i'm from denmark

but here they sort the post rather "smart" and by machine... everything from outside denmark lands at copenhagen and gets splitted in 2 piles... from outside EU and within EU... the one within EU goes to sorting center allmost at once. The other well takes forever to pass as they need to check customs value by hand... anything above about 10$ are takes aside.

so its very rare that we dont get tax added

boelle, 2016-08-07:
hmm.... if we figure this i guess it can be used for the printrboard also?

but would it be outside the scope of marlin to host hardware files? ie in the form of a zipfile and board manager link so IDE can install it automatic?

epatel, 2016-08-07:
@boelle So, order the items separately around $5 each like @Roxy-3D specified. Shipping is most often free from China (have no idea how they pull that off but I like it). And searching on eBay can be pretty fun I think :) you can find amazing other items.

Roxy-3D, 2016-08-07:
> Shipping is most often free from China (have no idea how they pull that off but I like it). 
I think this is an example of government subsidized industry.  By keeping the postage rates super low within the country (even things shipped with tracking numbers) they help their export industry.   The reason is most countries have reciprocity agreements.  They agree, "we will continue the shipping and delivery process for anything you deliver to our boarder if you do the same." 

The big problem comes in when you get a defective item.  You have to return it to the eBay vendor and use a tracking number to do it.   That costs many times what the item is worth.  So, they can ship junk not worry about it.

berryelectronics, 2016-08-07:
@boelle if they only pass things from within the EU, you could get that one from GB: http://www.ebay.de/itm/3D-Printer-Kit-Mega-2560-Board-RAMPS-1-4-4X-A4988-For-Arduino-RepRap-TE163-/301923620361?hash=item464c0cd609:g:T7YAAOSw9VZXOVij

they are still in the eu, so get it

berryelectronics, 2016-08-07:
@Roxy-3D my experience with chinese vendors (ebay and aliexpress):
As long as you pay with paypal/alipay, you are on the safe side because you can get the money back from them without having to ship it back.
and like 99% of the chinese vendors dont want you to send it back (if it isnt like 100€) and just give you the money back. Even better with aliexpress: if its DOA, you can just open a case where you get the money back before the vendors see it, because aliexpress keeps the money until you release it when the item arrived.

boelle, 2016-08-07:
@epatel & @Roxy-3D you are right... postage is free because its paid by the Chinese goverment

@berryelectronics why trash otherwise good electronics just because i miss a text file or 2? (the hardware files for the IDE)

thinkyhead, 2016-08-08:
> would a link to the old hardware files do any good?

Links to those files would start us on the path. Since I know practically nothing about hardware files, I would take the following naïve approach:
1. Get the hardware files for some project that has both old and new versions
2. Try to emulate the same changes to the old SAV-MKI hardware files.

Once these files are updated, we can just host them someplace on Github and link to them. We don't necessarily need to include them in the Marlin package.

boelle, 2016-08-10:
http://www.reprap.org/wiki/File:At90usb1286txt_SAV_MkI.zip

http://reprap.org/wiki/SAV_MKI

on last link search for "Step 3: Setup the SAV MKI on Arduino's IDE"

but there you see how i used to install under IDE 1.06

boelle, 2016-08-10:
i would have done as you said.... but since he just overwrites file with others its hard to do for me at least

at least its a bit over my head... and of course the people behind teensy cant be expected to help out... i even had the nerve to ask them :-/

boelle, 2016-08-14:
hmm... had an idea for this one... since its more or less 99% a teensy i planned to burn the bootloader for that teensy and just rename the board name in the hardware files and call it a day...

it's a risk thou as i could end up with a 50*50*50 cm brick most would call a 3d printer :-D

boelle, 2016-08-14:
nah bootloader for teensy is closed source :-/

but looking at this one again https://github.com/scwimbush/Printrboard-HID-Arduino-IDE-Support/issues/1

boelle, 2016-08-24:
killing this one and will try and focus on making the edits to make it compile in ide 1.0.6 instead...

almost there but last part is a bit over my head

closed 2016-08-24 by boelle