MarlinFirmware/Marlin #4549: new Advance Extrusion algorithm (LIN_ADVANCE) extruder motor not running

Macgyver001, 2016-08-06:
The Marlin version RC7 - 31 Jul 2016
    Machine Self build coreXY
    Electronics board ramps 1.4 with arduino 2560
    Machine components (direct drive e3d lite, with termistors (my stepper drivers are m542T and I run  at 64 micro step also the extruder)
    Host software (none
    Slicer (if relevant)-
    Printing method (Pronterface or Octoprint)

[Configuration_and_adv.zip](https://github.com/MarlinFirmware/Marlin/files/405163/Configuration_and_adv.zip)

Problem:
When I enable Lin_advance my extruder motor does not turn/extrude.
What Have I tried to do.

Turn off core XY, but still no extruding
K factor put to 0 no effect
Tried to lower the extrusion speed no effect
Extruding in reverse is also not working
When disable watchdog does not help

When I disable Lin_advance it extrudes well.

thinkyhead, 2016-08-07:
Sounds like a possible bug! Between myself and @Sebastianv650 we should be able to figure it out…

Sebastianv650, 2016-08-07:
If the problem is also there if K=0 then it should be something with the extruder ISR. This is the only big change when LIN_ADVANCE is enabled.
I had a short look on your config, the values look quite strange to me:

#define DEFAULT_AXIS_STEPS_PER_UNIT   {320,320,800,386}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {500, 500, 2, 50}   // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {60,60,200,500}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot. was 75

#define DEFAULT_ACCELERATION          60    // X, Y, Z and E acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  300    // E acceleration in mm/s^2 for retracts
#define DEFAULT_TRAVEL_ACCELERATION   80    // X, Y, Z acceleration in mm/s^2 for travel (non printing) moves

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK                500.0    // (mm/sec) was 20 was 700
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
#define DEFAULT_EJERK                50.0

Why do you use so incredible high jerk values combined with ultra-low acceleration values? I can't imagine this can work..! Do you ever printed good things with theese values?

Macgyver001, 2016-08-08:
Hi thanks for the quick response.

The jerk settings are not aligned very carefully so thank you for your comment.
I have lowered a lot of values one by one, but without any result, no extruding when Lin_advance is enable, final settings are listed below:

#define DEFAULT_AXIS_STEPS_PER_UNIT   {320,320,800,100}  // default steps per unit for Ultimaker 160*2,160*2,800,((200*64)*1.0/(10.56*3.14159))=386
#define DEFAULT_MAX_FEEDRATE          {200, 200, 2, 10}   // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {60,60,100,10}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot. was 75 e=500

#define DEFAULT_ACCELERATION          10    // X, Y, Z and E acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  10    // E acceleration in mm/s^2 for retracts
#define DEFAULT_TRAVEL_ACCELERATION   80    // X, Y, Z acceleration in mm/s^2 for travel (non printing) moves

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK               5.0    // (mm/sec) was 20 was 500
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
#define DEFAULT_EJERK                 2.0    // (mm/sec) 5.0 was 2000

Is there something changed in the extruder stepper motor signals when using Lin_advance? I have connected my ramps 1.4 via self made wiring towards my stepper drivers  m542T (but as I mentioned I do not change connections Lin_advance disabled or enabled and it works when I disable Lin_advance)

thinkyhead, 2016-08-09:
I recommend you keep testing daily with `RCBugFix` as we keep fixing things by mistake. This will also allow us to insert some debugging code to help track the issue – something we would generally only do with the working branch.

Sebastianv650, 2016-08-09:
No, LIN_ADVANCE isn't changing anything in the way the stepper is electricaly driven. The only difference is the timing, when enabled the extruder has it's own ISR.

I would start with common configuration values and see what happens. For example, maybe a value can overflow with an acceleration of only 10mm/s². Therefore, I would set all acceleration values to 500 and test again.
If a simple extruder only move, for example G1 E3 F50, isn't working with enabled advance, I would check the wiring or other values in the basic config. There is no reason I can think of why such a command should fail due to LIN_ADVANCE.

thinkyhead, 2016-08-10:
@Sebastianv650 Well, you will want to look through the `LIN_ADVANCE` code and make sure nothing got broken since it was first integrated.

esenapaj, 2016-08-10:
Possibly this?
2/4 phase Nema 23 Stepper Motor Driver 24-50VDC 1.5A-4.5A 256 Microstep M542T|M542T|Stepper Motor Drivers
http://www.omc-stepperonline.com/24-phase-nema-23-stepper-motor-driver-2450vdc-15a45a-256-microstep-m542t-p-293.html
>For reliable response, pulse width should be longer than 1.5μs.

Extruder working fine with A4988s, not with DRV8825, repetier works, marlin not · Issue #975 · MarlinFirmware/Marlin
https://github.com/MarlinFirmware/Marlin/issues/975#issuecomment-47515225
>From Pololu's site:
>>With the DRV8825, the high and low STEP pulses must each be at least 1.9 us; they can be as short as 1 us when using the A4988.

esenapaj, 2016-08-10:
To see the [Repetier-Firmware](https://github.com/repetier/Repetier-Firmware/tree/master/src/ArduinoAVR/Repetier), Additional delay is added in here and there (`STEPPER_HIGH_DELAY`) for stepper driver that it needs long palse.

Macgyver001, 2016-08-10:
Hello all,

Today I checked the enable signal on my stepper driver, that seems to work fine and is not the issue
I also tried the acceleration = 500 and jerk 50 but that also did not help.
It could be related to pulse width? Is there an option to change that in the code. 
I will check if I can find anything related to pulse with in the data sheet of the m542 driver?

Sebastianv650, 2016-08-11:
With a jerk of 50, you may end up with such high speed changes that the extruder can keep up with the pressure adjustment moves. Extruder only moves shouldn't be affected by LIN_ADVANCE, but I will have a look at the latest Marlin code after my holidays in about 2 weeks.

Macgyver001, 2016-08-11:
Ok, but last time I tested with jerk of 2!
I checked the timing of the step pulse and its indeed 1.5 us up and down and 5 us after a direction switch, I don't know what the current timing is, and if its shorter than that?

Thanks for checking, I will also go on holiday, so lets catch up after..

[image]

thinkyhead, 2016-08-12:
@esenapaj Do the Marlin settings `CONFIG_STEPPERS_TOSHIBA` and `MAX_STEP_FREQUENCY` seem to be equivalent to the repetier delay options?

esenapaj, 2016-08-12:
No, it isn't.
It looks like that `CONFIG_STEPPERS_TOSHIBA` of Marlin [suppress the `MAX_STEP_FREQUENCY` simply](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Conditionals_post.h#L175-L179), and [disable double / quad stepping mode automatically](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/stepper.h#L281-L292).

In case of [`STEPPER_HIGH_DELAY`](https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/Configuration.h#L1099-L1104) and [`DOUBLE_STEP_DELAY`](https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/Configuration.h#L1127) of Repetier-Firmware,
those add additional delay to section of stepper.
For example, they add additional delay to [A](https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/Printer.cpp#L1782), [B](https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/motion.cpp#L2397), [C](https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/motion.cpp#L2760).
Maybe those lines of Marlin are [A](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/stepper.cpp#L1076), [B](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/stepper.cpp#L369), [C](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/stepper.cpp#L665). 

And, Marlin decide threshold value between single and double stepping [automatically](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/stepper.h#L281-L292).
In case of Repetier-Firmware, threshold value between single and double stepping is [configurable](https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/motion.cpp#L2541-L2563) ([`STEP_DOUBLER_FREQUENCY`](https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/Configuration.h#L1118)).

boelle, 2016-08-14:
@thinkyhead does this make a bell ring ? :-D

thinkyhead, 2016-08-19:
I'm too shy to start messing with the stepper code at this junction.

Macgyver001, 2016-09-01:
Hi all,

I have tried to reduce MAX_STEP_FREQUENCY to 10000, but without any result. 
Is there a dirty way to add a delay in only the Extruder stepper, for testing purposes? If this is the correct direction.
In the older version of marlin from April 2016 I enabled the 128 u stepping, this gives me a resolution of 640 steps/mm and Extruder 772 steps/mm, at 20mm/s the processor is still capable of running this setup.

Roxy-3D, 2016-09-01:
> Is there a dirty way to add a delay in only the Extruder stepper, for testing purposes? If this is the correct direction.

Can you explain in a little more detail what you are thinking and why this would be beneficial?   I need to understand the problem a little better.

But the reason I'm asking for more detail is this:  Right now I'm messing with the mesh_buffer_line() routine to make it non-recursive (to lighten the CPU load).   On any given Planner request, it would not be difficult to stagger  when an axis moves relative to the other axis.

Macgyver001, 2016-09-02:
Hi Roxy,

The problem is that when I enable Lin_advance my extruder stepper does not work.
In the documentation of my external stepper controller it is stated that the steps (digital signals) should not be shorter than 1.5 u seconds. Therefore I would like to enlarge the timing of a high / low signal of only the extruder (or all stepper signals if necessary) (nothing else), to investigate if this is the problem. When I disable LIn_advance everything is working correctly. The XYZ axis are working fine in all occasions.

Blue-Marlin, 2016-09-02:
#4738 is already merged into RCBugFIX.

Try if https://github.com/MarlinFirmware/Marlin/pull/4738/files#diff-20025b9ffe01efeb4272ee752f32170dR712 helps.

thinkyhead, 2016-09-02:
@Blue-Marlin Is the overhead of invoking interrupts very large, I wonder? I was thinking that the advance ISR could start the pulse(s) for E, and instead of sitting and waiting, it could instead set up an interrupt to stop the pulse(s) and exit right away. Perhaps the same approach is possible with the main stepper ISR. This would allow them both to exit sooner and give time back to the main thread.

Blue-Marlin, 2016-09-02:
I might have seen something like that in ether grbl or RepetierFirmware. On the other hand i had a look into Prusa-Firmware today, where they not even have the interleave
````cpp
        counter_x += current_block->steps_x;
        if (counter_x > 0) {
          WRITE(X_STEP_PIN, !INVERT_X_STEP_PIN);
          counter_x -= current_block->step_event_count;
          count_position[X_AXIS]+=count_direction[X_AXIS];   
          WRITE(X_STEP_PIN, INVERT_X_STEP_PIN);
        }
```
But with supporting only 2-3 defined boards they pretty well know what timing to expect. (And they kicked Advance at all.)

>  not be shorter than 1.5 u seconds.
That is ~24 ticks. With only one extruder in the loop this could be a bordercase, especially with the old `STEP_E_ONCE(INDEX)`. (Have not really counted cycles.)

@Macgyver001 asked for a delay - there it is. But in reality i do not believe there is one needed.
Let' see if it helps.

thinkyhead, 2016-09-02:
> had a look into Prusa-Firmware today, where they not even have the interleave

Hmm! But apparently they have employed fixed-point maths where performance matters. I've been thinking about that a lot also.

Blue-Marlin, 2016-09-02:
I still try to find out what they have really done. The forks are pretty far apart from each other.
For now i was mostly interested what they have done with the new bed and the inductive sensor. Really interesting.

Blue-Marlin, 2016-09-02:
Comments like
```
//FIXME This routine is called 15x every time a new line is added to the planner,
// therefore it is a bottle neck and it shall be rewritten into a Fixed Point arithmetics,
// if the CPU is found lacking computational power.
```
do look more than a plan than like a done.

thinkyhead, 2016-09-02:
Searching the interwebs turns up a lot of discussion about pulse widths, decay time, etc. Lately I'm extra-interested in the topic because I'm not sure I'm a fan of DRV8825 for the current SCARA application. The machine I'm writing code for is currently set up to use 100:1 harmonic drives on the A/B axes, which means if using 32x micro-stepping, we need 1778 pulses per degree to move the steppers (which results in ~7mm per degree in the most extreme case, but only half of that per linkage).

Or in other words, total overkill in terms of movement accuracy. Even at 16x microstepping it amounts to 889 pulses per degree. When we start having this many pulses the bugs and errors start to get a bit weird. Imagine also, this is all on a Mega2560 platform, so again, really pushing the limits unnecessarily. I'm inclined to drop down even as low as 8x microstepping, but wondering how much extra sound that will produce.

Macgyver001, 2016-09-06:
Hey guys,

@Macgyver001 asked for a delay - there it is. But in reality i do not believe there is one needed.
Let' see if it helps.

I am sorry, but I am quite new in C, so Blue Marlin I don't know what I have to change where?
I downloaded the latest RC bugfix but I don't know where to find STEP_E_ONCE(INDEX). Unfortunately I don't have a oscilloscope else I would have checked it.
Advance and Lin_advance are both not extruding when one off them is enabled. including the use of CONFIG_STEPPERS_TOSHIBA to limit the frequency.

Thinkyhead, I have changed my drivers for m542T because they run smoother and I had some resonance problems. My CoreXY is quite stiff (ridgid) and I have better looking prints with higher mirco-stepping. If you like I can show the difference between lets say 16 and 128 (mirco stepping), I need some time for that?  My latest Treefrog is really nice except some small artifacts I hope to resolve with Lin_advance, and the Lin_advance concept is way smarter than the normal implementation, that's why I like to enable Lin_advance.

thinkyhead, 2016-09-08:
@Macgyver001 Did you test with `MINIMUM_STEPPER_PULSE` set around 9 or 10?

Macgyver001, 2016-09-10:
Thanks thinkyhead,
I now tested this option Minimum stepper pulse.
Started with 10 it worked, but marlin did not response well after.
Went down to 1 and the extruder is reacting, I have to see if  my marlin configurables are correct and see if everything is working well. Something went NOK with my G28 homing, Z was not responding correctly.
This MINIMUM_STEPPER_PULSE accepts only integers?
For now this Minimum_stepper_Pulse is doing the Job for me, I can start testing with LIN_ADVANCE.

Nice.
I will Keep you updated.

thinkyhead, 2016-09-10:
> This `MINIMUM_STEPPER_PULSE` accepts only integers?

It does. If you want to reduce the time it may accrue, you can strip off individual 1/16ths of a µs by increasing the value of `CYCLES_EATEN_BY_CODE` and/or `CYCLES_EATEN_BY_E` which are "hidden" in `stepper.cpp`.

thinkyhead, 2016-09-10:
@Sebastianv650 It looks like we need to add a small delay into the advance ISR to get a long enough pulse. If a stepper pulse needs to be 9 or 10µs long, then it should do something like this…
```cpp
void Stepper::advance_isr() {

  old_OCR0A += eISR_Rate;
  OCR0A = old_OCR0A;

  #define SET_E_STEP_DIR(INDEX) \
    E## INDEX ##_DIR_WRITE(e_steps[INDEX] <= 0 ? INVERT_E## INDEX ##_DIR : !INVERT_E## INDEX ##_DIR)

  #define START_E_PULSE(INDEX) \
    if (e_steps[INDEX]) E## INDEX ##_STEP_WRITE(INVERT_E_STEP_PIN)

  #define STOP_E_PULSE(INDEX) \
    if (e_steps[INDEX]) { \
      e_steps[INDEX] <= 0 ? ++e_steps[INDEX] : --e_steps[INDEX]; \
      E## INDEX ##_STEP_WRITE(!INVERT_E_STEP_PIN); \
    }

  SET_E_STEP_DIR(0);
  #if E_STEPPERS > 1
    SET_E_STEP_DIR(1);
    #if E_STEPPERS > 2
      SET_E_STEP_DIR(2);
      #if E_STEPPERS > 3
        SET_E_STEP_DIR(3);
      #endif
    #endif
  #endif

  // Step all E steppers that have steps
  for (uint8_t i = 0; i < step_loops; i++) {

    START_E_PULSE(0);
    #if E_STEPPERS > 1
      START_E_PULSE(1);
      #if E_STEPPERS > 2
        START_E_PULSE(2);
        #if E_STEPPERS > 3
          START_E_PULSE(3);
        #endif
      #endif
    #endif

    // For a minimum pulse time wait before stopping pulses
    #if MINIMUM_STEPPER_PULSE > 0
      #define CYCLES_EATEN_BY_E_CODE 10
      #define ADDED_DELAY (MINIMUM_STEPPER_PULSE) * (F_CPU / 1000000UL) - 20 * (E_STEPPERS - 1) - CYCLES_EATEN_BY_E_CODE
      #if ADDED_DELAY > 0
        static uint32_t pulse_start;
        pulse_start = TCNT0;
        while ((uint32_t)(TCNT0 - pulse_start) < (uint32_t)(ADDED_DELAY)) { /* nada */ }
      #endif
    #endif

    STOP_E_PULSE(0);
    #if E_STEPPERS > 1
      STOP_E_PULSE(1);
      #if E_STEPPERS > 2
        STOP_E_PULSE(2);
        #if E_STEPPERS > 3
          STOP_E_PULSE(3);
        #endif
      #endif
    #endif
  }

}
```

esenapaj, 2016-09-18:
I obtained oscilloscope in this month and now I'm observing stepping pulses, I found direction signal of `LIN_ADVANCE` (and `ADVANCE`)is broken.
Yellow: direction signal
Blue: stepping pulse
magenta: enable signal
```cpp
#define XYZ_FULL_STEPS_PER_ROTATION 400 // original is 200
#define E_FULL_STEPS_PER_ROTATION 1024 // 5.18 : 1 geared // original is 200
#define XYZ_MICROSTEPS 8
#define E_MICROSTEPS 8
#define XYZ_BELT_PITCH 2
#define XYZ_PULLEY_TEETH 16
#define E_GEAR_DIAMETER 10.94 // Bondtech QR: Effective diameter : 10.94mm?
#define XYZ_STEPS (double(XYZ_FULL_STEPS_PER_ROTATION) * XYZ_MICROSTEPS / (XYZ_BELT_PITCH * XYZ_PULLEY_TEETH))
#define E_STEPS (double(E_FULL_STEPS_PER_ROTATION) * E_MICROSTEPS / (E_GEAR_DIAMETER * M_PI))
```
`LIN_ADVANCE`

`g1 e10 f10` | `g1 e100 f1200`
--- | ---
[image] | [image]
This problem had been also occurred in [my 32bit branch](https://github.com/esenapaj/Marlin/tree/Duo),
but when I change the `Stepper::old_OCR0A` and `Stepper::eISR_Rate` from `unsigned char` to `uint32_t` and down a threshold of double-stepping, problem was solved roughly.
But in case of official 8bit Marlin, even if I change the `old_OCR0A` and `eISR_Rate` to `unsigned int` or `unsigned long`, problem isn't solved.

and, I awaken stepper pulse of `ADVANCE` and `LIN_ADVANCE` are inverted for some reason.

`g1 e100 f1200`

normal | `ADVANCE`
--- | ---
[image] | [image]

<details>
<summary>Configuration.h</summary>
```cpp
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
[... 2265 lines cut ...]
 * M155 B77  ; M
 * M155 B97  ; a
 * M155 B114 ; r
 * M155 B108 ; l
 * M155 B105 ; i
 * M155 B110 ; n
 * M155 S1   ; Send the current buffer
 *
 * ; Example #2
 * ; Request 6 bytes from slave device with address 0x63 (99)
 * M156 A99 B5
 *
 * ; Example #3
 * ; Example serial output of a M156 request
 * echo:i2c-reply: from:99 bytes:5 data:hello
 */

// @section i2cbus

#define EXPERIMENTAL_I2CBUS
#define I2C_SLAVE_ADDRESS  0 // Set a value from 8 to 127 to act as a slave

#endif // CONFIGURATION_ADV_H
```
</details>

thinkyhead, 2016-09-18:
Also: Is the pulse duration in `advance_isr` too short? There is no added delay, and with only a single E stepper, this might cause some issue.

But your report is interesting, because the stepper direction is definitely being set there:
```cpp
SET_E_STEP_DIR(0);
#if E_STEPPERS > 1
  SET_E_STEP_DIR(1);
  #if E_STEPPERS > 2
    SET_E_STEP_DIR(2);
    #if E_STEPPERS > 3
      SET_E_STEP_DIR(3);
    #endif
  #endif
#endif
```

thinkyhead, 2016-09-18:
I can see a couple possible changes that may help. Try this…

In the `Stepper::set_directions()` method change:
```cpp
  #if DISABLED(ADVANCE)
    if (motor_direction(E_AXIS)) {
      REV_E_DIR();
      count_direction[E_AXIS] = -1;
    }
    else {
      NORM_E_DIR();
      count_direction[E_AXIS] = 1;
    }
  #endif //!ADVANCE
```
…so it always sets the direction for E, even with regular `ADVANCE` enabled…
```cpp
  if (motor_direction(E_AXIS)) {
    REV_E_DIR();
    count_direction[E_AXIS] = -1;
  }
  else {
    NORM_E_DIR();
    count_direction[E_AXIS] = 1;
  }
```
And then in `Stepper::advance_isr()` completely get rid of the code that sets the E direction by removing this:
```cpp
#define SET_E_STEP_DIR(INDEX) \
  E## INDEX ##_DIR_WRITE(e_steps[INDEX] <= 0 ? INVERT_E## INDEX ##_DIR : !INVERT_E## INDEX ##_DIR)

```
…and this…
```cpp
SET_E_STEP_DIR(0);
#if E_STEPPERS > 1
  SET_E_STEP_DIR(1);
  #if E_STEPPERS > 2
    SET_E_STEP_DIR(2);
    #if E_STEPPERS > 3
      SET_E_STEP_DIR(3);
    #endif
  #endif
#endif
```

thinkyhead, 2016-09-18:
One other thing, maybe…
```diff
  #if ENABLED(LIN_ADVANCE)
-   volatile int Stepper::e_steps[E_STEPPERS];
+   volatile long Stepper::e_steps[E_STEPPERS];
```

esenapaj, 2016-09-19:
About `ADVANCE`:
>```diff
-  #if DISABLED(ADVANCE)
    if (motor_direction(E_AXIS)) {
      REV_E_DIR();
      count_direction[E_AXIS] = -1;
    }
    else {
      NORM_E_DIR();
      count_direction[E_AXIS] = 1;
    }
-  #endif //!ADVANCE
```

It seems that this is a precise fixing. Direction signal works rightly. (but stepper signal is still inverted)

`g1 e100 f1200` | `g1 e100 f3000`
--- | ---
![newfile5](https://cloud.githubusercontent.com/assets/15308844/18620428/352d5972-7e4d-11e6-852c-2d31a7dbd67e.png) | ![newfile6](https://cloud.githubusercontent.com/assets/15308844/18620430/3e387d30-7e4d-11e6-90ef-0d4363b0be37.png)

About `LIN_ADVANCE`:
>```diff
-#define SET_E_STEP_DIR(INDEX) \
-  E## INDEX ##_DIR_WRITE(e_steps[INDEX] <= 0 ? INVERT_E## INDEX ##_DIR : !INVERT_E## INDEX ##_DIR)
```
>```diff
-SET_E_STEP_DIR(0);
-#if E_STEPPERS > 1
-  SET_E_STEP_DIR(1);
-  #if E_STEPPERS > 2
-    SET_E_STEP_DIR(2);
-    #if E_STEPPERS > 3
-      SET_E_STEP_DIR(3);
-    #endif
-  #endif
-#endif
>

It seems that this is a precise fixing, too. (but stepper signal is still inverted, too.)

`g1 e100 f1200` | `g1 e100 f3000`
--- | ---
[image] | [image]

About pulse width and period:
>Is the pulse duration in advance_isr too short?

In case of `ADVANCED` and `LIN_ADVANCED` and no added any delays,
HIGH (but inverted) pulse width is 1.6 ~ 1.8μs, LOW pulse width on multi-stepping mode is under 1μs, pulse period is 2.6μs (≒ 384.61KHz).

From datasheets,
[A4988](http://www.allegromicro.com/~/media/Files/Datasheets/A4988-Datasheet.ashx?la=ja-JP) and [A5984](http://www.allegromicro.com/~/media/Files/Datasheets/A5984-Datasheet.ashx) needs 1μs HIGH/LOW pulse width at least,
[DRV8824](http://www.tij.co.jp/product/DRV8824/datasheet/specifications), [DRV8825](http://www.tij.co.jp/product/DRV8825/datasheet/specifications), and [DRV8834](http://www.tij.co.jp/product/DRV8834/datasheet/specifications) needs 1.9μs HIGH/LOW pulse width at least, and 250KHz step frequency(pulse period)is maximum.
[DRV8880](http://www.ti.com/product/DRV8880/datasheet/specifications) needs 470ns HIGH/LOW pulse width at least, and 1MHz step frequency(pulse period)is maximum.
[TMC2100](http://www.trinamic.com/_articles/products/integrated-circuits/tmc2100/_datasheet/TMC2100_datasheet.pdf) needs 120ns HIGH/LOW pulse width at least.
About [THB6128](http://www.datasheet.jp/download.php?id=817135)([RAPS128](http://reprap.org/wiki/RAPS128)), I don’t get it... ([needs 40ns HIGH/LOW pulse width at least?](https://www.stohn.de/3d/index.php/2014/09/06/thb-6128-driver-speed-test/))

esenapaj, 2016-09-19:
About `LIN_ADVANCE` part2:
>```diff
  #if ENABLED(LIN_ADVANCE)
-   volatile int Stepper::e_steps[E_STEPPERS];
+   volatile long Stepper::e_steps[E_STEPPERS];
```

I saw strange phenomenon.
When I changed the `e_steps[]` from  `volatile int` to  `volatile long`, 
HIGH pulse width grow to 3μs, LOW pulse width on multi-stepping mode grew to 1.4μs, pulse period grew to 4.4μs (≒ 227.27KHz) for some reason...

`g1 e100 f1200` | `g1 e100 f3000`
--- | ---
![newfile3](https://cloud.githubusercontent.com/assets/15308844/18621933/82232fce-7e64-11e6-9c44-c80b6d660a13.png) | ![newfile4](https://cloud.githubusercontent.com/assets/15308844/18621935/83872046-7e64-11e6-8793-71def92326c2.png)

>Allowing more than 32,767 E steps.

2,147,450,880?

A branch that it was used for test: https://github.com/esenapaj/Marlin/tree/testes

thinkyhead, 2016-09-20:
Good sleuthing! One of us should make a PR soon!

thinkyhead, 2016-09-20:
> When I changed the e_steps[] from volatile int to volatile long, 
HIGH pulse width grew to 3μs, LOW pulse width on multi-stepping mode grew to 1.4μs, pulse period grew to 4.4μs (≒ 227.27KHz) for some reason

Probably `long` takes `long`er to process. But this is probably a good thing. Pulse widths are probably too small. I think 8-9µs is considered best.

mosh1, 2016-09-20:
Very good sleuthing indeed. What is the theoretical effect of these issues when running with LIN_ADVANCE and these bugs not fixed?

thinkyhead, 2016-09-20:
Theoretically, erratic behavior.

esenapaj, 2016-09-20:
I submitted a [PR](https://github.com/MarlinFirmware/Marlin/pull/4852).

But, maybe more research and improvements are needed.

About pulse width:
`g1 e100 f3000`

Normal (not advance algorithms ) | `ADVANCE` | `LIN_ADVANCE`
--- | --- | ---
[image] | [image] | [image]

 | Normal | `ADVANCE` | `LIN_ADVANCE`
--- | --- | --- | ---
HIGH pulse width (μs)  | 6.7 | 1.81 | 3.0
LOW pulse width (μs) | 13.1 | __1.31__ | __1.38__
Pulse period (μs) | 19.8 | __3.12__ | 4.38

minimum requirements specification | DRV8825 | M542T
--- | --- | ---
HIGH pulse width (μs)  | __1.9__ | __1.5__
LOW pulse width (μs) | __1.9__ | __1.5__
Pulse period (μs) | __4.0__ | 3.0?

Marlin can adjust HIGH pulse width by `MINIMUM_STEPPER_PULSE`, but hasn't solution for changing LOW pulse width.

esenapaj, 2016-09-20:
__About between enable signal (EN) and direction signal (DIR), between direction signal and stepping signal (STP):__

minimum requirements specification | DRV8825 | M542T
--- | --- | ---
EN - DIR (μs)  | 0.65 | > 5
DIR - STP (μs) | 0.65 | > 5

But maybe anyone hasn't grasped  at this time that how much Marlin has those.

__About limitation of ISR frequency of `ADVANCE` and `LIN_ADVANCE`:__
I'm seeing that "dead zone" exists in transition of stepping rate with 5.18 : 1 geared extruder at 32 micro-stepping ([Bondtech QR](http://shop.bondtech.se/ec/extruders/bondtech-qr-175-universal.html), about 950steps / mm).
In case of `LIN_ADVANCE`, When I order `g1 e100 f600` (single-stepping), pulse timing fall into disorder, subsequently, Marlin freeze completely.
Green LED flash repeatedly, watchdog and reset button on RAMPS doesn't work. It needs complete power off.
But when I order `g1 e100 f630` (double-stepping), it looks like stable.
I'm still researching now.

`LIN_ADVANCE`

`g1 e100 f600` | `g1 e100 f630`
--- | ---
[image] | [image]

esenapaj, 2016-09-20:
>I'm guessing long takes long'er to process.

It appears so.
This is benchmark program that it was ported from [Arduino Playground - ShowInfo](http://playground.arduino.cc/Main/ShowInfo)
<details>
<summary>ShowInfo</summary>
```cpp
// -----------------------------------------------------
// Arduino ShowInfo
// Show what the Arduino has to tell you.
//
// 10 April 2012: Extra speed tests,
//                and a few other additions.
// 17 April 2012: The name 'ADCW' is preferred over 'ADC'.
// 23 April 2012: Changed PB5 into PORTB5.
// 1 July 2012  : Added i2c_scanner.
//                More speed tests.
// 21 July 2012 : Added divide speed tests.
// 25 July 2012 : Added sketchSize.
// 26 July 2012 : Changed sketchSize for more than 64kB.
//                Also show sketchSize in percentage.
//                Added processor defines.
//                Arduino Mega compatible (not tested).
// 5 November 2014 : Tested on Arduino Mega 2560
//                Added Arduino version "ARDUINO".
//                Replaced "double" with "float".
//                The i2c scanner scans now 1 up to 126,
//                  it was 0 up to 127.
//                Added UTF-8 test.
//                The delayMicroseconds(1) replaced by 
//                  2 micro seconds. I think one micro second
//                  was not called.
//                The analogReference() timing removed,
//                  I think it was not called.
// 4 December 2014 : Delay to allow the Serial.print to be printed.
//                More floating point and 32-bit integer tests.
//                Tested with Arduino 1.0.6 and Leonardo and Uno.
// -----------------------------------------------------

#ifndef __SAM3X8E__
  #include <avr/boot.h>
#endif
#include <Wire.h>

void setup() 
{
  // The system led is present on most Arduino boards.
  pinMode(13,OUTPUT);

  Serial.begin(9600);

#if defined (__AVR_ATmega32U4__)
  while(!Serial);        // For Leonardo, wait for serial port
#endif

  Serial.println(F("Arduino ShowInfo"));
  Serial.println(F("Show what the Arduino has to tell you."));

  ShowMenu();
}

void loop() 
{
  int incomingByte, i, n;

  if (Serial.available() > 0)
  {
    incomingByte = Serial.read();

    switch (incomingByte)
[... 1823 lines cut ...]
  divide long               : 0.069 us
  add long                  : 0.080 us

  multiply float            : 0.892 us
  divide float              : 7.219 us
  add float                 : 1.189 us

  itoa()                    : 1.509 us
  ltoa()                    : 3.194 us

  bitSet()                  : 0.070 us

  analogRead()              : 4.544 us
  analogWrite() PWM         : 3.629 us

  delay(1)                  : 999.994 us
  delay(100)                : 99999.992 us
  delayMicroseconds(2)      : 2.002 us
  delayMicroseconds(5)      : 5.003 us
  delayMicroseconds(100)    : 100.094 us
-----------
```
</details>

In case of MEGA2560, calculation of `long` is x 1.99 ~ x 4.58 slower than calculation of `int`.

Macgyver001, 2016-09-20:
**# Really great all that hard work.**
I will try LIN_ADVANCE part2:

      #if ENABLED(LIN_ADVANCE)
    -   volatile int Stepper::e_steps[E_STEPPERS];
    +   volatile long Stepper::e_steps[E_STEPPERS];

# **and**
About inverted stepper signal:
It seems that it's simple bug. In advance_isr(),

    #define START_E_PULSE(INDEX) \
-      if (e_steps[INDEX]) E## INDEX ##_STEP_WRITE(INVERT_E_STEP_PIN)
+      if (e_steps[INDEX]) E## INDEX ##_STEP_WRITE(!INVERT_E_STEP_PIN)

    #define STOP_E_PULSE(INDEX) \
      if (e_steps[INDEX]) { \
        e_steps[INDEX] <= 0 ? ++e_steps[INDEX] : --e_steps[INDEX]; \
-        E## INDEX ##_STEP_WRITE(!INVERT_E_STEP_PIN); \
+        E## INDEX ##_STEP_WRITE(INVERT_E_STEP_PIN); \

To see if that's working more stable ( without the : MINIMUM_STEPPER_PULSE = 0 )

thinkyhead, 2016-09-20:
#4852 is merged. Please confirm that it fixes the original issue.

thinkyhead, 2016-09-20:
> This is benchmark program

That is some very useful info. Perhaps it can be timed with all interrupts (temporarily) disabled, using `TCNT0` (or another counter) instead of `millis` to get completely accurate values.

Sebastianv650, 2016-09-21:
Great research!! I'm realy sad I still have no time to contribute in this topic (even my printer wasn't running the last months) but I definitly must so some tests with the merged PR :)

Thanks to everyone here :+1:

Macgyver001, 2016-09-21:
@thinkyhead is #4852 already landed/merged in the fixpatch branch? I can't find it back in the stepper.ccp file?

Macgyver001, 2016-09-21:
"fixpatch" I ment RCbugfix branch?

esenapaj, 2016-09-22:
PR #4852 (Fix for advance extrusion algorithms) was merged to [RCBugFix branch](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).

thinkyhead, 2016-09-22:
[image]

thinkyhead, 2016-09-22:
> In case of LIN_ADVANCE, When I order g1 e100 f600 (single-stepping), pulse timing fall into disorder, subsequently, Marlin freeze completely.
Green LED flash repeatedly, watchdog and reset button on RAMPS doesn't work. It needs complete power off.
But when I order g1 e100 f630 (double-stepping), it looks like stable.
I'm still researching now.

Did this resolve itself?

Macgyver001, 2016-09-22:
Sorry Thinkyhead for the question I am quite new with this.

I tried the PR4852 fix, but I have more or less the same as Esenapaj.
I tried it with **Lin_advance = 0** to start with. Then I god loads of this message:
Resend: 12307
Error:No Checksum with line number, Last Line: 12346
[ERROR] Error:No Checksum with line number, Last Line: 12346

The printer is working but slow and seems sometimes a bit off. (but the print finished, and the result is slightly worse than Lin_advance disabled. (1 inch cube and 5mm high)

I tried with **Lin_advance = 75**
Started the same print, but I stopped it after 1 minute, print result was **NOK** bad lines, but worse my heater controller didn't respond as normal with same intervals, but did unreliable on / off. Totally different than normal.

I have the feeling that the MINIMUM_STEPPER_PULSE =1 did a better job than the volatile long Stepper::e_steps[E_STEPPERS]; I think it's taking to long? While with MINIMUM_STEPPER_PULSE and CYCLES_EATEN_BY_E_CODE 12 (in my case) you can tune the minimum delay time and therefore minimize the impact to the code.

I will try the MINIMUM_STEPPER_PULSE =1 again with **Lin_advance = 75**, will update you next time.

thinkyhead, 2016-09-24:
@Macgyver001 I've made some tweaks to the way `MINIMUM_STEPPER_PULSE` is applied in #4887. Once that's merged, it should prevent the regular stepper ISR from gaining extra pulse-checking code when only the advance ISR needs it. You'll have to use a `MINIMUM_STEPPER_PULSE` of 4 or more, and/or tweak `CYCLES_EATEN_BY_E`.

Macgyver001, 2016-09-29:
I tried the latest RCBugFix branch 29-9-16, put Minimum_stepper_pulse to 4 and tried to get it working with Lin_advance = 75, still I have problems with my temperature controller.
I tried Minimum_stepper_pulse to 0 and still problems with the temperature controller, but yes the E stepper was extruding?
Then I tried Lin_advance = 0 & Minimum_stepper_pulse to 0 that was working, but got some Resend: 12307
Error:No Checksum with line number, Last Line: 12346
[ERROR] Error:No Checksum with line number, Last Line: 12346

After Lin_advance = 10 & Minimum_stepper_pulse to 0 the temperature controller got wrong again. 
The timing has changed, because the E stepper is extruding, but my system can't cope with I think the difference in timing of the temperature controller that expects the same time interval between samples?

Is is possible that the coreXY is using more processor power than the non-coreXY systems?
Is there any idea how many people use Lin_advane?

Sebastianv650, 2016-09-30:
We had the discussion about coreXY and the power it's using once before. It's quite possible that coreXY and LIN_ADVANCE together may overload the CPU. I don't know if somebody ever got Advance to work on a coreXY up to now..

Regarding how many people are using it: I know about one or two Lulzbot TAZ users are using it with success and myself of course. There might be others?

mosh1, 2016-09-30:
I've definitely been using it with good success / E3D Bigbox (Cartesian gantry).

esenapaj, 2016-10-01:
@thinkyhead 
>Did this resolve itself?

With newest RCBugFix, it respond erratic result.
Pulse timing became stable after the PR #4868 (Cleanups for the Autumn release), 
But when e steps/mm is 953.1 (Bondtech QR at 32 micro-step) and ordered a `g1 e100 f600`, results are
・Not freeze, but Repetier-Host spout a lot of "Error:checksum mismatch" and "Error:No Checksum with line number" and "Resend".
or
・Freeze silently and watchdog doesn't work, but reset button of RAMPS and Repetier-Host works.
or
・Freeze with green LED on RUMPS flushing, watchdog and reset button on RAMPS and Repetier-Host doesn't works.

Marlin has the threshold of between single-stepping and double-stepping that it's set 10000(steps/sec or Hz).
But in case of `ADVANCED` and `LIN_ADVANCED`, It looks like that it's too heavy for 16MHz AVR.

Macgyver001, 2016-10-01:
@thinkyhead  & @esenapaj 

With RCBugFix branch 29-9-16 and Lin_advance = 10 & Minimum_stepper_pulse to 0 I disabled the Watchdog, when I do this the temperature controller is just NOT making it, but way better than when I Enable Watchdog. So I also think that this Lin_advanced is using the 16MHz AVR to the limit.

I went back to the RC7 31 july 16, disabled the watchdog and enabled Lin_advance = 10
I adjusted the stepper.ccp code

  void Stepper::advance_isr() {

    old_OCR0A += eISR_Rate;
    OCR0A = old_OCR0A;

    #define STEP_E_ONCE(INDEX) \
      if (e_steps[INDEX] != 0) { \
        asm("nop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\n"); \
        E## INDEX ##_STEP_WRITE(!INVERT_E_STEP_PIN); \
        if (e_steps[INDEX] < 0) { \
          asm("nop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\n"); \
          E## INDEX ##_DIR_WRITE(!INVERT_E## INDEX ##_DIR); \
          e_steps[INDEX]++; \
         } \
        else { \
          asm("nop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\nnop\n"); \
          E## INDEX ##_DIR_WRITE(INVERT_E## INDEX ##_DIR); \
          e_steps[INDEX]--; \
        } \
        E## INDEX ##_STEP_WRITE(INVERT_E_STEP_PIN); \
      }

I don't know if this is the wright way, it did not succeed yet, I will try to add more nop's? (E stepper is not moving yet) temperature control seems ok
To be continued

thinkyhead, 2016-10-02:
> But in case of ADVANCED and LIN_ADVANCED, It looks like that it's too heavy for 16MHz AVR.

I'd be interested to know where we are hitting the limits, and compare this to older versions of Marlin to see where we stand in terms of comparative performance. I'm sure there is some added overhead in various places.

Mainly we should time the core operations…
- `Stepper::isr()`, stepping CPU requirements in typical printing situations.
- Planner `buffer_line` to see how long it takes to make an average block
  - Can the planner fill in a block in advance of waiting for a free block?
  - (Apparently it can pre-process some before that wait)
- LCD event handling: `lcd_update()` called from `idle()`.
- `idle()` and the things it calls: `host_keepalive()`, `manage_inactivity()`, `Temperature::manage_heater()`, `buzzer.tick()`,  `print_job_timer.tick()`.

Other performance-related questions:
- Should the watchdog get reset more often, not just when a temperature reading is ready? Does the watchdog reset belong in a more generic place, such as the main `idle()` function?
- Should the `idle()` function manage the time-slicing of the things it calls, instead of calling them only to get back a "not yet" most of the time?
- Can we use "throttling" to free up CPU –for example reducing the frequency of the `Temperature::isr` and the LCD update– whenever time becomes constrained?
- Display sleep? It can free up a lot of cycles to "sleep" the display.
- Low Power LCD mode? Drawing between 1 - 3 lines of information…
  - `195°_Z:5.40_SD25%___`
  - `_70°_1d_01:11:19____`
  - `F_95%_Short_Message.`

[9 more replies]

closed 2016-11-16 by thinkyhead