MarlinFirmware/Marlin #4550: Delta Printer Auto bedlevel

myschetzky, 2016-08-06:
Hello, 

I can't get the bed level feature to work on my Kossel XL delta printer, with a BLTouch probe.
The printer probes fine, but it seems like it doesn't use the probed data to compensate. 

I've tried with the latest RCBugfix version without luck.

I should probably mention, that the printer prints "okay" without the auto leveling.

I feel like I forgot to enable a feature somewhere?!

myschetzky, 2016-08-06:
Here is my Configuration.h

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/405170/Configuration.txt)

myschetzky, 2016-08-06:
And the log from a G29 sent in command

SENT: G29
RECEIVED: G29 Auto Bed Leveling
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: -0.11 -0.26 -0.41 -0.68 -0.95 
RECEIVED: 0.23 0.09 -0.04 -0.31 -0.57 
RECEIVED: 0.00 0.34 0.57 -0.22 -0.63 
RECEIVED: 0.88 0.60 0.31 0.32 0.32 
RECEIVED: 1.48 1.27 1.06 0.11 0.07 
RECEIVED: X:55.00 Y:103.00 Z:35.00 E:7.57 Count X: 21216 Y:28979 Z:35521
RECEIVED: ok

myschetzky, 2016-08-06:
And here is a test gcode. I removed most of the coordinate data to save upload space.
[caldeltaV2.txt](https://github.com/MarlinFirmware/Marlin/files/405173/caldeltaV2.txt)

thinkyhead, 2016-08-07:
#4520

aljscott, 2016-08-11:
I have to agree after going through #4520 - it appears that the auto level compensation is not being applied.

thinkyhead, 2016-08-11:
> it appears that the auto level compensation is not being applied.

Uncomment the debug output lines in the `adjust_delta` function and re-flash. After leveling with `G29` all delta moves should invoke this function and print detailed output to the serial console.

myschetzky, 2016-08-12:
These lines?       
//DEBUG_POS("prepare_kinematic_move_to", target);
//DEBUG_POS("prepare_kinematic_move_to", delta);

myschetzky, 2016-08-12:
With those lines uncommented the console is displaying a lot of "prepare_kinematic_move" data.

SENT: G1 X-65.307 Y-49.123 E21.2635
RECEIVED: target=(-66.28, -47.79, 0.10) : prepare_kinematic_move_to
RECEIVED: delta=(319.30, 248.04, 237.68) : prepare_kinematic_move_to
RECEIVED: target=(-66.03, -48.12, 0.10) : prepare_kinematic_move_to
RECEIVED: delta=(319.28, 248.30, 237.43) : prepare_kinematic_move_to
RECEIVED: target=(-65.79, -48.46, 0.10) : prepare_kinematic_move_to
RECEIVED: delta=(319.25, 248.57, 237.18) : prepare_kinematic_move_to
RECEIVED: target=(-65.55, -48.79, 0.10) : prepare_kinematic_move_to
RECEIVED: delta=(319.23, 248.83, 236.93) : prepare_kinematic_move_to
RECEIVED: target=(-65.31, -49.12, 0.10) : prepare_kinematic_move_to
RECEIVED: delta=(319.20, 249.09, 236.68) : prepare_kinematic_move_to
RECEIVED: ok
SENT: G1 X-65.257 Y-49.190 E21.2688
RECEIVED: target=(-65.31, -49.12, 0.10) : prepare_kinematic_move_to
RECEIVED: delta=(319.20, 249.10, 236.68) : prepare_kinematic_move_to
RECEIVED: ok
SENT: G1 X-65.254 Y-49.194 E21.2691
RECEIVED: target=(-65.26, -49.19, 0.10) : prepare_kinematic_move_to
RECEIVED: delta=(319.19, 249.15, 236.63) : prepare_kinematic_move_to
RECEIVED: ok

Does this tell you anything?

boelle, 2016-08-14:
@thinkyhead does it make sense to you?

aljscott, 2016-08-14:
I saw somewhere through some random browsing where G28 resets the auto level data?
So my slicer automatically starts the print with a G28, so it was resetting the bed levelling.
So I changed the default to a G28 followed by G29 at the start of the print.
Seems to be applying now.

myschetzky, 2016-08-16:
Thanks for the suggestion @aljscott. Unfortunately it isn't the case here. :/

thinkyhead, 2016-08-28:
@myschetzky Be sure to re-download `RCBugFix` and test again soon, as there have been a few tweaks in this area. And if you still see issues, please test one or both of my branches, which are attempting to produce the most accurate Z position at the end of leveling.

This one adjusts Z based on the homing position:
https://github.com/thinkyhead/Marlin/tree/rc_fix_leveling_maths

And this one adjusts Z based on the final probe position:
https://github.com/thinkyhead/Marlin/tree/rc_final_z_correction

boelle, 2016-09-03:
@myschetzky have you retested?

myschetzky, 2016-09-06:
Thank you @thinkyhead . I'll test asap but unfortunately my work is not giving me much spare time at the moment. 

@boelle No not yet. :-/

boelle, 2016-09-20:
@myschetzky had a bit of spare time?

closed 2016-10-26 by thinkyhead