MarlinFirmware/Marlin #4557: Feature Request: Auto meassure Z height

boelle, 2016-08-07:
I always struggle to get my Z height correct

Given i have defined Z max endstop i would be nice to issue a command that homed Z and then told my how far up the endstop is

thinkyhead, 2016-08-07:
There's a trick you can currently do, without needing a new feature. After homing Z to Max, move the nozzle all the way down to the bed, where you want "zero" to be. And then choose "Set Home Offset" from the LCD, or use `M428` from the console. This command will set the current Z height as your Z=0 and offset the position of your Z Max Endstop. You can save the result with `M500`. Next time you do `G28 Z` you will see that your Z position is offset from the `Z_HOME_POS` you set in your Configuration. Note that this works only if you set your `Z_HOME_POS` (`Z_MAX_POS`) a bit higher than actual.

strips, 2016-08-09:
This could be combined with this feature: https://github.com/MarlinFirmware/Marlin/issues/4531#issuecomment-238537694

closed 2016-12-06 by boelle