MarlinFirmware/Marlin #4560: Error:Z-Probe failed  move down to retract probe Delta

Anion-anion, 2016-08-07:
The problem with retracting Z-probe on Kossel Delta printer.
Deployed well, is the whole table, too, but when you revert, just for the bolt and stops. The display says Error:Z-Probe failed move down to retract the probe.
Apparently triggered z-probe, as the limiter in top.
Tell me , what did I miss?

my configuration here.

Проблема с убиранием Z probe на дельта коссель принтере.
Развертывается хорошо, проходит весь стол тоже, но при убирании, чуть касается болта и останавливается. На дисплее пишет Error:Z-Probe failed  move down to retract probe.
Видимо срабатывает z-probe, как ограничитель во высоте.
Подскажите , что я упустил?
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/405716/Configuration.h.txt)

Blue-Marlin, 2016-08-07:
Likely yesterday solved by "Make Allen Key Probe stow #4551" (RcBugFix)

Anion-anion, 2016-08-07:
Blue Marlin, thank you)
Works!

boelle, 2016-08-11:
he said it works so why no close it? and its a dublicate

closed 2016-08-11 by Anion-anion