MarlinFirmware/Marlin #4561: [Help] How to make the probe stay deployed during the entire G29 E routine

smokinpuppy, 2016-08-07:
Hi,
I used to use the G29 E code to have my probe stay down/deployed during the entire auto bed leveling routine. Ever since I upgraded to 1.0.2-1 the G29 E looks just like the regular G29. How can I make it behave the same way as before?

Thank you in advance,

Robert

Blue-Marlin, 2016-08-07:
>  * E/e By default G29 will engage the Z probe, test the bed, then disengage.
   *     Include "E" to engage/disengage the Z probe for each sample.
   *     There's no extra effect if you have a fixed Z probe.
   *     Usage: "G29 E" or "G29 e"

If you do not want the probe to deploy/stow for every probe, don't use the 'E' parameter. Sorry if 'E' is working now on your system.

smokinpuppy, 2016-08-07:
Thanks Blue-Marlin for your reply. The behavior is the same either way with the G29 alone or G29 E it engages/disengages the z probe for each sample. Before, G29 E used to leave the probe down. So how can I get that behavior back? (i.e leave the probe down during the auto level routine)

Blue-Marlin, 2016-08-07:
What kind of probe are you talking about?
Config. please.

smokinpuppy, 2016-08-08:
Sorry, I'm using a servo probe on a prusa I2 printer. Please find my config file here: [http://pastebin.com/x8qEz20N](http://pastebin.com/x8qEz20N)

thinkyhead, 2016-08-08:
The [G29 routine in 1.0.2-1](https://github.com/MarlinFirmware/Marlin/blob/1.0.x/Marlin/Marlin_main.cpp#L1666) doesn't take any arguments. The fancier G29 is only being introduced with 1.1.0.

closed 2016-08-08 by thinkyhead

Blue-Marlin, 2016-08-08:
Your config. belongs to Marlin 1.0.2. Sorry, this is from 18 month ago. I have no clue, how things have been then. We will not make any code changes for that branch anymore. Please update to RCBugFix.

thinkyhead, 2016-08-08:
Note to the @MarlinFirmware/documentation-maintainers that we need to make a note of the version of Marlin where a feature appears. The only versions we need to be concerned with are 1.0.1, 1.0.2, and 1.1.0.