MarlinFirmware/Marlin #4565: LCD Timer resets to 00:00 after 24 Hour Print

psavva, 2016-08-08:
The Timer on the LCD Resets back to 00:00 after 24 Hour Print.
ie: 25 Hours will become 01:00 and not 1day 01H 00 MI or alternatively 25:00  (HH:MI) or similar

thinkyhead, 2016-08-09:
Are you referring to the clock on the Info Screen? You can get up to 99:59 with…
```diff
  void toDigital(char *buffer) const {
-   int h = this->hour() % 24,
+   int h = this->hour(),
        m = this->minute() % 60;

    sprintf_P(buffer, PSTR("%02i:%02i"), h, m);
  }
```

psavva, 2016-08-09:
I guess this must be from dogm_lcd_implementation.h
 Line ~375
```
    uint16_t time = print_job_timer.duration() / 60;
    if (time != 0) {
      lcd_print(itostr2(time/60));
      lcd_print(':');
      lcd_print(itostr2(time%60));

```

thinkyhead, 2016-08-09:
I see. There's more space on the graphical display, so a days count could fit there. On the character-based display we only have space for hh:mm.

boelle, 2016-08-11:
are we keeping this to add days also?

jbrazio, 2016-08-11:
Fixed by #4572.
Please reopen this ticket if the problem still persists.

closed 2016-08-11 by jbrazio