MarlinFirmware/Marlin #4566: Compile error in macros.h

fitch22, 2016-08-08:
I just tried to build from snapshot taken today.  My printer uses a printrboard.  I tried Arduino 1.6.9 and 1.6.10, both seem to generate the same error.  Probably something I am doing, but for the life of me I can't figure it out.  The errors are all similar to:

macros.h:68: error: pasting "SWITCH_ENABLED_" and "(" does not give a valid preprocessing token
 #define ENABLED(b) _CAT(SWITCH_ENABLED_, b)

Can anyone help?

Blue-Marlin, 2016-08-08:
As a workaround try:
SanityCheck.h
```diff
@@ -742,8 +742,8 @@
 #endif
 
 /**
  * emergency-command parser
  */
-#if ENABLED(EMERGENCY_PARSER) && ENABLED(USBCON)
+#if ENABLED(EMERGENCY_PARSER) && defined(USBCON)
   #error "EMERGENCY_PARSER does not work on boards with AT90USB processors (USBCON)."
 #endif
```

fitch22, 2016-08-08:
Awesome!  That worked.  Thank you so much.

closed 2016-08-08 by fitch22