MarlinFirmware/Marlin #4568: "MINTEMP BED" working not as expected

jangrewe, 2016-08-08:
Hi, i'm currently working on (mostly with success) getting the current RC to work on a Fabrikator Mini with a heated bed, which i haven't wired in yet because i'm still waiting for the PSU... but anyways, so i set `TEMP_SENSOR_BED` to `5` and flashed the new firmware, but was greeted with a "MINTEMP BED" error and was told to reset the printer because it was halted. Bummer.

I had to jury-rig the thermistor to the board so that i could at least reflash the firmware to revert`TEMP_SENSOR_BED` to `0`.

Would it be possible to only halt the printer if the bed is actually heating? Of course not by meassuring the temperature, which is exactly why this feature exists, but by checking if the heater has been enabled.

That way it would be reset when you reset, and you can use your printer without the bed (e.g. if the thermistor is broken).

Maybe i'm just a bit naive and totally don't understand what's going on, but in my head this would make sense! ;-)

thinkyhead, 2016-08-09:
I believe this will do what you want…
```diff
- if (bed_minttemp_raw GEBED current_temperature_bed_raw) _temp_error(-1, PSTR(MSG_T_MINTEMP), PSTR(MSG_ERR_MINTEMP_BED));
+ if (bed_minttemp_raw GEBED current_temperature_bed_raw && target_temperature[3] > 0.0f) _temp_error(-1, PSTR(MSG_T_MINTEMP), PSTR(MSG_ERR_MINTEMP_BED));
```
…and I will submit it as a proposed patch.

jangrewe, 2016-08-09:
Thank you, i will test and report back when i get home!

jangrewe, 2016-08-09:
Works perfect now! Shorting the thermistor leads (by accident) still results in MAXTEMP & halt, but that is probably a good idea.
Enabling the heating for the bed results in MINTEMP, so that is still working as expected. 

Now everything behaves as i would it expect it, Thank you!

closed 2016-08-09 by jangrewe