MarlinFirmware/Marlin #4575: Compiler warning after merge #4574

WheresWaldo, 2016-08-09:
Downloaded this Aug 9 and getting the following warning trying to compile on Arduino 1.6.10

> Arduino: 1.6.10 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
>
> In file included from sketch\ultralcd.cpp:48:0:
>
> sketch\ultralcd_impl_DOGM.h: In function 'void lcd_implementation_status_screen()':
>
> sketch\ultralcd_impl_DOGM.h:387:43: warning: integer overflow in expression [-Woverflow]
>
>     bool has_days = (elapsed.value > 60*60*24);
>                                            ^
>
>
> Sketch uses 117,548 bytes (46%) of program storage space. Maximum is 253,952 bytes.
> Global variables use 4,970 bytes (60%) of dynamic memory, leaving 3,222 bytes for local variables. Maximum is 8,192 bytes.
> An error occurred while uploading the sketch
>
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
>

WheresWaldo, 2016-08-09:
I am not getting this consistently, it appears to compile correctly randomly, will try to investigate possible causes.

Roxy-3D, 2016-08-09:
I do not like the sound of that!   It should be consistent what ever it does!

Blue-Marlin, 2016-08-09:
Random? No. But only if a graphical display and SDSUPPORT is on and you are compiling the first time. (If you don't change the source file, or one of its includes it will not be compiled again because the already existing *.o file can be reused. (make))

60\*60\*24=86400 > 2^16. So a int is to small.

`elapsed.value` is uint32_t so ok.
To expand the right side to a long use
  
60\*60\*24 __L__

WheresWaldo, 2016-08-09:
@Blue-Marlin That appears to be what I am seeing. I do have LCD and SDSUPPORT enabled. It will still upload to the RAMPS board, and I have never done a print exceeding 22 hours, so I wouldn't have days listed.

thinkyhead, 2016-08-10:
The "L" has been added and merged.

WheresWaldo, 2016-08-10:
Does this need a test or can I just close it?

closed 2016-08-11 by thinkyhead

boelle, 2016-08-11:
@thinkyhead ^^ what he just asked

boelle, 2016-08-11:
upps.... @thinkyhead i forgot the "read 1x extra before you click comment"