MarlinFirmware/Marlin #4576: Error in Printer Stats -> "Extruded total" jumps around when turning encoder

Kaibob2, 2016-08-09:
This is a little vague, but my programming skills came to an end here, so i cannot say exactly where and what the problem is. This is what i observe in 1.1.0-RCBugFix.

**Printer in standby:**
When i open the "Printer Stats" screen the "Extruded total" value is somewhere, lets say 30000m (which is pretty much!! Realistic would be 20m) 
When i turn the encoder clockwise the menu scrolls down. When the last line shows nothing happens.

**Printer printing:**
When i open the "Printer Stats" screen the "Extruded total" value is somewhere, lets say 30000m 
When i turn the encoder clockwise the menu scrolls down. As soon as the menu reaeches the last line the "Extruded total" value can be altered by turning the encoder. The value jumps quite fast, also in negative direction.
The "Extruded total" value doesn't change by itself while extruding.

thinkyhead, 2016-08-10:
Here's part of the problem. First, let's see the definition of `filamentUsed`:
```cpp
double filamentUsed;
```
And here's where it is printed:
```cpp
sprintf_P(buffer, PSTR("%im"), stats.filamentUsed / 1000);
```
Formatting a `double` (i.e., a `float`) as an integer won't do! So first we need to patch that… then we can see whether the value is being changed by rotating the knob, or if it's actually changing by a tiny amount, but the conversion to `int` makes it look like more.

Kaibob2, 2016-08-10:
These where the two lines i expected to be the cause, but, like mentioned above, my coding skills are limited :)
I will test the fix as soon as i'm home and report back.

boelle, 2016-08-11:
have you had time to test the fix?

Kaibob2, 2016-08-11:
@boelle: Yes and no, i can confirm that the Extruded total value was 56m yesterday evening (which is realistic) with the most recent RCBugFix. I forgot to check what turning the encoder does to it. I expect it will be what @thinkyhead mentioned:
> then we can see whether the value is being changed by rotating the knob, or if it's actually changing by a tiny amount, but the conversion to int makes it look like more.
Further i didn't see a decimal place, it was just "56m". Maybe because of beeing exactly 56.0m and the 0 won't show or the decimal place isn't there always. I would appreciate at least one decimal place.
But i will verify this today and report back.

I had to revert back to a preview branch because i experienced PWM not worling for PIN4 fan. I will try to figure out why today, but this got nothing to do with the Extruded total thing.

Kaibob2, 2016-08-11:
@thinkyhead Everything is fine now. The Extruded total value shows a realistic value and it doesn't change when turning the encoder. Thanks!
It would be nice to have at least one decimal place. I guess this should be possible, i just don't know how without breaking something :)

thinkyhead, 2016-08-11:
> It would be nice to have at least one decimal place

```cpp
sprintf_P(buffer, PSTR("%ld.%im"), long(stats.filamentUsed / 1000), int(stats.filamentUsed / 100) % 10);
```

Kaibob2, 2016-08-12:
Works like a charme. This looks really good now. Many thanks!
I would suggest to merge this line to RCBugFix.

[image]

boelle, 2016-08-12:
so now we can close this one?

Kaibob2, 2016-08-12:
Absolutely. Thanks for the support.

commit 58c8e6ce (2016-08-10): Fixes #4576: Convert filamentUsed to long for display
closed 2016-08-12 by Kaibob2