MarlinFirmware/Marlin #4579: Set auto fan pins to RAMPS D8 pin

diegotrap, 2016-08-09:
I have set the `Configuration_adv.h` parameters to be able to use the D8 (MOSFET) port in my RAMPS 1.4 to connect an extruder fan. This fan has to be ON whenever the hotend is working (typical case for an all metal extruder). These are the resulting lines of code:

> #define EXTRUDER_0_AUTO_FAN_PIN 8 //RAMPS_D8_PIN
> #define EXTRUDER_1_AUTO_FAN_PIN -1
> #define EXTRUDER_2_AUTO_FAN_PIN -1
> #define EXTRUDER_3_AUTO_FAN_PIN -1
> #define EXTRUDER_AUTO_FAN_TEMPERATURE 5
> #define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed

This doesn't turn on the fan when the temperature is above 5ºC (always). I have tested it in both RAMPS_14_EFB and RAMPS_14_EFF pin maps.

I think the problem is that pin number 8 is already been used for the bed/other fan and this setting is overriden, but it should be a way to make it work. Many printers use a noisy hotend fan and it would be a great way to turn it off without disconnecting the whole printer.

thinkyhead, 2016-08-11:
What version of Marlin are you using?

diegotrap, 2016-08-11:
Marlin 1.10 RC7

thinkyhead, 2016-08-11:
Do you experience the same issue with `RCBugFix`? I'm looking at the RAMPS pins now, and it seems that as long as you're using `RAMPS_14_EFB`, `EEB`, or `EFF` you should have no conflict. In fact, you ought to only have a conflict in the first 2 cases if you are also trying to use a heated bed.

Try with `RCBugFix` and if you still can't get it to work, post your `Configuration.h` and `Configuration_adv.h` so we can test-compile with them and figure out what's going on.

boelle, 2016-08-11:
hmm... he might be on to something

one output cant rarely be tied to more than one pin in the pins file...

maybe test this theory by setting all FAN_PIN to -1

just a thought

thinkyhead, 2016-08-11:
@boelle I have a technique to see exactly what's going on. I use the given config, and then I add code that checks the values of the pins I'm interested in. When I compile, I can see whether they are set to expected values or something else. If needed, I can add code to print out the values of all the pins I'm interested in at runtime.

diegotrap, 2016-08-12:
I have upgraded to `RCBugFix` and tried again, the fan still doesn't turn on. I upload the `Configuration.h` and `Configuration_adv.h` as `.txt` so you can replicate this error and we can learn from it.

Thank you very much for your help, thinyhead.

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/415434/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/415435/Configuration.txt)

thinkyhead, 2016-08-14:
Your configuration is good, and everything seems to be setting up correctly, so the `checkExtruderAutoFans` function is certainly being invoked. I'll test the auto-fan code on one of my printers in a little while, but in the meantime here's one thing you can try. In `temperature.cpp` change the code in the `Temperature::init` function that initializes `EXTRUDER_0_AUTO_FAN_PIN` to the following…

```cpp
  #if HAS_AUTO_FAN_0
    #if EXTRUDER_0_AUTO_FAN_PIN == FAN1_PIN
      SET_OUTPUT(EXTRUDER_0_AUTO_FAN_PIN);
    #else
      pinMode(EXTRUDER_0_AUTO_FAN_PIN, OUTPUT);
    #endif
  #endif
```

thinkyhead, 2016-08-14:
I just did a test of `RCBugFix` without that change, and it seemed to work fine. As soon as the extruder reached the configured temperature the fan came on at full speed. When the temperature fell the fan turned off. As you do your testing, double-check to make sure the fan is connected to the correct plug, and that all the wiring is good.

boelle, 2016-08-14:
@diegotrap had any chance to test this?

diegotrap, 2016-08-16:
I have a fan connected with the right polarity to the D8 RAMPS pin. I have changed the fan, measured with a multimeter and there is no signal in that output. 

Should I try all the new commits?

diegotrap, 2016-08-16:
I have tested the latest version of `RCBugFix`, with the new commits related to the fans.

I still can't make it work. I have used two different RAMPS, checked all the hardware part and have not being able to get the fan to turn on.

Have you make it work with my same settings in a physical machine?

thinkyhead, 2016-08-19:
> Have you made it work with my same settings in a physical machine?

More or less. I was easily able to override the pin and have it function on my Delta, which is also a RAMPS 1.4 variant. As I wrote above:

> As soon as the extruder reached the configured temperature the fan came on at full speed.

I didn't borrow all your settings, but there aren't any other settings that could prevent the auto fan code from being invoked. You only need to set `EXTRUDER_0_AUTO_FAN` to something other than -1 and the code will be enabled.

Try this: Scroll down to the `Temperature::checkExtruderAutoFans` in `temperature.cpp` and make the following change, then try and compile. It should throw the error if all is set correctly.
```diff
#if HAS_AUTO_FAN

  #error "Yep! This code is being compiled!"

  void Temperature::checkExtruderAutoFans() {
    . . .
```

Jochem77, 2016-08-21:
I had the same problem. But when I changed the heated bed pin in pins_RAMPS_13.h to -1 the fan turned on.

#if MB(RAMPS_13_EFF) || MB(RAMPS_13_EEF) || MB(RAMPS_13_SF)
  #define HEATER_BED_PIN   -1    // NO BED
#else
  #define HEATER_BED_PIN    -1    // BED was: 8
#endif

thinkyhead, 2016-08-21:
> when I changed the heated bed pin

@Jochem77 What is your `MOTHERBOARD` setting? And are you using `RCBugFix`? Because it has changed the way that the MOSFET pins are reckoned. Based on your feedback, I'll take a look again at the pins files and make sure it's handling various cases properly.

Jochem77, 2016-08-21:
I have 2 printers: one with a Megatronics v3.1 and one with Ramps 1.4. I have no heated bed. Extruder is connected to MOSFET D10, filament fan to MOSFET D9 and E3D extruder fan to MOSFET D8. 
I'm using Marlin 1.1.0-RC6.

thinkyhead, 2016-08-21:
@Jochem77 I see. Since RC6 this has changed. Marlin `RCBugFix` now auto-assigns the pins based on your configuration. It presumes for Extruder-Fan-Fan setups that the Extruder is (as always) on D10, the first (PWM) Fan is on D9 and the second fan is on D8. So you should find that `RCBugFix` does the right thing without needing to do any pin-tweaking.

boelle, 2016-08-28:
@Jochem77 @thinkyhead maybe close this one as it seems fixed?

closed 2016-08-29 by thinkyhead

diegotrap, 2016-08-29:
I couldn't fix my issue and tried several times with your proposed solutions. Nevertheless, I don't have much more time to test this and I understand that the thread is closed.

I'll try to have a look again in the future, thanks for your help anyways :)

thinkyhead, 2016-08-30:
@diegotrap If you are able to collect more diagnostics on your end I'm sure we can track it down. The feature still works for me just fine with `RCBugFix`, using `BOARD_RAMPS_14_EFF` and…
```cpp
#define EXTRUDER_0_AUTO_FAN_PIN FAN1_PIN
```
The pin gets defined and then the auto fan just works.

a29er, 2017-06-07:
I have had the exactly the same problem and spent few hours solving it. It turned out that there was no input power on 10A 12V which is used by D8's MOSFET. I just connected 10A input to 5A and the problem gone.

diegotrap, 2017-06-08:
Thank you very much a29er! That was the issue, I am sure. Since I wasn't
using a heated bed I hadn't connected the 10A power line.

Thank you very much for your help and for taking the time to share it here
with the community.

viktorglekler, 2018-12-12:
Hi guys! Do you know how to enable SOFT_PWM for AUTO_FAN? In my 
Configurations_adv.h : 
#define E0_AUTO_FAN_PIN FAN1_PIN

and in Configurations.h : 
#define FAN_SOFT_PWM

but the FAN_1 spin with standard 500Hz frequency.. I have tried to define FAST_PWM_FAN, but it does change the PWM frequency too. 

If I use  #define E0_AUTO_FAN_PIN -1 in Configurations_adv.h then changing the frequency works for FAN1_PIN

viktorglekler, 2018-12-12:
Hi, AnHardt. With 500Hz I do not mind the rps of the fan, but PWM frequency on FAN1_PIN Output. The FAN_SOFT_PWM or FAST_PWM_FAN does not work for FAN_PIN assigned to AUTO_FAN_PIN. It seems that If I define AUTO_FAN it rund only with standard analogWrite(). I want to change pwm frequency cause this 500hz are very annoying..

thinkyhead, 2018-12-12:
Correct. At this time `FAN_SOFT_PWM` is not implemented for the auto fans, only for the three fans which are controlled by `M106`.

viktorglekler, 2018-12-13:
What a pity.. Then 

> > …`FAN_SOFT_PWM` … does not work for `FAN_PIN` assigned to `AUTO_FAN_PIN`…
> 
> Correct. At this time `FAN_SOFT_PWM` is not implemented for the auto fans, but only for the three fans which are controlled by `M106`.

What a pity..  Then I have an another solution for me, but I need your help. I want to turn the FAN1 on by turning on my 3D printer. Can I just set the pwm value for that FAN_PIN in loop() function? What for function do I need for setting the pwm value if I want to use SOFT_PWM? Thanks in advance!

thinkyhead, 2019-01-03:
> So it works now with "fanSpeeds[FanNumber] = val;" inside main loop()

That's a fine solution.