MarlinFirmware/Marlin #4581: Nozzle heating problem.

BraiEc, 2016-08-09:
Hi Guys.

I have a problem with my 3d printer. I choose the Marlin RC6 and i changed the parameters in configuration.h and configuration_adv.h but i can´t heat the extruder in RC6 and RC7.In pronterface i put m104 s90 or something like but it couldn´t work. In Marlin 1.0.2 works well and the parameters are similar in RC6 and 1.0.2. 

Sorry for my english.

brainscan, 2016-08-10:
When you send m104 or m109 do you get any response from the printer or pronterface? Can you hear the load on the power supply? Could you please post the config files you've changed, might be worthwhile trying latest RCBugFix too.

BraiEc, 2016-08-10:
Hey bro, thanks for your answer... m104 printer send me PRINTER HALTED ERROR and the m109 printer don´t send me anything. Both of them are sent with 1.1.0 RCBugFix. I have a mistake the last comment i do the test with Marlin before the Release Candidates. i don´t know the version number... but this marlin heat the extruder in a good way... but the RC versions don´t do that. I post the code with Marlin before the RC and the Marlin RC6. The bed is heated well in both.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/411524/Marlin.zip)

brainscan, 2016-08-10:
In those pictures you've just sent M104 and M109 without any other parameters, do you get the same response if you tell it a temperature.

BraiEc, 2016-08-10:
I put the values in other test but the extruder can't heat the value are display in the pronterface and the RepRap LCD but don't heat.

brainscan, 2016-08-10:
/ The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB
#endif

Should be like this, I don't think 34 works anymore.

brainscan, 2016-08-10:
Maybe it does though or surely it wouldn't compile?!?

BraiEc, 2016-08-10:
you can see in the boards.h file... you can use 34 or BOARD_RAMPS_14_EFB. this code compile but the heat bed can heat but the extruder can´t heat.

brainscan, 2016-08-10:
Yeah my mistake, I missed that change as mine has always been the default anyway. It was the first thing that jumped out at me, I should have thought more before commenting. 

I can't see anything obvious but if it works on an old version then the hardware must be ok. Are you using the latest RCBugFix (as in today) or one that's a few days old?

BraiEc, 2016-08-10:
I use all the RC versions but any version works with my printer but the other code works well.

thinkyhead, 2016-08-11:
@BraiEc Welcome to Github! I hope we can solve this for you quickly.

closed 2017-10-31 by thinkyhead