MarlinFirmware/Marlin #4582: Delta kossel homing weirdness RC7

mjmeans, 2016-08-10:
Delta Kossel homing in RC7 (using the sample delta config's) operates differently than in RC6. In RC6 the homing would move to the top and stop there. In RC7 the homing operation moves to the top and then then moves down and diagonally to approximately X60, Y60, Z201.75. There are several problems with this:

1. The screen shows X120, Y120, Z201.75. There's no way the head is out at x120, y120 as the screen shows.

2. Where is this Z move to 201.75 coming from? It doesn't exist anywhere in my configuration.ini.

3. It should home to X0, Y0; and not off to the side. RC6 had #defines for MANUAL_X_HOME_POS and MANUAL_Y_HOME_POS set to 0 and RC7 has them commented out. So either BED_CENTER_AT_0_0 is not calculating correctly for this configuration, or the other manual hope positions should not be commented out.

Oh, I should also say I commented out the Z_PROBE_ALLEN_KEY. My Kossel is manually calibrated and I don't use a Z_PROBE.

mjmeans, 2016-08-10:
Okay, I figured out where the Z201.75 comes from. But I still think MANUAL_X_HOME_POS and MANUAL_Y_HOME_POS should either be uncommented (set to 0), or the comment about automatic processing should be fixed so it actually does go to 0,0.

thinkyhead, 2016-08-10:
The Marlin package contains several example configurations. All the ones for delta have `BED_CENTER_AT_0_0` enabled. Additionally uncommenting the `MANUAL_[XY]_HOME_POS` – which are both "0" – would have no additional effect in these configurations. We don't want to uncomment them in the other (Cartesian, SCARA) configurations, because uncommenting them also enables them.

mjmeans, 2016-08-10:
Yes. But it's not going to X0 Y0 unless I enable MANUAL_[XY]_HOME_POS. It is moving to the correct Z, but it's also moving toward the X tower which is I think -60,-60. So to restate G28 for delta should move to X0 Y0 Z201.75, but it is instead moving to X-60 Y-60 Z201.75.

For confirmation purposes, my MIN and MAX values are the default ones in the delta config:
```
// Travel limits after homing (units are in mm)
#define X_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Y_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Z_MIN_POS 0
#define X_MAX_POS DELTA_PRINTABLE_RADIUS
#define Y_MAX_POS DELTA_PRINTABLE_RADIUS
#define Z_MAX_POS MANUAL_Z_HOME_POS

```

mjmeans, 2016-08-10:
The home position should be at (0,0) and it should be the center of the bed. Could the problem be because ` #define X_MAX_LENGTH (X_MAX_POS - (X_MIN_POS))` and `#define X_HOME_POS ((X_MAX_LENGTH) * (X_HOME_DIR) * 0.5)` in conditionals_post.h doesn't result in a value of 0?

thinkyhead, 2016-08-11:
@mjmeans Please post your Configuration. All the home-position logic works fine for me.

mjmeans, 2016-08-11:
So during homing, the printer raises to the MAX end stops, then lowers to a Z height where the entire XY axis can be realized. When using the default example_configurations/delta/kossel_mini/Configuration.h you can see that both MANUAL_[XY]_HOME_POS are commented. This doesn't work because when the printer homes it moves off toward the Z tower instead of strait down to reach a Z height where the entire XY range can be reached. So either the example configuration is wrong (and these two should not be ommented out), or something is wrong with the callculation of the movement to a Z height a the end of the homeing procedure.

Here is my configuration.h: [https://github.com/mjmeans/MarlinFirmware-Marlin/tree/RC7-KosselLegacy](https://github.com/mjmeans/MarlinFirmware-Marlin/tree/RC7-KosselLegacy)

zenmetsu, 2016-08-11:
Your Conditionals_post.h is missing the check for the DELTA declaration for both the X_HOME_POS and Y_HOME_POS blocks...

Should be:
`#ifdef MANUAL_X_HOME_POS
    #define X_HOME_POS MANUAL_X_HOME_POS
  #elif ENABLED(BED_CENTER_AT_0_0)
    #if ENABLED(DELTA)
      #define X_HOME_POS 0
    #else
      #define X_HOME_POS ((X_MAX_LENGTH) * (X_HOME_DIR) * 0.5)
    #endif`

You has:
` #ifdef MANUAL_Y_HOME_POS
    #define Y_HOME_POS MANUAL_Y_HOME_POS
  #elif ENABLED(BED_CENTER_AT_0_0)
    #define Y_HOME_POS ((Y_MAX_LENGTH) * (Y_HOME_DIR) * 0.5)
  #else
    #define Y_HOME_POS (Y_HOME_DIR < 0 ? Y_MIN_POS : Y_MAX_POS)
  #endif`

thinkyhead, 2016-08-11:
@mjmeans It looks like you tried to work around the issue and forgot you had modified `Conditionals_post.h`. Does it work properly if you use the unmodified file?

mjmeans, 2016-08-12:
I did not modify that file. I never got that change in the first place. I forked the repo during RC6. 14 days ago I updated by copy of the repo from RC to RC. The last changes were labeled "Changes for 1.1.0-RC7" and "Merge Branch RCBugFix into RC" (both 16 days ago) and 2 more changes labeled "Merge Branch RCBugFix into RC" (both 14 days ago). The final change I got for RC7 had SHA 3c2d8bc034088cd9af6ba67bfdb378bea1d795ad. So it appears that I started with fork of RC as of July 26 (which was the first RC7), then it appears that another RC7 commit was made on July 31, which I didn't get. I've tried to pull the changes but it tell me there are conflict even though I never changed my fork of RC. I then tried to fetch -f and rebase -f but that fails too. I don't know how to repair this and get the correct version of RC7 in my fork RC branch. I don't want to replace my entire repo, because I have made several branches that I used to experiment with that I don't want to lose.

mjmeans, 2016-08-12:
Well, I think I rebuilt my form's RC branch and the RC7 branch I made and the RC7-KosselLegacy. But I am tired now. I lost my change history but I had a copy of my configuration.ini so I so I think the correct delta settings are in now in the correct RC7 version. But it's late and I'll try it tomorrow after I get some sleep. Thanks for the help. I don't know why, but somehow I got an RC7 branch that was dated July 26 and that was the cause of the problems.

mjmeans, 2016-08-12:
That did solve the problem. Thanks.

closed 2016-08-12 by mjmeans