MarlinFirmware/Marlin #4583: Z Probe Homing with Min Switch as Abort

StevenChristy, 2016-08-10:
I couldn't find a way to configure Marlin to what I felt was a sensible setup using both an induction probe and a Z min switch. In my case I expected the z min switch to act as a safety mechanism in case of induction probe failure. Right now my Z min switch right near where the nozzle and the plate meet, but I'd rather the min switch only be used as a backup.

Roxy-3D, 2016-08-10:
You can have both a Z-Probe and a Z-Min End Stop switch.   You need to follow the directions in the Configuration.h file to set that up.

StevenChristy, 2016-08-10:
The main problem with this is that the z-min switch gets used for homing if you use both a min switch and a probe. I want homing to be based on the probe and the z-min switch to be used only as a fail safe. I only had two real choices using the latest RC7 - to use both but the switch will be used when homing (what I don't want) or to use only the probe. I ended up just using the probe for now. I'd really like both a min switch for a safety (stop/abort and lift) and a probe for both homing and leveling.

Roxy-3D, 2016-08-10:
You can set the Z-Min switch 1 or 2 mm below the bed.   And then wire up the Z-Probe and Z-Min in series (with them both Normally Closed).  If something happens where the Z-Probe doesn't trigger, the Z-Min will trigger before much force is exerted on the Z-Probe.

StevenChristy, 2016-08-10:
Originally I was just going to use a probe making this a non-issue, but after the first time the extruder crashed into the plate I thought a safety mechanism might be prudent. Specifically this would help reduce damage to expensive build plates if the extruder was hot forcing it to abort and lift on crash. I imagined that it might save the plate from extensive damage (some damage is probably unavoidable). Implementing it in software seemed a bit more reasonable as it would make it simple to lift the z-axis. For now I suppose I can probably make due without it. Thanks for your suggestion though.

Roxy-3D, 2016-08-10:
Once you get setup and running, I think you will find nozzle crashes don't happen very often.   And personally, if I was that worried about it, I would not want software in the middle of things.   I might wire up some switches that actually cut power to the Z-Stepper motors so they can't step any further down.  In that case, you would need a 'Double Throw' micro switch instead of the normal 'Single Throw' because you need to cut the power to two different phases.   But you get the idea.

landodragon141, 2018-11-08:
@thinkyhead I think that the auto kill for max z travel in the probing routine should cover this.

boelle, 2019-02-18:
@StevenChristy this seems more like a question than a feature request so

Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/ 
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

closed 2019-07-11 by thinkyhead