MarlinFirmware/Marlin #4585: Discussion: Z probe in delta might be able to measure tower height, tile, convex/concave, etc

mjmeans, 2016-08-10:
The known steps per unit of a stepper motor and a Z probe constitutes the equivalent of a micrometer. So it seems that there could be an algorithm that would actually measure the geometry of a delta printer given only the known printable radius of the printer; even if the z tower max end stops are not exactly the same height.

After moving all carriers to the top end stops, move all three down slowly equally. The overall height of the bed can be measured. Then move all three steppers up something like 5% of the known radius. Then move down each tower down one at a time until it touches and move it back. This would mean that the Z probe will not be moving strait down, but actually along a curve. So the normal Allen Z probe would not work for this. Repeat at another value, like 10% of the known radius. And perhaps again at 15%. At each iteration you can calculate an ever finer resolution average tilt of the plane of the bed. Of course we know that the bed is essentially a plane, so any tilt revealed would be due to unequal end stop heights. Once we know that, we can compensate. Then a normal-ish bed levelling process (adjusted for unequal end stops) can measure if the bed is convex or concave. And once that is known, a smooth rod offset can be calculated because, again, we know the bed is essentially a plane. Has this been discussed before?

Roxy-3D, 2016-08-10:
> Of course we know that the bed is essentially a plane, so any tilt revealed would be due to unequal end stop heights.

Or it could be caused by Delta Rod errors.  Or Delta Radius Errors.   Or slop in any of the bearings.  Or Tower radius angles being non-symmetrical.   Or tower normal angles not being normal.    

I'm sure I missed a couple of possibilities.

mjmeans, 2016-08-10:
Roxy-3D: I think that with the errors you mention it will not be possible to even manually calibrate the printer. You have to start with the assumption that the printer is an equilateral triangle and that the verticals are strait lines parallel to each other. If the rod lengths are not the same, that can also be measured by using the stepper motors and the z probe. Even the diameter of the build platform can be measured by the z probe (though that doesn't necessarily equate to the printable radius since the user might not want to print close to the edge).

Roxy-3D, 2016-08-10:
Yes.   I agree.   I have two thoughts.   First, if you (or somebody else) could come up with an algorithm that did some probing and made suggestions on what to change, and the results quickly converged, you would be a hero!   If I can just wish for things, I would want it to automatically converge the Delta Radius and Delta Rod trim numbers.  I would also like it to address the radial angle and normal angle of the towers.    

The second thought is this:   The UBL code was primarily designed to help Delta printers get a perfect first layer across the entire bed.   It is up and running on Cartesian machines but it isn't running on Delta's yet.   I expect it to be able to handle very bad Cup and Bowl problems (caused by Delta Radius and Delta Rod problems).   It won't correct X & Y dimensions that get shifted in the really bad Delta configuration problems.   But it will get us 1/2 to the finish line.      You can't try it on a Delta yet.  But you can get a feel for the system on a Cartesian machine and make recommendations for how it behaves on Delta's.

mjmeans, 2016-08-10:
I don't have a Cartesian. For a delta such as mine (Kossel), the carriage is mounted directly to the verticals, so no pair of rids that could get significantly out of sorts. But directly measuring the DELTA_SMOOTH_ROD_OFFSET is a difficult thing to do on either a Kossel or traditional delta. But at least on a Kossel, it is very easy to calculate this. In a printer that is an equilateral triangle the center is simply the Circumradius of the triangle of the three towers: R = (1/3 * SQRT(3) * a). Where a is the known distance between the centers of each tower. It's easy to measurement of the distance between each two towers at the top and bottom. You get 6 measurements. If they are equal to each other within 0.5mm you know that you have an equilateral triangle and that don't have varying geometries at different heights and you easily calculate this. `#define DELTA_SMOOTH_ROD_OFFSET (1/3 * SQRT(3) * 383)` in my case. Now that I know that they are equal, I know that the DELTA_RADIUS is the hypotenuse of a right triangle. Because I know that, and I know it is a fixed length, all I should need to do is raise all three towers an equal number of steps up and then move two towers down and count the steps. As the pair of steppers are move down, the head is moving in a circular path along the surface of a sphere whose center is the stationary tower carriage pivot. The head stops when it touches the plane of the bed. Move the pair back up to the original point, and move one of the pair of steppers at a different rate. Doing this for all three towers gives us enough informaiton to calculate the intersection of the plane of the bed with all three spheres which I think is enough to calculate all necessary parameters for the delta geometry including correction for convex/concave and unequal end stop heights. The user will still need to specify printable radius they want and the z-probe XYZ offsets to the print head if they are not using FSR's. The only problem I see is that if someone's delta has side walls, the print head may not actually be able to reach the side edge of the build plate without impacting the walls. So the calibration process could be iterative, working forst with small motions for gross values, enough to find good enough approximations to be sure the head is nowhere near the sides, then a wider fine tune calcualtion.

LVD-AC, 2017-05-10:
Writing a routine that calculates the height, end-stops and delta radius by probing the center and a point near each towers is pretty straight forward. This is like manual calibration: probing the center gives you an indication of the overal height,  probing near the towers gives you the end-stop adjustment needed to make the 3 towers equal and the difference in height between the equal towers and the center determines the delta radius.

An accurate calibration routine is highly dependent on accurate and repeatable probing. I do not think that probing the bed with a circular move instead of a vertical move and hitting the bed under a more horizontal angle, scraping the bed instead of pushing the probe down, will give sufficient accurate probing results, at least with probes that physically touch the bed.

mjmeans, 2017-05-10:
The stationary tower could be at a low position so that the other two towers are moving the probe in a nearly vertical manner. Visualize a sphere with the center being the pivot at stationary tower intersecting with the bed. This will intersect in an arc across the bed opposite the stationary tower. Like a crescent moon.

But I do see your point that this method of calculation will not work well on a delta that must contact the bed and has a very sloppy or wobbly probe. But then on such a Kossel you will probably never have a very good calibration. :) On a Kossel that uses a tool steel hardened Allen key, the key may scratch the bed which would be even worse. For my Kossel I use a bent brass rod which for which I smoothed and curved the tip. Of course a ball tip probe would be the best and I may just make one some day.

thinkyhead, 2017-11-01:
See `G33` in the latest Marlin.

closed 2017-11-01 by thinkyhead