MarlinFirmware/Marlin #4591: Compile failure when using 3drag / k8200 board

ntoff, 2016-08-10:
https://gist.github.com/ntoff/f4ef715413ce93ab9984990543ff64e5

pins_3drag appears to need #undef FAN1_PIN added somewhere, probably around here: https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/pins_3DRAG.h#L55 (tested it and it compiles)

thinkyhead, 2016-08-11:
It seems to have already been patched.

ntoff, 2016-08-11:
considering I still get the issue in rcbugfix I don't see how it's been patched unless there's another branch out there?

jbrazio, 2016-08-11:
RC7 as of today ?

ntoff, 2016-08-11:
rc whatever as downloaded seconds before I posted that reply to test. both "rc" and "rcbugfix"

thinkyhead, 2016-08-11:
@ntoff Sorry! I missed the "1" in the pin name. Is it that the 3Drag board has only two power output connectors? The `FAN1_PIN` will assigned to `RAMPS_D8_PIN` as the second fan, but only if you're using 1 hotend and no bed heater. Then it should be -1.

ntoff, 2016-08-11:
3drag \ k8200 only has 1 heat bed, one extruder and one fan pin http://reprap.org/wiki/3Drag_controller

thinkyhead, 2016-08-11:
@ntoff I see. So the solution is…
```cpp
. . .
#define RAMPS_D8_PIN -1

#include "pins_RAMPS.h"
. . .
```

thinkyhead, 2016-08-11:
The changes in #4605 should do the trick.

boelle, 2016-08-14:
So we can close this one right?

closed 2016-08-14 by thinkyhead