MarlinFirmware/Marlin #4592: LCD screen not working.

drluigi92, 2016-08-10:
Hey!

Just try'd to get a new marlin firmware instead of the older one of makerfarm,

Changed the config as it should be for the printer but not alot after sensors and steps,

When uploading the firmware rebooting the printer the LCD stopped working,
When using Prontrface everything does seem to work, except for the LCD.

Here is a picture as a example.

http://puu.sh/qwAiC/d90fe04f83.jpg

brainscan, 2016-08-10:
Changed the config as it should be for the printer but not alot after sensors and steps

Did you define an lcd or not?

http://puu.sh/qwAiC/d90fe04f83.jpg

-

WheresWaldo, 2016-08-11:
It used to be that the small LCD controller was enabled by default, I don't recall that is the case on any version of 1.1.0. Maker sure this line is uncommented in configuration.h
`#define REPRAPWORLD_GRAPHICAL_LCD`

drluigi92, 2016-08-11:
reprapworld Graphical isnt that the larger one?

WheresWaldo, 2016-08-11:
No the FULL Graphics is the larger one, you will see that entry lower down in the file.

drluigi92, 2016-08-11:
Just tried that,

I now get an error:

`sketch\ultralcd.cpp: In function 'void lcd_quick_feedback()':

ultralcd.cpp:2338: error: 'LCD_FEEDBACK_FREQUENCY_HZ' was not declared in this scope

     lcd_buzz(LCD_FEEDBACK_FREQUENCY_DURATION_MS, LCD_FEEDBACK_FREQUENCY_HZ);

                                                  ^

exit status 1
'LCD_FEEDBACK_FREQUENCY_HZ' was not declared in this scope`

thinkyhead, 2016-08-11:
I noticed this the other day. the LCD_FEEDBACK defines are not always set. I will patch it shortly.

thinkyhead, 2016-08-11:
This seems to be patched in `RCBugFix` with this block in `Conditionals_post.h`:
```cpp
  /**
   * Buzzer/Speaker
   */
  #if ENABLED(LCD_USE_I2C_BUZZER)
    #ifndef LCD_FEEDBACK_FREQUENCY_HZ
      #define LCD_FEEDBACK_FREQUENCY_HZ 1000
    #endif
    #ifndef LCD_FEEDBACK_FREQUENCY_DURATION_MS
      #define LCD_FEEDBACK_FREQUENCY_DURATION_MS 100
    #endif
  #else
    #ifndef LCD_FEEDBACK_FREQUENCY_HZ
      #define LCD_FEEDBACK_FREQUENCY_HZ 5000
    #endif
    #ifndef LCD_FEEDBACK_FREQUENCY_DURATION_MS
      #define LCD_FEEDBACK_FREQUENCY_DURATION_MS 2
    #endif
  #endif
```

drluigi92, 2016-08-12:
Thanks error is gone, screen still not working tho :/ still the same blocks.

WheresWaldo, 2016-08-12:
It could be physical, is the a potentiometer to control contrast, if it's too high you will see blocks instead of letters.

drluigi92, 2016-08-12:
Doubt it, as my old firmware also marlin but from 2013-2014 does work fantastic, no problems with the screen what so ever, have it on atm.

zenmetsu, 2016-08-13:
OK, I just ran into some stupidity on my behalf.  I was able to get character display to work in `RC7` as well as `RCBugFix`.

My issue was that another LCD was uncommented, specifically `#define PANEL_ONE` from the Kossel Mini example config.  I commented that line and uncommented `#define REPRAPWORLD_GRAPHICAL_LCD` per @WheresWaldo and still had no display.  I then commented `#define REPRAPWORLD_GRAPHICAL_LCD` again, and the display worked fine.

You need to have `#define ULTRA_LCD` uncommented, but every display panel beneath that line needs to be commented...

drluigi92, 2016-08-13:
That will probably be it, will test it in a min

drluigi92, 2016-08-13:
I only have 2 lines atm,
Cant goto my menu neither
just says bed and nozzle temp and that its ready..

As far as easy to use, it didnt improve alot did it ? :P

zenmetsu, 2016-08-13:
Which LCD panel do you have?  Some of the 4-line displays have two Enable lines... one for odd numbered lines and one for even numbered lines.  I have one like that and I cannot use it without modifying the code.

drluigi92, 2016-08-13:
Hey,

Well i have this one like on the picture at the start of the topic:

http://www.makerfarm.com/media/catalog/product/cache/1/image/e50596b62d4a48a983f848d3568a5b89/p/h/photo_2_1.jpg

Blue-Marlin, 2016-08-14:
Great news. Behind a cover all 20x4 displays look about the same.
My first guess would be `REPRAP_DISCOUNT_SMART_CONTROLLER` that is the most common one.

thinkyhead, 2016-08-14:
I concur. It's most likely `REPRAP_DISCOUNT_SMART_CONTROLLER` or `G3D_PANEL` which also looks very similar.

drluigi92, 2016-08-14:
So i have to comment ultra LCD again and try reprap discount or g3d panel now?

zenmetsu, 2016-08-14:
You don't have to comment it.  For some reason I thought that you had a 2-line display when I said to uncomment it.

What he's saying is that whether or not `ULTRA_LCD` is commented, enabling `REPRAP_DISCOUNT_SMART_CONTROLLER` will automatically enable `ULTRA_LCD` within `Conditionals_LCD.h`

```
  #if ENABLED(ULTIMAKERCONTROLLER)              \
   || ENABLED(REPRAP_DISCOUNT_SMART_CONTROLLER) \  
   || ENABLED(G3D_PANEL)                        \
   || ENABLED(RIGIDBOT_PANEL)                   \
   || ENABLED(REPRAPWORLD_KEYPAD)
    #define ULTIPANEL
    #define NEWPANEL
  #endif
```

This will set `ULTIPANEL` and `NEWPANEL` if any of those 5 defines are made, and then since `ULTIPANEL` is enabled, the second block within that file will do additional setup of your LCD, including setting up the width and height to give you 4 rows of text.

```
  #if ENABLED(ULTIPANEL)
    #define NEWPANEL  //enable this if you have a click-encoder panel
    #define ULTRA_LCD
    #ifndef LCD_WIDTH
      #define LCD_WIDTH 20
    #endif
    #ifndef LCD_HEIGHT
      #define LCD_HEIGHT 4
    #endif
  #else //no panel but just LCD
    #if ENABLED(ULTRA_LCD)
      #ifndef LCD_WIDTH
        #define LCD_WIDTH 16
      #endif
      #ifndef LCD_HEIGHT
        #define LCD_HEIGHT 2
      #endif
    #endif
  #endif
```

drluigi92, 2016-08-14:
The discount reprap thing seems to work,

Thanks to everybody,
I bet this topic will be handy for a ton of other users that didnt seem to find it, as it used to not be necassary.

best regs
Simon

closed 2016-08-15 by thinkyhead