MarlinFirmware/Marlin #4593: Question about homing and ABL

apballard, 2016-08-10:
Hi

This most probably a silly question, but I can't seem to figure it out.

I callebrate the z-axis by homing it then adjusting the offset so that when I make z=0 a piece of paper can barely fit under.

If after that I do an ABL, it's always too close and my nozzle scrapes the base plate.  It's roughly about 0.3mm out.

So I've determined for it to work I need to do ABL then set my offset from that. Then my first layer is perfect.

Does this make sense? Am I doing things correctly?

zenmetsu, 2016-08-11:
If your nozzle is 0.3mm too close to the platform after doing ABL, it could be because your probe offset is incorrect.  If you do an M501, you should see a listing for the M851 value.  Assuming that it is something like -5.0mm, you can add 0.3mm (your error) to this value and issue a M851 Z-4.7   and then save it with M500 before doing another G28 + G29 and seeing if the issue is corrected.

If not, you could alternatively do a G1 Z10 followed by a G92 Z9.7 to tell the printer that it's really at Z=9.7 instead of Z=10.0.

Roxy-3D, 2016-08-11:
As zenmetsu hinted...  please be aware that if your EEPROM is turned on, you will be getting values from it instead of your Configuration.h file.   If you make changes to the values in your Configuration.h file, do a M502 followed by a M500 to get everything synchronized.   Otherwise, you will be pulling your hair out and not understanding why things are not as you said they are.

zenmetsu, 2016-08-11:
My apology for overlooking that.  I shouldn't have assumed that the user had EEPROM support enabled.  

I did mention using G92 because I run into problems within Simplify3D.  After doing a G28/G29 as my beginning-of-print scripts, the nozzle is about 1.5mm above the bed.  I cannot solve this with M851 changes as they appear to have no effect... If I do a G28 after the G29, then going to Z0 results in a correct placement of the nozzle, but I fear that G28 also wipes out the bed level matrix from the prior G29...  So instead, I just issue a G1 Z10, then G92 Z11.5 to correct the issue.  Incidentally, this problem doesn't bite me within Pronterface...

thinkyhead, 2016-08-11:
G92 can't be used this way any more. It moves the software endstops. For example, if you are at Z=0 and you issue `G92 Z10` thinking you can now move below the bed, you will be mistaken. The new lower limit will now be 10.

zenmetsu, 2016-08-12:
unless I disable software endstops... which I have done until I can figure out what the heck is wrong with my Simplify3D setup.

apballard, 2016-08-12:
I can get the offset set correctly, and I can print fine. Thanks for all your help.

I was just asking if it's normal for the offsets between a G28 only and a G28/G29 to be different.

Is it standard practice when calibrating the z-offset to set the offset only after a G29?

In my mind I always imagined that after setting the offset correctly after a G28, the G29 would use that offset and simply provide leveling "sub-offsets" across the build plate.  I did not think that the offset would need to be set after a G29.

bgort, 2016-08-13:
I'm having the same issue:  the `M851` Z offset necessary when using `G28` alone is different than the offset necessary when using `G28`->`G29`.  I don't believe this is (or should be) 'normal' behavior.  The Z probe offset should be the Z probe offset; believe the nozzle height should be the same at the same Z coordinate before and after G29.

FYI, this is being discussed in #4577, also.

thinkyhead, 2016-08-16:
You are correct that the offset should apply in the same way in both cases, and that there is a bug.

apballard, 2016-08-16:
Thanks for all the replies!

Will wait for a bug fix.

closed 2016-08-16 by apballard

thinkyhead, 2016-08-16:
I could use some helpful replies too!