MarlinFirmware/Marlin #4596: Z-probe will not store in 1.1.0RC7

zenmetsu, 2016-08-11:
I am recently migrating to Marlin from Repetier for my RepRap mini Rostock.  I tried the 1.1.0RC7 branch first, and got everything working well.  I attempted to add a Z-probe (allen key type) and got as far as having G29 properly deploy the probe and probe the bed.  

The problem is that after probing, when the printer is told to go to the second "stow" position which "pushes down" the probe, the movement stops as soon as the switch is opened, and the printer moves on to the third "stow" position as stored in firmware.  It doesn't matter what I use for the Z position in the second stow position... in my case it needs to be -2 as manual G1 Z-2 Z0 Y-98 will stow the probe reliably every time.  I've tried setting it to ridiculous values such as Z-500 and the code just aborts the move the instant the switch on the probe is opened. 

I am using the stock Kossel Mini as a template and just adjusting my delta values, my probe offset, and enabling auto-leveling... Configs can be provided if requested.  The thing to note is that when I moved to the RCBugFix branch, everything worked exactly as expected.  I suspect that recent changes to the RC7 code broke the stow probe functionality.  If someone with a Z-probe could perform some regression testing on RC7 and validate this issue, it would restore some faith in my sanity.

Blue-Marlin, 2016-08-11:
Fixed by #4551 in RCBugFix.

zenmetsu, 2016-08-11:
Go ahead and close this issue.  I had a misunderstanding regarding the RC7 vs. RCBugFix repos.  Thanks.

closed 2016-08-11 by zenmetsu