MarlinFirmware/Marlin #4598: Cancel after pause doesn't work

spilz87, 2016-08-11:
Hi,
I don't know if it's an issue, or if you want it's like this, but :
When I pause the print for any technical reason, and finally I have to cancel the print, I can't directly cancel, I have to "play" again the print to cancel.
When it's for technical issue, it can be a problem to continue the print and have new movements.

Regards

thinkyhead, 2016-08-11:
Are you talking about the LCD menus? As far as I can see, the "Stop print" command is always available when SD printing.

spilz87, 2016-08-12:
Hello,
Yes I'm talking about led menu, when SD printing, if I pause first, and then stop, the menu still shows pause and cancel line, the "print from sd card" doesn't appear. 
If I stop without pause or after replay after pause, the menu "print from sd card" appear

spilz87, 2016-08-12:
Led -> read "LCD"

thinkyhead, 2016-08-12:
Ah… verrrry interesting. There are a couple of flags involved in the state of an SD print. Likely one of them is failing to be reset. I'll poke around and see what's going on.

thinkyhead, 2016-08-12:
#4610 should clear this issue. Thanks for bringing it to our attention!

closed 2016-08-12 by thinkyhead