MarlinFirmware/Marlin #4599: Linear Advance extrusion breaks Fan PWM on PIN 4

Kaibob2, 2016-08-11:
When #define LIN_ADVANCE in Configuration_adv.h is uncommented the fan on PIN 4 looses the PWM capability.
In other words: M106 S255 switches the fan on, M106 S254 - M106 S0 switches it off.
#define FAN_KICKSTART_TIME 100 still works as always.
When commenting #define LIN_ADVANCE the fan PWM works perfect again.

PS: The same happens when uncommenting #define ADVANCE. But this was only for testing as i read that it isn't working properly. The prints with LIN_ADVANCE enabled looked really promising yesterday, but PLA without a PWM fan isn't an option.

Blue-Marlin, 2016-08-11:
There is no simple way to change that. Both use timer 0.
Better take an other pin.

thinkyhead, 2016-08-11:
@Blue-Marlin Looks like something we might need to account-for. Perhaps in `Conditionals.h`…

```cpp
#if (HAS_FAN0 || HAS_FAN1) && ENABLED(LIN_ADVANCE)
  #define FAN_SOFT_PWM
#endif
```

Blue-Marlin, 2016-08-12:
At first lets wait for @Kaibob2 s answer - if FAN_SOFT_PWM helps.

HAS_FAN0 or HAS_FAN1 is wrong. It depends on the pin number and the type of the processor, and if analogWrite() is used to get a PWM.

I looked it up for the ATmega2560-Arduino Mega in `pins_arduino.h`. There you can find an array `digital_pin_to_timer_PGM[]`. Therein you can find the timers related to the pin numbers.
```
const uint8_t PROGMEM digital_pin_to_timer_PGM[] = {
	// TIMERS		
	// -------------------------------------------		
	NOT_ON_TIMER	, // PE 0 ** 0 ** USART0_RX	
	NOT_ON_TIMER	, // PE 1 ** 1 ** USART0_TX	
	TIMER3B	, // PE 4 ** 2 ** PWM2	
	TIMER3C	, // PE 5 ** 3 ** PWM3	
	TIMER0B	, // PG 5 ** 4 ** PWM4	
	TIMER3A	, // PE 3 ** 5 ** PWM5	
	TIMER4A	, // PH 3 ** 6 ** PWM6	
	TIMER4B	, // PH 4 ** 7 ** PWM7	
	TIMER4C	, // PH 5 ** 8 ** PWM8	
	TIMER2B	, // PH 6 ** 9 ** PWM9	
	TIMER2A	, // PB 4 ** 10 ** PWM10	
	TIMER1A	, // PB 5 ** 11 ** PWM11	
	TIMER1B	, // PB 6 ** 12 ** PWM12	
	TIMER0A	, // PB 7 ** 13 ** PWM13	
	NOT_ON_TIMER	, // PJ 1 ** 14 ** USART3_TX	
	NOT_ON_TIMER	, // PJ 0 ** 15 ** USART3_RX	
	NOT_ON_TIMER	, // PH 1 ** 16 ** USART2_TX	
	NOT_ON_TIMER	, // PH 0 ** 17 ** USART2_RX	
	NOT_ON_TIMER	, // PD 3 ** 18 ** USART1_TX	
...
```
```
	TIMER0B	, // PG 5 ** 4 ** PWM4	
	TIMER0A	, // PB 7 ** 13 ** PWM13	
```
are  relevant for timer 0.

But also
```
	TIMER1A	, // PB 5 ** 11 ** PWM11	
	TIMER1B	, // PB 6 ** 12 ** PWM12	
```
can not be used for hardware PWM because of the stepper interrupt at timer 1.

For other processors/boards the register names and pin numbers may be different.

Kaibob2, 2016-08-12:
Enabling FAN_SOFT_PWM works.
Does this have any negative impact on something else like the extruder heaters, bed or MOSFET heating?

Kaibob2, 2016-08-12:
I built an 4 channel extension PCB for the servo port which uses PIN 4,5,6 and 11.
PIN 4: Filament cooling fan (controlled via S3D Gcode M106)
*works*
PIN 5: Controller Fan (controlled via Marlin CONTROLLERFAN_PIN 5 / CONTROLLERFAN_SPEED 200)
*works*
PIN 6: Extruder cooling fans (controlled via Marlin EXTRUDER_0_AUTO_FAN_PIN 6 / EXTRUDER_AUTO_FAN_SPEED   200)
*works*
PIN 11: LEDs for building room (Manual control via M42 P11 Sxxx )
No PWM. Only on and off

**Here is the question:** PIN 11 is not PWMable. Why?

Blue-Marlin, 2016-08-12:
To cite myself:
> But also
```
	TIMER1A	, // PB 5 ** 11 ** PWM11	
	TIMER1B	, // PB 6 ** 12 ** PWM12	
```
can not be used for hardware PWM because of the stepper interrupt at timer 1.

Blue-Marlin, 2016-08-12:
> // Use software PWM to drive the fan, as for the heaters. This uses a very low frequency
// which is not as annoying as with the hardware PWM. On the other hand, if this frequency
// is too low, you should also increment SOFT_PWM_SCALE.
//#define FAN_SOFT_PWM

> // Incrementing this by 1 will double the software PWM frequency,
// affecting heaters, and the fan if FAN_SOFT_PWM is enabled.
// However, control resolution will be halved for each increment;
// at zero value, there are 128 effective control positions.
#define SOFT_PWM_SCALE 0

I don// Use software PWM to drive the fan, as for the heaters. This uses a very low frequency
// which is not as annoying as with the hardware PWM. On the other hand, if this frequency
// is too low, you should also increment SOFT_PWM_SCALE.
//#define FAN_SOFT_PWM

// Incrementing this by 1 will double the software PWM frequency,
// affecting heaters, and the fan if FAN_SOFT_PWM is enabled.
// However, control resolution will be halved for each increment;
// at zero value, there are 128 effective control positions.
#define SOFT_PWM_SCALE 0

I don't expect more negative effects.

Kaibob2, 2016-08-12:
Sorry for that. In Germany we say (free translated) "The ones who can read have an advantage". 

Does this mean, with "FAN_SOFT_PWM" enabled and adding "#define FAN1_PIN 11" in "pins_RAMPS.h", PWM on Pin 11 would work?

Blue-Marlin, 2016-08-12:
Never tried it. But i expect it to work.
In the time i write this, you could simply have mad a test at your own.

Kaibob2, 2016-08-12:
Sure, sorry to bug you with this.

Kaibob2, 2016-08-13:
Enabling Soft PWM works.

closed 2016-08-13 by Kaibob2