MarlinFirmware/Marlin #4601: Feature request: Add ability to drive servo in place of PWM fan

zenmetsu, 2016-08-11:
In trying to drive down the weight of my effector, I am attempting to use a single fan for cooling the coolend as well as the extruded plastic.  Funny thing about fans is that they cannot spin at two speeds at once.

The coolend fan needs to run all the time to prevent jamming, but the main cooling fan needs to be turned off when printing the first layer or two, and it needs to be going full speed while printing bridges.  This is my dilemma.  

I decided that I could put a single fan blowing across the coolend, and then use a servo to open/close some vanes which would either allow the air to pass out the back without going to the nozzle area, or when closed, direct air downwards to the nozzle.  

Granted, this is a corner case which might not have much demand, but who knows, someone might be crazy enough to run a fan from an R/C nitro motor and need the servo for the throttle.

hoffstadt1004, 2016-08-12:
Sounds good

thinkyhead, 2016-08-22:
@zenmetsu For niche features like this, it is preferable that we leave it to users to figure it out for themselves or find help in their local community. We simply cannot implement every request. If you cannot find a developer in your local area to help you, myself and others are happy to offer services for a nominal fee.

zenmetsu, 2016-08-22:
No biggie.  I can close this.  I will add the feature and then see if there is merit in incorporating it into the mainline code.  I've identified a few other uses for such control, such as opening vents in the enclosure, as well as milling/laser uses. I actually found a way to accomplish what I needed without using a servo... I just rely on air pressure from the fan once it spins up past 30% and I set the minimum speed of the fan to 20% in my host software.

closed 2016-08-22 by zenmetsu