MarlinFirmware/Marlin #4604: Free Memory Watch M100 only works one time until a reset is issued.  [RCBugFix]

zenmetsu, 2016-08-11:
Upon enabling and issuing an `M100`, the expected stack is dumped and the report is made as expected.  If one attempts to issue another `M100`, no output is generated.

Blue-Marlin, 2016-08-11:
```
/**
 * M100 Free Memory Watcher
 * 
 * This code watches the free memory block between the bottom of the heap and the top of the stack.
 * This memory block is initialized and watched via the M100 command.
 * 
 * M100 I Initializes the free memory block and prints vitals statistics about the area
 * M100 F Identifies how much of the free memory block remains free and unused.  It also
 *    detects and reports any corruption within the free memory block that may have
 *    happened due to errant firmware.
 * M100 D Does a hex display of the free memory block along with a flag for any errant
 *    data that does not match the expected value.
 * M100 C x Corrupts x locations within the free memory block.   This is useful to check the
 *    correctness of the M100 F and M100 D commands.
 * 
 * Initial version by Roxy-3DPrintBoard
 */
```
For the subsequent call you need `M100 F`.

Roxy-3D, 2016-08-11:
The command is smart enough to know the first time through that it needs to be initialized.  So with no parameters, it will 'Initialize'.   But then, you need to tell it what to do.  So the second command is often M100 F

thinkyhead, 2016-08-11:
@MarlinFirmware/documentation-maintainers

zenmetsu, 2016-08-11:
Thanks.  I was just looking at `M100_Free_Mem_Chk.cpp` and was about to close this.  Great addition, by the way.  Thanks Roxy.

esenapaj, 2016-08-12:
By the way when I executed `m100` commands from Repetier-Host, it looks like that results strings of `m100 i`, `m100 d`, and `m100 c` were broken.

<details>
<summary>log</summary>
```
23:29:33.453 : N23 M100 I*122
23:29:33.459 : Initializing free memory block.
23:29:33.538 : bss_end : ??????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????
23:29:33.538 : Stack Pointer :
23:29:33.539 : 2214 bytes of memory initialized.
23:29:40.996 : N26 M100 F0*64
23:29:41.002 : Found 2219 bytes free at 0x1811
23:29:52.976 : N31 M100 D*116
23:29:52.979 : bss_end : 1811
23:29:52.983 : Stack Pointer : 2102
23:29:52.983 : 1810:00 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |?
23:29:52.987 : 1820:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:52.991 : 1830:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:52.991 : 1840:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:52.995 : 1850:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.000 : 1860:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.000 : 1870:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.004 : 1880:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.008 : 1890:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.008 : 18A0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.012 : 18B0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.016 : 18C0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.020 : 18D0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.020 : 18E0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.024 : 18F0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.028 : 1900:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.028 : 1910:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.032 : 1920:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.036 : 1930:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.040 : 1940:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.040 : 1950:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.045 : 1960:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.049 : 1970:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.049 : 1980:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.053 : 1990:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.057 : 19A0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.057 : 19B0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.061 : 19C0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.065 : 19D0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.065 : 19E0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.069 : 19F0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.073 : 1A00:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.073 : 1A10:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.077 : 1A20:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.081 : 1A30:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.086 : 1A40:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.086 : 1A50:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.089 : 1A60:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.094 : 1A70:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.094 : 1A80:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.098 : 1A90:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.102 : 1AA0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
23:29:53.106 : 1AB0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
[... cut ...] ?????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????
23:30:05.252 : Stack Pointer : ?\!??!?
```
</details>

Roxy-3D, 2016-08-12:
I just verified M100 is not currently working.   One problem is all of the malloc()'s and free()'s have been eliminated from the rest of the Marlin code base.    The way the C run time library works is __brkval is pre-initialized to NULL.   That is used by the malloc() routine as a flag that it has to go find a more appropriate value for __brkval (which will be after the global variables).

I'm getting the fixes in place right now.

Roxy-3D, 2016-08-12:
I have M100 fixed in the devel_ubl branch.  If you are in a hurry, you can grab M100 and the hex_prt_routines from that branch and use them in RCBugFix branch.

zenmetsu, 2016-08-12:
Sweet.  I'll give it a test.

Roxy-3D, 2016-08-12:
You will also need to declare   bool code_has_value(); somewhere within the M100 scope.   But if you know C, you will have it running in no time.

zenmetsu, 2016-08-12:
No problem.  I'll get to it as soon as I figure out why M48 is giving me standard deviation of 0.000000 all the time.

Roxy-3D, 2016-08-12:
Geeeze???   Another one of my children.    Can you start another thread dedicated to M48 misbehaviour and I'll figure it out.   But it is working just fine in the devel_ubl branch!

zenmetsu, 2016-08-12:
I think it is just my probe and printer's accuracy.  I got a couple of 0.010130 or something... I then slid a piece of paper under the probe for a few of the measurements and it threw the value way off... so I guess it is working fine... but the resolution might be a bit low if I got the same values repeatedly.

Roxy-3D, 2016-08-12:
If you are getting really low Standard Deviation numbers, that is good!   If you can get numbers less than .0005 you are doing really well!

boelle, 2016-08-14:
@zenmetsu did you give this a go?

zenmetsu, 2016-08-16:
Yeah, I couldn't get it to produce output.  It is most likely due to my lousy C++ "skills".  :(  Hell, I only do this stuff to learn.

Anyway, I grabbed the `hex_print_routines.cpp` and tossed it into the mix, along with the updated M100.  I see the forward declarations for the hex print functions, but I don't think that the functions as defined in hex_print_routines.cpp are getting used, thus the lack of output.

zenmetsu, 2016-08-16:
Correction, I are a dumbass.  I forgot that I had disabled the M100 in Configuration.h to troubleshoot another problem I had.  Uncommenting led to compilation errors.  I wasn't sure how to properly deal with  adding code_has_value() since the forward declaration was in Marlin.h which was sourced by M100, so I just slapped the code from `Marlin_main.cpp` into the `M100_Free_Mem_Chk.cpp` file, in addition to adding `static char* seen_pointer;`   If this is the totally wrong way of handling the issue, then let me know since I really want to learn to not suck in C++  :)

ANYWAY...

The output produced some interesting values which may or may not be intended.  I verified that issuing `M100` by itself did indeed initialize as expected.  Then `M100 D` output ended with the following:

```
1F10:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
1F20:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
1F30:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |               E1 00 00 3E 00 00 |????????????????
Done.
```

Not sure that those 6 bytes, and the corresponding `?` are supposed to represent, but I suspect that there should at a minimum be a linefeed beforehand.  :)

Also, when attempting to corrupt the space, I don't think it is working since I see no change after the first corruption attempt...

```
>>> M100 F
SENDING:M100 F
Found 4583 bytes free at 0x0EEA
Done.
>>> M100 C 1
SENDING:M100 C 1
Corrupting free memory block.
__brkval : 3818
Stack Pointer : 8532
Corrupting address: 0x1803
>>> M100 F
SENDING:M100 F
Found 2328 bytes free at 0x0EEA
Found 2253 bytes free at 0x1804
Memory Corruption detected in free memory area.
Done.
>>> M100 C 1
SENDING:M100 C 1
Corrupting free memory block.
__brkval : 3818
Stack Pointer : 8532
Corrupting address: 0x1803
>>> M100 F
SENDING:M100 F
Found 2328 bytes free at 0x0EEA
Found 2253 bytes free at 0x1804
Memory Corruption detected in free memory area.
Done.
>>> M100 C 2
SENDING:M100 C 2
Corrupting free memory block.
__brkval : 3818
Stack Pointer : 8532
Corrupting address: 0x1803
>>> M100 F
SENDING:M100 F
Found 2328 bytes free at 0x0EEA
Found 2253 bytes free at 0x1804
Memory Corruption detected in free memory area.
Done.
```

`M100 C x` appears to only corrupt a single byte, regardless of value.

Roxy-3D, 2016-08-16:
The function code_has_value() is used to see if there is a number specified for the C (corrupt) feature to use.   If code_has_value() is false, a 1 is assumed.  

The 'correct' way to deal with that out of scope function is put this declaration at the top of M100.cpp    Do not try to put the actual function in M100.cpp because it is reaching into the serial buffer and won't have access to all of the things it needs.   Just put:

~~~~
bool code_has_value();
~~~~ 

towards the top of the M100.cpp file and the compiler will know how to call the function and the linker will find it where ever it is hiding.

zenmetsu, 2016-08-16:
I tried that, and received an undefined reference during linking :/

Roxy-3D, 2016-08-16:
Oooops!  Sorry!   You need to remove any inline and/or static declarations to use a function in another file.   I didn't even think to tell you that!  

If you add sufficiently large delay();  in your code, the columns should line up nicely.

You would declare:
~~~~
void delay(unsigned long);
void idle();
~~~~
(or what ever the 'correct' declaration is) towards the top of your M100.cpp file.   With those declarations, the compiler will know how to call them, and the linker will find them.

zenmetsu, 2016-08-16:
I appreciate the help.  I plan on going back to school for programming, but everything seems to be Java now, and I'm on a 24/7 electronic leash from my employer...  :(  I tend to be able to figure things out eventually... i just have to be persistent.

closed 2016-08-19 by Roxy-3D