MarlinFirmware/Marlin #4613: M100 Clean Up

Roxy-3D, 2016-08-12:
@thinkyhead @jbrazio    I fixed the problems in M100.   I was trying to update devel_ubl but I forgot to specify the branch I was changing.   I zapped RC's M100.cpp file.   The code is correct and everything compiles clean.   

But RC should not have been changed and the changes I made to RC really belong in RCBugFix.

I do have devel_ubl updated to be correct.    Nothing needs to be done there.   But RC needs to have the commits I did reverted and those commits probably should be applied against RCBugfix to get M100 working again.

SORRY!!!!

mjmeans, 2016-08-13:
This begs the ancillary question of whether a new branch should be created when an RC is completed. It would set a clear date for the branch and make it easier for people to 'go back' to a default RC7 without having to do git acrobatics in their own repo forks. I had a problem with RC7 when I started my testing on an RC7 'release' on July 28, when not all the RC7 RCbugfixes and readme changes had been applied. At the point where RC7 was 'complete' on August 3 or whatever, I think it would have been better to have had an RC7 branch created.

Roxy-3D, 2016-08-13:
Yes, but each Release Candidate is significantly better than the previous one.  It would add to the confusion with people wondering which branch they should grab.   We already have enough confusion with the RC vs. RCBugFix.   It does make sense to flag the start of the current RCBugFix branch.  But really, most people should be using RCBugFix.   

The good news is soon, these issues are going away.   Probably at the end of RC8's life, it will turn into a Stable Branch.

thinkyhead, 2016-08-14:
> whether a new branch should be created when an RC is completed

No, we'd rather not make a branch. I only added some commits and moved the RC7 tag on July 31 because there were some bad behaviors that I didn't want propagating, but otherwise it is meant to be left alone.

The latest news is that RC8 will be released shortly. We are pretty clear about making sure users test with `RCBugFix` before reporting bugs, in case we already caught them. But, again, we only want `RCBugFix` to be changing, and not the tagged pre-releases.

thinkyhead, 2016-08-14:
@Roxy-3D I removed your commits from RC. Thanks for pointing it out!

closed 2016-08-14 by thinkyhead