MarlinFirmware/Marlin #4614: Compilation error with #define U8GLIB_SSD1306

esenapaj, 2016-08-12:
When `#define U8GLIB_SSD1306` is enabled with newest RCBugFix, it occurs compilation error.

<details>
<summary>error message</summary>
```
AppData\Local\Temp\build6674025c445f92d0b07185e4e4080804.tmp\sketch\ultralcd.cpp: In function 'void lcd_update()':

ultralcd.cpp:2699: error: 'currentScreen' was not declared in this scope

           (*currentScreen)();

             ^
AppData\Local\Temp\build6674025c445f92d0b07185e4e4080804.tmp\sketch\ultralcd.cpp:426:13: warning: 'void lcd_status_screen()' defined but not used [-Wunused-function]

 static void lcd_status_screen() {

             ^

exit status 1
'currentScreen' was not declared in this scope
```
</details>

thinkyhead, 2016-08-14:
I think this might be the right way to fix it…
```diff
#if ENABLED(DOGLCD)  // Changes due to different driver architecture of the DOGM display
  static int8_t dot_color = 0;
  dot_color = 1 - dot_color;
  u8g.firstPage();
  do {
    lcd_setFont(FONT_MENU);
    u8g.setPrintPos(125, 0);
    u8g.setColorIndex(dot_color); // Set color for the alive dot
    u8g.drawPixel(127, 63); // draw alive dot
    u8g.setColorIndex(1); // black on white
-   (*currentScreen)();
+   #if ENABLED(ULTIPANEL)
+     (*currentScreen)();
+   #else
+     lcd_status_screen();
+   #endif
  } while (u8g.nextPage());
#elif ENABLED(ULTIPANEL)
  (*currentScreen)();
#else
  lcd_status_screen();
#endif
```

esenapaj, 2016-08-14:
I got new compilation warnings.
<details>
<summary>new warning message</summary> 
```
AppData\Local\Temp\build6674025c445f92d0b07185e4e4080804.tmp\sketch\ultralcd_impl_DOGM.h:513:13: warning: 'lcd_implementation_drawmenu_generic' defined but not used [-Wunused-function]

 static void lcd_implementation_drawmenu_generic(bool isSelected, uint8_t row, const char* pstr, char pre_char, char post_char) {

             ^

AppData\Local\Temp\build6674025c445f92d0b07185e4e4080804.tmp\sketch\ultralcd_impl_DOGM.h:531:13: warning: '_drawmenu_setting_edit_generic' defined but not used [-Wunused-function]

 static void _drawmenu_setting_edit_generic(bool isSelected, uint8_t row, const char* pstr, const char* data, bool pgm) {

             ^
```
</details>

I'm not sure that these are problem or not, but compilation error and warning in my previous comment is gone.
Thanks for the fixing.

thinkyhead, 2016-08-14:
It looks like they just aren't used if `ULTIPANEL` isn't set. So it's another patch I need to make.

zenmetsu, 2016-08-14:
XD  @thinkyhead i'd kiss ya if I could.  I am working on adding my SSD1351 SPI OLED to the code and ran right into this issue.  Your suggested code change came just seconds after I needed it.  It works brilliantly.

boelle, 2016-08-14:
@esenapaj does it now compile without error?

esenapaj, 2016-08-14:
Yes it is, compilation error was fixed by commit 132322e4d94cf776ea3d42af68406bf7b012120a.
But new compilation warning was occured, and it looks like that @thinkyhead work in preparation for another patch,
therefore I reopened this Issue.

boelle, 2016-08-16:
maybe close this one and open a new on with the new warning?

commit 132322e4 (2016-08-14): Fix #4614: currentScreen only defined with ULTIPANEL
closed 2016-08-17 by esenapaj