MarlinFirmware/Marlin #4615: Strange move after G29

zenmetsu, 2016-08-13:
When I execute the probe reliability test with M48, the printer moves to the probe exployment position, deploys the probe, then probes the bed at X0 Y0 before stowing the probe and returning to X0 Y0 Z10 which is the fourth position defined in the firmware for stowage.

When I execute G29, the printer performs the above listed steps, and pauses for about 50ms at X0 Y0 Z10 before moving on to X0 Y32 Z10.  I'm trying to find the source of this move as it is causing me some problems. I've got video of the behavior and I can post my configuration on my repo if need be.

Roxy-3D, 2016-08-13:
Something else is going on.   What are the exact commands you are giving M48 ?   The reason I'm asking is M48 decides where to probe based on this logic:
~~~~
    float  X_current = current_position[X_AXIS], Y_current = current_position[Y_AXIS];
    float X_probe_location = code_seen('X') ? code_value_axis_units(X_AXIS) : X_current + X_PROBE_OFFSET_FROM_EXTRUDER;
    float Y_probe_location = code_seen('Y') ? code_value_axis_units(Y_AXIS) : Y_current + Y_PROBE_OFFSET_FROM_EXTRUDER;
~~~~
It should either probe where it is currently located, or it should probe the (X,Y) location you specify.

zenmetsu, 2016-08-13:
I am just issuing an M48.

If the probe is deployed, it moves down and probes the bed.  I usually issue it from home position or X0 Y0 Z10.  If the probe is not deployed, as is usually the case, it moves to X56 Y86 Z100, swipes across the Z-tower to X48 Y86 Z100, then moves to X0 Y0 Z100 which is the 3rd and final point of my deploy coordinates.  Thus if M48 issued without a coordinate just tests the current X,Y, then yes, it would probe X0 Y0 because that is where it is after the probe deploys.

Likewise, the four positions for my probe stow routine are
X0 Y-96 Z10
X0 Y-96 Z-2  ; pushes probe up and locks it
X0 Y-96 Z10 ; moves up to allow the nozzle to clear the bed
X0 Y0 Z10 ; return to origin

Those movements are carried out with M48 and the printer remains at X0 Y0 Z10.  When I issue a G29, however, it goes to X0 Y0 Z10, pauses for a fraction of a second while it computes the bed parameters, and then moves to X0 Y32 Z10.  That final move is what I am trying to figure out.

zenmetsu, 2016-08-13:
Actually, I think I know what is going on.  The final position after G29 is actually X0 Y29 Z10.

The final probe position is X0 Y54 Z10.  My probe offset is X0 Y25.   54 - 25 = 29.  After probing, the printer is returning to the final probe point.

That explains the move.  I'll go ahead and modify the code in my branch to avoid this.  I'll go ahead and close the issue since the origin of the post-probe translation is understood.  Thanks!

closed 2016-08-13 by zenmetsu

Roxy-3D, 2016-08-13:
OK!    It is possible ThinkyHead will have some additional questions for you just to make sure things are operating as they should.   There have been recent changes to the probe deploy and stow to clean them up and optimize them.

zenmetsu, 2016-08-13:
Sure thing.  The probe deploy worked fine in RC7.  The stow, not so much.  The triggered switch cause an abort of the probe push move.  Consider my printer on tap if you need me to test the code for anything... despite it being smaller than my big delta, it's stupid fast and less of a cost to fix when experimental code breaks something.