MarlinFirmware/Marlin #4617: Hotend frequency

KokinSSJ, 2016-08-13:
Hello,
Could you explain me how can I change hotend frequency in Marlin software?
I can use SOFT_PWM setting but it decrease control resolution and it's effective on hotend and fan.

I would like to change hotend frequency to around 100Hz ? Could you explain me how can I do that?
How can I change Timer2 - only for heater (PB4 or pin 10). I couldn't find settings for heater.
RAMPS use pin 10 for hotend and pin 9 for fan - both operates on Timer2.  But hotend has special software divider, am I right? Where is software divider for heater and how to change freq?

Ramp 1.4 (bed + FAN + hotend ) + arduino MEGA

Thanks for your help.
Best regards
Dominik

thinkyhead, 2016-08-14:
In Marlin the heaters are always software-PWM driven and as you point out, `SOFT_PWM_SCALE` applies to fans too. So you would have to make a separate PWM handler just for the heaters, basically a second copy of the state machine, plus a separate `SOFT_PWM_SCALE_HEATERS` setting, which you would set to 2 to get a heater PWM period of ~122Hz. With the current implementation this would make the control resolution 1/4 as fine.

KokinSSJ, 2016-08-14:
OK,
Maybe, heater PWM is 488.28Hz but on mosfet output I have maybe 20-50Hz - when hotend achieve temperature around I set - diode blinking but it must glow. 
I would like to increase hotend software frequency for hotend.

I need to avoid frequency which is lower than 50Hz - it's related to my power supply and main grid problems when I use so small frequency.

Could you help me with this problem?

Thanks for your help.
Best regards,
Dominik

thinkyhead, 2016-08-15:
> Could you help me with this problem?

I offer consulting services for a nominal fee.

birkett, 2016-08-15:
> KokinSSJ commented 14 hours ago
> I need to avoid frequency which is lower than 50Hz - it's related to my power supply and main grid problems when I use so small frequency.

Stick an extra filter capacitor between your power supply positive and negative?

closed 2017-10-31 by thinkyhead