MarlinFirmware/Marlin #4622: support for USB on LCD menu

alan707, 2016-08-14:
Rigidbot printers have a USB slot instead of a SD Card slot.  I am working on a pull request to replace the the strings from the LCD menu that have the word "Card" for "USB".  In other words, the LCD would print the word "USB" instead of "Card".  E.g. "USB Inserted".

My question: which branch should I use to make that PR?  Is this something that would be accepted?

boelle, 2016-08-14:
hmmm

i think it would have to be more "intelligent" ie if the controller has usb port them print USB.. and if the controller have SD then do as now

JustAnother1, 2016-08-14:
If detection is not possible maybe something more generic better? External memory available - is probably bad and too long though,...

alan707, 2016-08-14:
My thought was start with something simple that is manually set in the configuration.h file and then work towards more complex things like automatic detection. 

what is the appropriate process to contribute a feature like this? Submit a Pull Request to RC branch?

Grogyan, 2016-08-15:
You would need significantly more code than what Marlin already has, to be
able to mount and unmount a USB, initiate USB data comms, set the BAUD as
well as the need for a USB host to serial converter on the main board.

Most of this can be done with OctoPrint, and I know that Gina has already
got USB as a future support item.

Handling dynamic USB may be a lttle too much for an 8bit micro.

thinkyhead, 2016-08-15:
> Rigidbot printers have a USB slot instead of a SD Card slot.

Marlin receives data from USB (serial) and when printing from SD it actively reads from the SD file. Are you saying that Rigidbot connects to something by USB that acts like an SD card?

alan707, 2016-08-15:
> Are you saying that Rigidbot connects to something by USB that acts like an SD card, and that the SD_DETECT_PIN is attached to the USB port?

@thinkyhead I am actually referring to the front LCD Panel which has a USB connector so you can plug in a USB stick instead of an SD Card (see picture below).  @Grogyan I've tried it with the latest Marlin RCBugFix and it works great, as it thinks it's a SD Card volume being inserted.

[image]

My simple proposal is to create a conditional statement to replace the word "Card" to "USB".  So if you stick a USB drive into the front LCD panel slot (Type A USB connector), it will read correctly: "USB inserted" instead of "Card inserted".  That's all 😄 

The USB connector that you are referring to is also present in the Rigidbot but it is a Type B connector and I actively use it to send commands over serial.  I've included the standard nomenclature of 3 USB connectors below to facilitate this conversation.

[image]

thinkyhead, 2016-08-15:
Aha, I see! They happen to plug into a USB port, but they are called "Flash Drive" or "USB Stick". So perhaps instead of just "USB" use "USB Drive" or "USB Stick."

alan707, 2016-08-15:
Yes, exactly!  I think I'll go with "USB Drive".  Will report back with progress.

Grogyan, 2016-08-15:
@alan707 can you please provide a link to the schematic for the display.

As a flash drive works through the USB protocol, though it can work without it, it isn't advised.
I suspect that there must be at least a USB host device on it, which is then connected via the SPI port back to the main controller.
If this is the case, then I also have an even more serious concerns about whether there is a way to break out the SPI bus to use it for additional tasks (MAX31855 thermocouple for example)

thinkyhead, 2017-10-31:
Please post a new issue detailing your progress. Link back to this one.

closed 2017-10-31 by thinkyhead

marcio-ao, 2018-02-26:
This thread is interesting. We would like to implement USB on our printers. How does Rigidbot do it? Have they released their code? What chip do they use?

thinkyhead, 2018-02-26:
I suspect something like https://shop.clickandbuild.com/cnb/shop/ftdichip?productID=62&op=catalogue-product_info-null