MarlinFirmware/Marlin #4627: Feature request: Retract and lift when pause printing

Kaibob2, 2016-08-15:
A nice feature would be to lift the toolhead a little, retract and move to X0Y0 or somewhere else when pause SD is selected via display. Now, when you select SD Pause the hot toolhead melts the print object and, if it stops on a perimeter, it leads to a defective print.

Fixx, 2016-08-18:
Good idea, +1.

adamfilip, 2016-08-18:
I often pause a print early on to confirm the gap between the nozzle and the bed to check its in the right range. so this feature would prevent me from doing this.

How about Pause and then option for Pause and lift.

Kaibob2, 2016-08-19:
I tried coding something like this but ran into some difficulties due to my limited coding skills. I added this to ultralcd.cpp:
```
#if ENABLED(SDSUPPORT)
    static void lcd_sdcard_pause() {
      card.pauseSDPrint();
      print_job_timer.pause();
      #if ENABLED(LIFT_ON_PAUSE)
        enqueue_and_echo_commands_P(PSTR("T0\nG91\nG0 E-2 F1000\nG1 Z20 F540\nG90\nG1 X0 Y0 F3000"));
        #endif
    }

    static void lcd_sdcard_resume() {
      if (card.sdprinting){
      card.startFileprint();
      print_job_timer.start();}
      else {
      enqueue_and_echo_commands_P(PSTR("T0\nG91\nG1 Z-20 F540\nG90"));
      card.startFileprint();
      print_job_timer.start();}
    }
```

And this to Configuration.h:

```
#define LIFT_ON_PAUSE
#if ENABLED(LIFT_ON_PAUSE)
  #define PAUSE_RETRACT_SPEED   100
  #define PAUSE_RETRACT_LENGTH  2
  #define PAUSE_X_POSITION      0
  #define PAUSE_Y_POSITION      0
  #define PAUSE_Z_LIFT          20
  #define PAUSE_Z_SPEED          540
  #define PAUSE_XY_SPEED         3000
#endif
```

This works, but i assume this is the worst implementation one can imagine.
Further all defines in Configuration.h for length and speed etc. are obsolete because i don't get the syntax correct for implementing it to 

`enqueue_and_echo_commands_P(PSTR("T0\nG91\nG0 E-2 F1000\nG1 Z20 F540\nG90\nG1 X0 Y0 F3000"));`

The way to recover Z-position is also very very dumb, but maybe someone more skillful will attend to this :)

alejandro-anv, 2016-11-21:
It would be usefull an option for pause on next layer. The program could wait to the next z movement for avoiding a pause in middle of a layer.

Blue-Marlin, 2016-11-21:
Marlin has no clue about layers.
Detecting layes by z-moves is no option. Z-lift on retracts, spiral vase mode, ... all would be detected as layer changes.

alejandro-anv, 2016-11-21:
No problem. A z lifts on retract (or even any retract) are good moments for pausing the print.

closed 2017-07-26 by Kaibob2