MarlinFirmware/Marlin #4628: ST7920 damaged fonts

c64pl, 2016-08-15:
Hi
In file ultralcd_st7920_u8glib_rrd.h is bug.

#elif F_CPU == 16000000
  #define CPU_ST7920_DELAY_1 DELAY_0_NOP
  #define CPU_ST7920_DELAY_2 DELAY_0_NOP
  #define CPU_ST7920_DELAY_3 DELAY_1_NOP

I fix it.

#elif F_CPU == 16000000
  #define CPU_ST7920_DELAY_1 DELAY_1_NOP
  #define CPU_ST7920_DELAY_2 DELAY_1_NOP
  #define CPU_ST7920_DELAY_3 DELAY_2_NOP

Processor ATMEGA2561 16MHz

Blue-Marlin, 2016-08-15:
No, That's not a bug. Most of the displays seem to work with the suggested delays. If your panel does not work with the suggested delays, make a new exception specific to your board.

c64pl, 2016-08-15:
The same display on MEGA2560 working properly but MEGA2561 need to do fix. Why?

Blue-Marlin, 2016-08-15:
Likely additional capacities, resistances or inductivities, delaying the signal and or its rise/fall.
As said before - it's board depended. Even different boards with the same processor behave different see `3DRAG`.

0,0,1 are the theoretical delays when you count the processor cycles and compare the times to the ST7920 data sheet. The RAMPS 1.4 - Arduino Mega2560 - REPRAP_DISCOUNT_SMART_CONTROLLER combinations are known to work well with 0,0,0-delays, if the cables are not too long.

If the 2561 needs less cycles then a 2560 for the used assembler instructions, you have to search there.
https://github.com/AnHardt/Marlin/pull/43#issuecomment-223574911

closed 2016-08-19 by thinkyhead