MarlinFirmware/Marlin #4630: How to add new electronics?

c64pl, 2016-08-15:
I made new electronics. What do I need to be added in a later version of Marlin?

Roxy-3D, 2016-08-15:
It would be helpful to know more about the new electronics so we can better answer the question.

c64pl, 2016-08-15:
I proportioned software MERLIN.
Electronics is working properly.
I want a newer version of the software could select it.
What I have to do?

Blue-Marlin, 2016-08-15:
Concluding from #4628 i assume you are building a new board like RAMPS but with a processor on board. 
The required bare minimum is a new entry in boards.h and a pins_*.h file. There are more than enough examples.
If your new hardware is not "open" or, if you don't intend to produce and sell, at least several hundred of this boards, we are not very interested in integrating into mainstream Marlin. We are already carrying too much bloat.

c64pl, 2016-08-15:
processor ATmega2561
4 engines
1x hotend
1x headbed
hotend fan, cooling and electronics cooling
1x led. lighting print - does not have this in MARLIN
USB or Bluetooth
6x endstop

For this two different displays.
1. GLCD 12864 encoder and buzzer.
2. 4x20 44780 of the encoder.

I think that the electronics is better thought out than RAMPS.

I do not know how much electronics for sale.

I have already prepared files. Is it enough that it change here
https://github.com/MarlinFirmware/Marlin

Roxy-3D, 2016-08-15:
> 4 engines

4 stepper motors?   Can you tell me why the electronics is better thought out than the RAMPS board?

c64pl, 2016-08-15:
Smaller, single PCB, integrated slots for the fans, better outlet. Ramps in the trash - could not stand and burned :(
But what is the significance?
I ask how to add the electronics.

Roxy-3D, 2016-08-15:
The easiest way would be to grab a Pin.h file for a board that is similar to what you have.   And then go through the definitions line by line and change anything that is not correct.   You would also want to grab an unused Board # and assign it to your board.

c64pl, 2016-08-15:
So I will. How much I have to wait so that you are added to the MARLIN? Permanently.

Roxy-3D, 2016-08-15:
Well, there isn't a set formula.  But we would need to see a well debugged version of the code.  And we would like some expectation that it the additional code will be of value to the broader Marlin community.   This board would need to be available for sale and people will need to believe it is worth buying and using.

Usually, the path that makes sense is for the Hardware vendor to offer a fork of the Marlin code base with the required changes done in a clean fashion to support the new hardware.    When there is enough demand for native support, the new code gets pulled into the main Marlin branch of the code base.

zenmetsu, 2016-08-16:
Saying something is better than RAMPS is not saying much.  There are quite a few poor design decisions on the RAMPS and I would love to see improvements.  I mentioned a few deficiencies prior to the 1.4 release of the board and the changes were not implemented.  The rest of you can ignore the remainder of this post, this is directed to anyone designing their own board, like @c64pl is doing.

If you are still designing your board, I'll give you the rundown on what happened to me with my first Arduino Mega + RAMPS.  I hooked everything up, verified stepper drivers were inserted properly, attached power, and POP... 5V regulator on the Arduino blew up.  I removed the Mega, verified 5V on the RAMPS was shorted to ground, and found out that the short went away when one of the Pololu drivers was removed.  Long story short, that driver had a fault where the "MS3 pin" was shorted to ground internally within the stepper driver, but only when socketed into the RAMPS.  The "sleep" pin was shorted to ground, and "MS3" was shorted to "reset".  When socketed into the RAMPS, "sleep" and "reset" are connected, thus completing the path from MS3 to GND.  Since MS3 is fed raw 5V without current protection, it is obvious what happened next.

Granted my stepper driver was defective, but current limiting resistors on the "MSx" lines would have prevented the failure.  Also, another common failure mode for these boards is people connecting their endstop switches across 5V and GND... again, current limiting resistors would prevent the damage.

thinkyhead, 2016-08-16:
> I made new electronics. What do I need to be added in a later version of Marlin?

Once Marlin 1.1.0 is released, you should make a branch in your fork from it. Then, apply your changes as a patch to the release version 1.1.0. Then, direct people who use your board to your fork. This approach has the advantage that you don't have to update your fork, and you can distribute your hardware support as a simple .patch file that specifically applies to Marlin 1.1.0. As newer versions of Marlin are released, keep doing the same thing.

If in the long run, if your board becomes widely-used, we will consider adding support for it into the main Marlin project.

zenmetsu, 2016-08-16:
@thinkyhead Do you have a rough estimate on the 1.1.0 release timeline?  I've got an Arduino Due and I know that the RADDS / RAMPS-FD boards are in production.  I heard mention that there could be a pass at porting Marlin once 1.1.0 was released, but I'm taking a crack at bringing the older Marlin4Due code current ahead of this for my own use.  Seeing as I'm admittedly a C++ novice, I'm trying to get a feel for whether I should hold off or not, and that is dependent upon the likelihood of more experienced coders jumping into the mix in the near future.

Roxy-3D, 2016-08-16:
> Do you have a rough estimate on the 1.1.0 release timeline? 

The Marlin team is really hoping RC-8 will get promoted to 'Stable'.   We will have to wait and see. 

> I've got an Arduino Due and I know that the RADDS / RAMPS-FD boards are in production. I heard mention that there could be a pass at porting Marlin once 1.1.0 was released, but I'm taking a crack at [...]

Right now, we are trying to identify the 'correct' 32-Bit hardware to use as a 'Reference Platform'.   If you want to play around and see how far you can get:  GO FOR IT!   

But it is worth while pointing out why I'm trying to get a 32-Bit Reference Platform defined.   I expect we are going to run into some very difficult problems.  It is my hope that if we have 6 or 8 developers all with the same hardware, we will have a better chance of getting to the finish line.  I expect some of the problems we hit to be very difficult and we may need several people to take a look at them and try to figure out how we proceed.  So, in that environment, it is helpful to have everybody on the exact same hardware.

zenmetsu, 2016-08-16:
Well, the Duet does use the same processor as the Due, so it should be a minimal difference mainly dealing with board/pin files.  Anyways, I've somewhat strayed from the topic of the issue.  Is there an IRC channel or hangout or something where simple things can be discussed without cluttering the issues threads?

Roxy-3D, 2016-08-16:
No IRC...  But just start a thread.  People can ignore its contents if they are not interested in the subject matter.

thinkyhead, 2016-08-19:
RC8 will follow once we figure out whether there are real discrepancies seen between Z after G28 versus G29, or if this turns out to just be a very uneven bed plane throwing off the Z.

closed 2017-10-31 by thinkyhead