MarlinFirmware/Marlin #4631: IDE requirements

boelle, 2016-08-15:
Hi

was just wondering if the requirement for IDE 1.6.x so far is just a check

or is there something in IDE 1.6.x that is really used?

Roxy-3D, 2016-08-15:
I think the answer was we had too much 'flakey' behavior and were not sure if we were fighting compiler bugs.  We did not know what was causing trouble so limiting the build environment to 'current' versions of the IDE was thought to be wise.

I think some number of the problems we were seeing were conflicts with the U8G Graphics library.   It maybe that constraint can be relaxed now.   If you need to v1.05 to compile and build for your hardware, I suspect it is going to work OK.   There may be a few syntax issues that have crept into the code over the last year.   

I found one with the preprocessor not liking boolean expressions with the ENABLED() and DISABLED() macros when trying to get the UBL code up and running.   If you find something like that, we should try to fix it, but in the mean time, you can just look at the #ifdef block and decide whether it should be included or not and force the correct behavior.

boelle, 2016-08-15:
oki... so if it yells over the IDE version i will suppress that in sanity check and see what else comes up

i'm not after a big fancy compile... just the very basic... bed level will the most crazy i will do... LCD no, i will use raspberry pi and octoprint... SD card.... maybe, a very small maybe

boelle, 2016-08-15:
commenting out the sanity check for version only gave me this short list:

In file included from /nozzle.h:27,
                 from Marlin_main.cpp:62:
/point_t.h: In constructor 'point_t::point_t(float, float)':
point_t.h:49: error: type 'point_t' is not a direct base of 'point_t'
/point_t.h: In constructor 'point_t::point_t(float, float, float)':
point_t.h:59: error: type 'point_t' is not a direct base of 'point_t'
In file included from Marlin_main.cpp:63:
/duration_t.h: In constructor 'duration_t::duration_t()':
duration_t.h:36: error: type 'duration_t' is not a direct base of 'duration_t'

Roxy-3D, 2016-08-15:
Is this with v1.05?   Or maybe v.22 ?

My guess is the preprocessor is behaving differently between the versions.   It maybe the easiest way to 'fix' your compile is force the #include of the files where point_t and duration_t are defined.   But that is just a guess.

boelle, 2016-08-15:
where would i put those includes ?

not a rush, but could be nice to preserve some comp. with older ide's.

Roxy-3D, 2016-08-15:
You would put the #include "filename.h" towards the top of the file, probably with the other #include directives.   But doing a quick search I don't see point_t and duration_t anywhere in the code base that I searched.  (RCBugFix with UBL code from July 4th 2016).   So those types may have been added recently.

boelle, 2016-08-15:
@thinkyhead can you cast a small bone?

boelle, 2016-08-15:
oh... IDE 1.0.6

Roxy-3D, 2016-08-15:
If you post your configuration.h and configuration_adv.h file it makes it possible to reproduce the issue.

boelle, 2016-08-15:
is unchanged so its the same as on github...

and the other i only change for the SAV MK1 board but will throw both in paste bin and report back in a jiffy

boelle, 2016-08-15:
http://pastebin.com/yV77EvAe

and 

http://pastebin.com/XDwYjMfs

thinkyhead, 2016-08-16:
I notice that `duration_t` and `point_t` are both struct objects rather than full classes. Perhaps the older compiler dislikes that. Try changing them to classes instead.

boelle, 2016-08-16:
and how do a noob do that ? :-D

thinkyhead, 2016-08-19:
A n00b studies until he becomes a l33t.

boelle, 2016-08-20:
:-/ a what?

anyways you know i'm not a programmer so a hit of what to edit where would be usefull so i could test it... and i think the work could benefit more than just myself

@Roxy-3D would be one and everyone else with hardware that no longer gets updated hardware files

Saying to trash your better than Ramps electronics is not a good way to go. it will push people away and in the arms of firmware that do support it

Roxy-3D, 2016-08-20:
>> A n00b studies until he becomes a l33t.
> :-/ a what?

https://en.wikipedia.org/wiki/Leet

> @Roxy-3D would be one and everyone else with hardware that no longer gets updated hardware files
> Saying to trash your better than Ramps electronics is not a good way to go. it will push people away and in the arms of firmware that do support it

Yes...   But there is a flip side to the coin.   It takes a lot of effort and expertise to maintain those Arduino Board files.   I'm sure I could learn how to do it and make a new file for my PrintRboard.   But I would rather spend my time working on software I enjoy putting together.   If the company doesn't care enough about their board or their customers, does it make sense for the Marlin Team to use its limited resources to extend the life of some obsolete boards?   Especially when replacement hardware can be obtained for $15 ?

At some point, we are going to have this same discussion with regard to 8-Bit vs. 32-Bit boards.  When we get onto a 32-Bit platform, we will have the CPU power to do many things that we can not do now.  There will be people saying "If you spent extra time to code the loops tighter, you could make those new features work on the old 8-Bit AVR boards too!"     Well... it might make sense at that point in time to suggest to those people still on 8-Bit AVR boards that they have two options.  They can move to a supported 32-Bit board, or they can invest the time to rewrite the loops that are not tight enough for the 8-Bit AVR.

thinkyhead, 2016-08-21:
> a hint of what to edit where would be useful

I've traveled halfway across the country to work on a very complicated project, and I simply don't have the time to get into it. Hopefully one of the other programmers around here has more time to help. Good luck!

boelle, 2016-08-24:
>of some obsolete boards? Especially when replacement hardware can be obtained for $15 ?

still being sold, the maker just refers to use IDE 1.0.6

> "If you spent extra time to code the loops tighter, 

what happened to Erik Zalm's wish that the very basic stuff should work of a 644 chip? ie usb print

closed 2016-08-24 by Roxy-3D

thinkyhead, 2016-08-24:
> what happened to Erik Zalm's wish that the very basic stuff should work of a 644 chip?

It is alive and well in my heart. I think currently the code has grown due to having safety features enabled by default and in filling in other oversights. We will continue to find ways to bring down the code size as time goes on. Often in software design we first elaborate, and then later we optimize. All this takes time and effort, but we are very short on skilled help at this time.

My recommendation for those with more modest boards is to stick with older versions of Marlin if they do the job and check back with the newer versions from time-to-time to see how they're fitting.