MarlinFirmware/Marlin #4632: Issue: Heating failed

Aschie4589, 2016-08-15:
I'm not completelu sure what causes this, but I'll post here anyway. If there's something wrong, just delete it.

Everytime I try and heat up my printer's hotend, I get the "Heating failed" error. I tried tuning the PID values, but nothing. Once it reaches a temperature that is 10°C from the target temperature (and I know this is when the PID control mechanism should kick in), the temperature stops increasing and remains constant. I know it's not a heater problem, because it's not temperature dependant (it doesn't matter whether I choose 240°C, 230°C or 220°C for example, temperature will aways stay 10°C below that). Could it be a firmware error? Possibly some kind of bug? If it's not, I'm at loss on what it could be...

mjmeans, 2016-08-15:
I too have the same problem. I'm using repetier-host and when I look at the graph, I see a jagged line for the bed temperature with constant bed power applied. I have already eliminated outside environmental causes, such as variances of power supply voltage (which doesn't vary enough to make the voltage change from 12.5 to anything else) and also eliminated drafts as a concern (doesn't correspond to any actual fan turning on or off or any room air flow). I have not further diagnosed this on my system. My bed has a 2mm cork bottom insulator attached to a capton heater attached to an 2mm aluminum heat spreader with a 3mm borosilicate glass plate on top. So I was thinking that maybe heating of the cork was causing an unexpected and random pull on the thermistor sandwiched under it and thereby randomly pulling the thermistor away from the bed. Or maybe it could just be a bad ramps or arduio board. As I said, I have no idea why it's happening. BTW, I'm using thermistor code 11.

mjmeans, 2016-08-15:
Here is an excerpt from the ACK output in Repetier-Host and I've highlighted the anomalous temperature readings. As you can see, the hotter the reading is, the more likely there is an anomalous reading.

> 13:07:37.515 : ok T:23.0 /0.0 B:24.0 /90.0 @:0 B@:127
> 13:07:40.557 : ok T:23.0 /0.0 B:24.0 /90.0 @:0 B@:127
> 13:07:43.608 : ok T:23.0 /0.0 B:24.0 /90.0 @:0 B@:127
> 13:07:46.679 : ok T:23.0 /0.0 B:24.5 /90.0 @:0 B@:127
> 13:07:49.746 : ok T:23.0 /0.0 B:24.8 /90.0 @:0 B@:127
> 13:07:52.789 : ok T:23.0 /0.0 B:25.1 /90.0 @:0 B@:127
> 13:07:55.826 : ok T:23.0 /0.0 B:25.7 /90.0 @:0 B@:127
> 13:07:58.860 : ok T:23.0 /0.0 B:26.2 /90.0 @:0 B@:127
> 13:08:01.901 : ok T:23.0 /0.0 B:27.0 /90.0 @:0 B@:127
> 13:08:04.952 : ok T:23.0 /0.0 B:27.5 /90.0 @:0 B@:127
> 13:08:07.957 : ok T:23.0 /0.0 B:28.0 /90.0 @:0 B@:127
> 13:08:10.998 : ok T:23.0 /0.0 B:28.4 /90.0 @:0 B@:127
> 13:08:14.002 : ok T:23.0 /0.0 B:29.1 /90.0 @:0 B@:127
> 13:08:17.051 : ok T:23.0 /0.0 B:29.5 /90.0 @:0 B@:127
> 13:08:20.087 : ok T:23.0 /0.0 B:30.1 /90.0 @:0 B@:127
> 13:08:23.160 : ok T:23.0 /0.0 B:30.6 /90.0 @:0 B@:127
> 13:08:26.179 : ok T:23.0 /0.0 B:31.3 /90.0 @:0 B@:127
> 13:08:29.198 : ok T:23.0 /0.0 B:31.7 /90.0 @:0 B@:127
> 13:08:32.202 : ok T:23.0 /0.0 B:32.4 /90.0 @:0 B@:127
> 13:08:35.207 : ok T:23.0 /0.0 B:32.8 /90.0 @:0 B@:127
> 13:08:38.234 : ok T:23.0 /0.0 B:33.5 /90.0 @:0 B@:127
> 13:08:41.338 : ok T:23.0 /0.0 B:34.0 /90.0 @:0 B@:127
> 13:08:44.374 : ok T:23.0 /0.0 B:34.6 /90.0 @:0 B@:127
> 13:08:47.412 : ok T:23.0 /0.0 B:35.3 /90.0 @:0 B@:127
> 13:08:50.515 : ok T:23.0 /0.0 B:35.9 /90.0 @:0 B@:127
> 13:08:53.571 : ok T:23.0 /0.0 B:36.4 /90.0 @:0 B@:127
> 13:08:56.638 : ok T:23.0 /0.0 B:37.1 /90.0 @:0 B@:127
> **13:08:59.694 : ok T:23.0 /0.0 B:36.6 /90.0 @:0 B@:127**
> 13:09:02.767 : ok T:23.0 /0.0 B:38.5 /90.0 @:0 B@:127
> 13:09:05.788 : ok T:23.0 /0.0 B:39.0 /90.0 @:0 B@:127
> 13:09:08.791 : ok T:23.1 /0.0 B:39.6 /90.0 @:0 B@:127
> 13:09:11.810 : ok T:23.0 /0.0 B:40.2 /90.0 @:0 B@:127
> 13:09:14.914 : ok T:23.0 /0.0 B:40.8 /90.0 @:0 B@:127
> 13:09:17.986 : ok T:23.1 /0.0 B:41.4 /90.0 @:0 B@:127
> 13:09:21.071 : ok T:23.0 /0.0 B:42.0 /90.0 @:0 B@:127
> 13:09:24.077 : ok T:23.1 /0.0 B:42.4 /90.0 @:0 B@:127
> 13:09:27.134 : ok T:23.0 /0.0 B:42.9 /90.0 @:0 B@:127
> 13:09:30.206 : ok T:23.0 /0.0 B:43.5 /90.0 @:0 B@:127
> 13:09:33.257 : ok T:23.0 /0.0 B:44.1 /90.0 @:0 B@:127
> 13:09:36.376 : ok T:23.0 /0.0 B:44.3 /90.0 @:0 B@:127
> 13:09:39.367 : ok T:23.0 /0.0 B:44.4 /90.0 @:0 B@:127
> **13:09:42.385 : ok T:23.0 /0.0 B:40.8 /90.0 @:0 B@:127**
> 13:09:45.443 : ok T:23.1 /0.0 B:45.0 /90.0 @:0 B@:127
> 13:09:48.462 : ok T:23.1 /0.0 B:45.1 /90.0 @:0 B@:127
> 13:09:51.497 : ok T:23.0 /0.0 B:47.2 /90.0 @:0 B@:127
> 13:09:54.503 : ok T:23.0 /0.0 B:47.7 /90.0 @:0 B@:127
> 13:09:57.576 : ok T:23.1 /0.0 B:47.8 /90.0 @:0 B@:127
> 13:10:00.580 : ok T:23.1 /0.0 B:48.3 /90.0 @:0 B@:127
> 13:10:03.615 : ok T:23.0 /0.0 B:48.9 /90.0 @:0 B@:127
> 13:10:06.657 : ok T:23.1 /0.0 B:49.2 /90.0 @:0 B@:127
> 13:10:09.693 : ok T:23.1 /0.0 B:49.8 /90.0 @:0 B@:127
> 13:10:12.743 : ok T:23.2 /0.0 B:51.4 /90.0 @:0 B@:127
> **13:10:15.800 : ok T:23.3 /0.0 B:49.2 /90.0 @:0 B@:127**
> 13:10:18.822 : ok T:23.3 /0.0 B:52.2 /90.0 @:0 B@:127
> **13:10:21.827 : ok T:23.4 /0.0 B:51.7 /90.0 @:0 B@:127**
> 13:10:24.831 : ok T:23.3 /0.0 B:53.0 /90.0 @:0 B@:127
> 13:10:27.835 : ok T:23.2 /0.0 B:53.5 /90.0 @:0 B@:127
> **13:10:30.870 : ok T:23.4 /0.0 B:53.2 /90.0 @:0 B@:127**
> 13:10:33.960 : ok T:23.4 /0.0 B:54.4 /90.0 @:0 B@:127
> 13:10:36.966 : ok T:23.5 /0.0 B:54.8 /90.0 @:0 B@:127
> 13:10:39.985 : ok T:23.5 /0.0 B:55.2 /90.0 @:0 B@:127
> 13:10:43.027 : ok T:23.5 /0.0 B:55.6 /90.0 @:0 B@:127
> 13:10:46.047 : ok T:23.5 /0.0 B:56.0 /90.0 @:0 B@:127
> 13:10:49.053 : ok T:23.5 /0.0 B:56.4 /90.0 @:0 B@:127
> 13:10:52.104 : ok T:23.5 /0.0 B:56.9 /90.0 @:0 B@:127
> 13:10:55.176 : ok T:23.5 /0.0 B:57.3 /90.0 @:0 B@:127
> 13:10:58.181 : ok T:23.5 /0.0 B:57.8 /90.0 @:0 B@:127
> 13:11:01.268 : ok T:23.5 /0.0 B:58.0 /90.0 @:0 B@:127
> 13:11:04.274 : ok T:23.5 /0.0 B:58.5 /90.0 @:0 B@:127
> 13:11:07.310 : ok T:23.5 /0.0 B:58.0 /90.0 @:0 B@:127
> 13:11:10.315 : ok T:23.5 /0.0 B:59.2 /90.0 @:0 B@:127
> **13:11:13.355 : ok T:23.5 /0.0 B:58.4 /90.0 @:0 B@:127**
> **13:11:16.361 : ok T:23.6 /0.0 B:57.9 /90.0 @:0 B@:127**
> **13:11:19.365 : ok T:23.7 /0.0 B:59.0 /90.0 @:0 B@:127**
> 13:11:22.371 : ok T:23.7 /0.0 B:60.7 /90.0 @:0 B@:127
> 13:11:25.492 : ok T:23.8 /0.0 B:61.0 /90.0 @:0 B@:127
>

Roxy-3D, 2016-08-15:
My guess is you either have some electrical noise on the thermistor wires or you have a bad connection somewhere on them.   It isn't likely your temperature really did this:
~~~~
13:10:12.743 : ok T:23.2 /0.0 B:51.4 /90.0 @:0 B@:127
13:10:15.800 : ok T:23.3 /0.0 B:49.2 /90.0 @:0 B@:127
13:10:18.822 : ok T:23.3 /0.0 B:52.2 /90.0 @:0 B@:127
~~~~

Blue-Marlin, 2016-08-16:
If your temperature noise level, for the bed, is that high, for what ever reason, you can tune:
```
  /**
   * Whenever an M140 or M190 increases the target temperature the firmware will wait for the
   * WATCH_BED_TEMP_PERIOD to expire, and if the temperature hasn't increased by WATCH_BED_TEMP_INCREASE
   * degrees, the machine is halted, requiring a hard reset. This test restarts with any M140/M190,
   * but only if the current temperature is far enough below the target for a reliable test.
   *
   * If you get too many "Heating failed" errors, increase WATCH_BED_TEMP_PERIOD and/or decrease
   * WATCH_BED_TEMP_INCREASE. (WATCH_BED_TEMP_INCREASE should not be below 2.)
   */
  #define WATCH_BED_TEMP_PERIOD 60                // Seconds
  #define WATCH_BED_TEMP_INCREASE 2               // Degrees Celsius
```
`WATCH_BED_TEMP_INCREAS` has to be at least the magnitude of the noise! With `WATCH_BED_TEMP_INCREASE` = 4 and `WATCH_BED_TEMP_PERIOD` = 120, you have the same slope, but a better noise resistance. If that is not enough, increasing the time further will help.

Blue-Marlin, 2016-08-16:
Best would be to find the reason for the noise. In many case using shielded thermistor wires, or at least twisting them, helped.
The thermistor table for type 11 looks ok to me.

psavva, 2016-09-07:
This will also be evident when your thermistor wires are bundled with your fan wires. Try a test, disconnect the fans from your Hotend, and see if that helps... 

Another thing is to ensure that your fans are not cooling down the heatblock while the heater is trying to warm it up.  Once the PID kicks in, your heater will have a hard time trying to reach the target temperature, whist your fan is cooling it down too fast/direct...

closed 2016-09-08 by thinkyhead

fefeaneta8, 2018-08-23:
Hi ich wollte meinen Anet A8 auch endlich mal auf Marlin upgraden.
(Hatte bis jetzt das Werksprogramm oben)

Habe schon den Hardware ordner in den Arduino ordner verschoben.
Bei Malin die Anet A8 dateien in den Hauptordner kopiert und gestartet.

Ich bekomme immer im arduino fenster beim Durchchecken eine Fehlermeldung: pasting "/* (bed)*/" and "_RPORT" does not give a valid preprocessing token und es makiert mir dabei folgende Zeile:
#define HEATER_BED_PIN 12 // (bed) 

Was kann ich dagegen machen?