MarlinFirmware/Marlin #4633: Question: Kossel Mini sample configuraiton

mjmeans, 2016-08-15:
Does anyone know why the Kossel Mini sample configuration has a `#define DELTA_EFFECTOR_OFFSET 19.9 // mm`?

If you examine [https://github.com/jcrocholl/kossel/blob/master/effector.scad] you can clearly see `offset = 20;  // Same as DELTA_EFFECTOR_OFFSET in Marlin.`. Aso if you add this line to the scad file `translate([0,-20,height/2]) rotate([0,90,0]) cylinder(r=1.4,h=separation,$fn=16,center=true);` and click CTRL+6 and zoom in you can visually see that 20 is the actual distance, not 19.9.

thinkyhead, 2016-08-16:
The source of this is here:

https://github.com/jcrocholl/Marlin/blob/kossel/Marlin/Configuration.h#L115

mjmeans, 2016-08-17:
I'll then have to direct my query to Johann Rocholl. Thanks.

mjmeans, 2016-08-17:
[here](https://github.com/jcrocholl/Marlin/commit/6ba423bd619a20bfb81a38bee45c0fac12c789ba) on June 13, 2013 is the actual source change merge for this value. But there was no reason given for this. He didn't comment specifically on this line.

mjmeans, 2016-08-17:
Looking at the source jcrocholl's source for the effector.scad there wasn't any change in the rod mounting locations after June 13, 2013.

mjmeans, 2016-08-18:
I have sent him an email to ask why this value is 19.9 instead of 20.0. If he replies to that, I'll post it here.

thinkyhead, 2016-08-21:
I presume he changed it because it worked better for his prototype. After all, engineering tolerances are what they are.

closed 2017-10-31 by thinkyhead