MarlinFirmware/Marlin #4635: LCD Lines shifted during M600 Filament change

Kaibob2, 2016-08-16:
When i start a Filament change routine the Display shows:
[image]
When i turn the encoder the display shows:
[image]

The empty line at the top of display stays there even if the filament change routine is over, The Main screen has an empty line at the top then too.

thinkyhead, 2016-08-19:
The glitch is interesting. Hardware-related, perhaps display timing-related. You may have some luck playing with these lines in `ultralcd_st7920_u8glib_rrd.h`:
```cpp
// If you want you can define your own set of delays in Configuration.h
//#define ST7920_DELAY_1 DELAY_0_NOP
//#define ST7920_DELAY_2 DELAY_0_NOP
//#define ST7920_DELAY_3 DELAY_0_NOP
```

> The empty line at the top of display
It appears when you turn the encoder clockwise, when you turn the encoder counter-clockwise, or both?

boelle, 2016-08-28:
@thinkyhead maybe also close this one? ie due to lack of repsonse from @Kaibob2

Kaibob2, 2016-08-29:
@boelle As i seem to be the only one with this problem feel free to close here. I'm not using the filament change routine anyway and right now i have other hardware related things to fix.

thinkyhead, 2016-08-30:
I'll mess with it later, see if it still needs a fix, and then close it as fixed once I know it's patched. Later on today / tonight.

boelle, 2016-09-03:
@thinkyhead remember to close this one

@Kaibob2 i only posted this as we want to close bug reports if they are fixed... if the issue is still there with latest bugfix (updates many times each week) let us know

you can always ping @thinkyhead on this issue and it can be reopened

boelle, 2016-09-20:
@Kaibob2 have you had time to retest this one? maybe @thinkyhead did fix it by "mistake" :-D

Kaibob2, 2016-09-20:
@boelle I'm not using it by default, so I did no further testing. I'm on holiday for 2 weeks. I will test is when I get home and report back.

thinkyhead, 2016-09-20:
I haven't changed the LCD scrolling code, so I expect it's still extant. Probably simple to fix. I'll await your feedback @Kaibob2.

Kaibob2, 2016-10-02:
I tested the filament routine again and, like expected, it still behaves like before.
See attached video.
The routine itself works perfect, it's just the shifted lines that are a little strange.

[20161002_211040_001.zip](https://github.com/MarlinFirmware/Marlin/files/505198/20161002_211040_001.zip)

thinkyhead, 2016-10-07:
Try this:
```diff
#define START_SCREEN_OR_MENU(LIMIT) \
  ENCODER_DIRECTION_MENUS(); \
  encoderRateMultiplierEnabled = false; \
  if (encoderPosition > 0x8000) encoderPosition = 0; \
  static int8_t _countedItems = 0; \
  int8_t encoderLine = encoderPosition / ENCODER_STEPS_PER_MENU_ITEM; \
  if (_countedItems > 0 && encoderLine >= _countedItems - LIMIT) { \
-   encoderLine = _countedItems - LIMIT; \
+   encoderLine = max(0, _countedItems - LIMIT); \
    encoderPosition = encoderLine * (ENCODER_STEPS_PER_MENU_ITEM); \
  }
```

Kaibob2, 2016-10-09:
@thinkyhead Great! That's it. No more shifted lines with the modification above. Please merge and close this one here.

closed 2016-10-10 by Kaibob2