MarlinFirmware/Marlin #4636: Core XY - Auto level and Z homming.

Moztho, 2016-08-16:
I´m working on a Core XY 3D printer. I want to add the auto level feature but i ran into a problem, the Marlin firmware requires me to connect the capacitive sensor to Zmin and my Z homing switch is on the lower part of the 3d printer. How can i configure this in order to get both working properly?

adamfilip, 2016-08-19:
use Z Max Pin?

thinkyhead, 2016-08-22:
The newest versions of Marlin give you alternatives. You can use the Z min pin if you want, and so `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` is enabled by default (though I think it should probably only be default in the delta configurations). If you want to use a different pin, you should enable `Z_MIN_PROBE_ENDSTOP` instead. Most of the current pins files now define a `Z_MIN_PROBE_PIN` and either assign a unique pin to it, or if no unique pin is available, they assign it to the `Z_MAX_PIN`.

closed 2017-10-31 by thinkyhead