MarlinFirmware/Marlin #4638: UBL Z_OFFSET

adamfilip, 2016-08-17:
@Roxy-3D Just starting a new thread related to UBL  UBL Z_OFFSET issues.

> I understand who has what... But I agree. @adamfilip let's start a separate thread for UBL Z_OFFSET issues. Otherwise, we are going to confuse everybody.
Here is one more thing you are doing differently than me. You are actually measuring things, and trying to plug in numbers. In theory, that should work just fine. But the way I operate is I've always looked at what the error was, and corrected most of it, but always made sure I would not drive the nozzle into the glass. So, it takes me 3 or 4 tries to get the nozzle's Z_OFFSET_FROM_EXTRUDER exactly correct. But it is perfect after those 3 or 4 iterations.
The reason I'm mentioning it is this: With the Z-Fade-Height active, it is possible there is a disconnect and either too much or too little Z-Height compensation is being done. And that compensation is going to change based on how much you are tying to correct.
Please try 3 or 4 (or 7) iterations to get Z_OFFSET_FROM_EXTRUDER set perfectly. And try to do it on a place on the bed where you know things are relatively flat. And don't forget to do a M502 and M500 each cycle (I always forget to do that!!!)
Mean while... I'm going to start setting up to go the opposite route and see what I find.

I will  keep trying and let you know.

adamfilip, 2016-08-18:
I tried three times..
Adjusted (reduced) Z_OFFSET_FROM_EXTRUDER from -17mm to -15
the end result is that the Gap increased. :\

im going to try and try it at -17.55 which should increase the gap but 
im not sure logic is at play here..

adamfilip, 2016-08-18:
Tried a -17.55 Z_OFFSET_FROM_EXTRUDER and I had to babystep it up .8mm to get a .1mm gap
next im trying a -16.75

adamfilip, 2016-08-18:
-16.75 Z_OFFSET_FROM_EXTRUDER seems to be working.
i have around a .2mm gap when printing the first layer.

Roxy-3D, 2016-08-18:
> Adjusted (reduced) Z_OFFSET_FROM_EXTRUDER from -17mm to -15
the end result is that the Gap increased. :\

I've been out playing darts.  And I always win a couple of beers.   But I'm pretty sure that as you shrink the negative offset, the nozzle goes higher and higher.    What ever offset you have, if you say you have less magnitude (because you are negative) you are really raising the nozzle.

adamfilip, 2016-08-18:
is there a way to save the mesh between firmware updates.. if i do a M502 it get erased

Roxy-3D, 2016-08-18:
If you don't change the size of the Mesh, it is still there.    Just say:  G29 L 1     Or what ever number you stored it at.    And then a G29 A  to activate.    And if you do a G29 S 1   it will save the new state information and things will automatically load each time the machine powers up or resets.

adamfilip, 2016-08-23:
Ive noticed that when running a G29 P1 to create the initial mesh. when it moves between points it doesnt raise the head before each move. it should as the probe is sliding across the surface during this move. I have taken a video to show this. BTW I use a cherry Keyboard switch for my probe.

https://youtu.be/6UdZ1psBSq0

It would really be nice if you could get a multipler for when doing a G29 P2 for the manual probing
the way my printer is setup I use geared steppers with a very fine acme screw so it has alot of resolution but that makes scrolling down each point take one minute per point. also if it didnt raise up as much that would help too. it starts at +3mm but i end up scrolling it down to around -5mm, should just take the previous reading and raise by 1mm

adamfilip, 2016-08-23:
Reloaded most recent firmware,
recreated the Mesh with P1
manually found all missing spots
but the manual spots are much lower then the automatic spots
see mesh report below

RECEIVED: Bed Topography Report:
RECEIVED: 
RECEIVED: (0,9)                                                                                                                                                             (9,9)
RECEIVED: (0,250)                                                                                                                                                         (280,250)
RECEIVED:  -4.05000    -3.81000    -3.57000    -0.29620    -0.13475     0.00697     0.16214     0.32298     0.47188    -1.76163   
RECEIVED:  
RECEIVED:  -4.02000    -3.84000    -3.63000    -0.26416    -0.12357     0.03130     0.19228     0.32335     0.43834     0.53859   
RECEIVED:  
RECEIVED:  -3.95000    -3.67000    -3.50000    -0.28486    -0.12879     0.02842     0.13917     0.29072     0.42277     0.48676   
RECEIVED:  
RECEIVED:  -3.84000    -3.69000    -3.57000    -0.23092    -0.10673     0.01656     0.14333     0.25476     0.34292     0.43690   
RECEIVED:  
RECEIVED:  -3.80000    -3.68000    -3.53000    -0.20584    -0.05543    -0.00761     0.11304     0.21283     0.31557     0.36831   
RECEIVED:  
RECEIVED:  -3.81000    -3.77000    -3.46999    -0.19519    -0.10899    -0.00353     0.07904     0.15368     0.25393     0.29593   
RECEIVED:  
RECEIVED:  -3.79000    -3.66000    -3.47999    -0.17865    -0.10491    -0.02189     0.05978     0.12157     0.18427     0.24162   
RECEIVED:  
RECEIVED:  -3.69000    -3.61000    -3.43000    -0.17139    -0.11360    -0.06034     0.01241     0.07428     0.14794     0.17649   
RECEIVED:  
RECEIVED:  -3.56000    -3.42000    -3.40000    -0.17547    -0.10801    -0.04516     0.03515     0.07481     0.12474     0.14348   
RECEIVED:  
RECEIVED: [-3.46000]  -3.24000    -3.18000    -3.20000    -3.13000    -3.09000    -3.03000    -2.99000    -3.08000    -3.07000

adamfilip, 2016-08-24:
@Roxy-3D Do you think that me using H0 confused the system?

Roxy-3D, 2016-08-24:
> Do you think that me using H0 confused the system?

That shouldn't confuse it.  But I would think you want a small, positive amount for the H number because you do want the nozzle to not scrape as it moves across the bed.    It is very possible that there is a bug in the manual probing.   I'll try what you did and see what happens.

But all that aside...   I'm getting more and more of the opinion that Manual Probing is too much of a hassle.  I think if a user just manually fills the unprobed locations in the mesh with a 'safe' number, that is the way to go.     It looks like the left edge of where you probed was -.25 or so.   If you were to do a G29 P3 C0.0 R   you would fill all those areas.  And probably the Mesh Points would be too high initially.   But that lets you move on to do a G26 at which point, you can dial in the correct height based on what got printed.

I don't even look at the LCD Panel when I'm editing the Mesh.  I kind of know how far the Encoder Wheel turns for +/- .1mm and I just guess at what is needed based on how much the filament squished out from under the nozzle.   The first G26 will get things pretty usable.  The second G26 gets it perfect.

(But back to what you saw:   I'll see if I can duplicate it, and if so, I'll get it fixed.  Your Topology Map does look 'wrong' and if G29 P2 H0 did that...  I should be able to get that fixed!)

Roxy-3D, 2016-08-24:
I just did a G29 P1 M  and then followed it up with a G29 P2 H0.0 B0.1 M     It did everything right for me.  I've got a G26 P C O3.0 heating up the printer right now to check the Mesh.   

What you posted does look wrong.    If you can find the time, can you do it again, except add the M onto the P2 command so you can see the numbers it is generating as it is doing its work?   And...  If you don't mind spending a little time trying to help figure this out, what I did was a G29 P1 M to get the whole Mesh probed (except for the unreachable areas).  I then saved the Mesh in an out of the way slot so I could keep going back to it and trying different things with the G29 P2 command.

Roxy-3D, 2016-08-25:
@adamfilip I am now able to duplicate what you are seeing with the Z-Offset.    I'm working on figuring out where the problem is coming from.

adamfilip, 2016-08-26:
@Roxy-3D Great, Its nice to know it isnt just my incompetence at work :)
haven't had more time to play. also leaving tomorrow for a camping trip. be back in 8 days or so.
Good Luck! tracking down the issue.. im sure you will figure it out.

Roxy-3D, 2016-08-26:
> also leaving tomorrow for a camping trip. be back in 8 days or so.

Cool!  The bug will be gone by the time you are back.   It might be possible to have the Grid Based Leveling going by that time.  We will see.    But before I do the Grid Based, I'm thinking I should make it so the X_MIN and Y_MIN can be negative numbers.   I need to do that anyway way to get the Delta's going.    And it will be much easier to debug that on a Cartesian printer than on a Delta.

Enjoy your trip!

adamfilip, 2016-08-26:
Thanks..

BTW check out this new Gantry design im working on for my CoreXY machine and let me know your thoughts :)

http://www.thingiverse.com/thing:1737901

closed 2017-10-31 by thinkyhead