MarlinFirmware/Marlin #4639: Delta Z-axis resolution

zenmetsu, 2016-08-17:
I got to thinking about the fact that I was routinely getting 0.000000 for M48 probe reliability standard deviation.  I realized that one reason that this could happen, besides having really good hardware, is a low resolution in either position calculation, or actual physical movement.  I then realized that under Marlin, when I try to step up or down in the z-axis via pronterface in 0.1mm increments, the stepper motors only engage every two clicks.  Is there a reason for 0.2mm limitation on resolution?  I do not recall this ever happening under repetier.  This also explains why my 0.1mm layer height prints end up looking really bad, or failing entirely.

Is there a configuration option somewhere that I am overlooking?

thinkyhead, 2016-08-20:
Possibly this points to an issue with loss of resolution, but I presume that you do see movement of the correct distance as shown on the display (once it moves)?

Do you see also this issue if you reduce `dropsegments` to 0?

Roxy-3D, 2016-08-20:
> I got to thinking about the fact that I was routinely getting 0.000000 for M48 probe reliability standard deviation.   I realized that one reason that this could happen, besides having really good hardware, is a low resolution in either position calculation

Whoa!   I hope you are not right about this!   The M48 code should be able to use your:
```cpp
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,90}     
```
to identify what the minimum quantum of Standard Deviation is possible.    I'm concerned this could happen.    I think we are OK,  but I'm going to have to look at the lowest layers of the probe code to verify that.      In different words, the M48 code is never trying to do such a small move the segment gets dropped.  But I'm going to verify that!!!

zenmetsu, 2016-08-21:
Yes, this fixes the problem.  Thanks.  I somehow overlooked this option in the advanced config file.  Feel free to close if this is a "works as intended" matter.

thinkyhead, 2016-08-22:
@zenmetsu I presume you're referring to `dropsegments`, yes?

closed 2016-08-22 by thinkyhead