MarlinFirmware/Marlin #4641: Disable extruder during travel.

timmit99, 2016-08-17:
I'm looking to see if there is a way to make it so that the extruder on my machine is diabled (ENABLE pin is high) during travel movements. I am using an extruder that only requires a 5V signal to begin operation and want to be able to control it with the ramps board.

Roxy-3D, 2016-08-17:
> I would like to be able to just have marlin turn a pin HIGH when the printer is supposed to extrude and LOW when it is to stop. 

Do you realize the extruder is treated as a 4th axis?   Just as you can move different amounts in X, Y & Z, the E axis can move varying amounts too.  Just turning it on and not saying how far to travel is probably not going to work very well.

zenmetsu, 2016-08-18:
Extrusion rate is dependent upon travel rate.  While a constant-speed extruder could work, you'd also have to ensure that the velocity of the hotend is unchanged, and you'd have to match the speed to the extrusion rate.  I am not even aware of a slicer which can figure this out.  In short, you might have a very legitimate reason to do this, but it would have to be due to some considerable deviation from normal design.

Roxy-3D, 2016-08-18:
> In short, you might have a very legitimate reason to do this, but it would have to be due to some considerable deviation from normal design.

As I was typing in my thoughts up above...  I was a little hesitant to press the Green Comment button because some street artist was doing just this.  He had his stepper motors hitched up to a pair of 2"x4" boards and he was extruding paint from a spray can.   He could graffiti the side of building in high resolution very quickly.   I think this was about 6 months ago.  It would be neat to see some pictures of the stuff he did.

timmit99, 2016-08-18:
Basically my extruder is an experimental jetting nozzle that is on or off. I am controlling the spray value with an inserted gcode on layer change. I really just need to make it so when a normal printer is SUPPOSED to be extruding a pin should be high as a trigger.

zenmetsu, 2016-08-18:
Ahh, that makes sense now.

I'm sure that similar functionality is needed for laser cutters or spindle tools.  The functionality might already be in Marlin, and if not, it could be implemented... Then instead of using an "extruder", you'd be using something akin to a laser which is on or off.

As I am planning on making a 6-axis mill, this will be of interest to me eventually, so I'll let the real devs chime in as to Marlin's capabilities.

zenmetsu, 2016-08-18:
It looks like you are needing `M3`, `M4`, and `M5` support.  These are absent from Marlin at the present time.  Instead of hacking up the current code to allow for an alternate extruder behavior, I think the devs will elect to implement those gcodes.

Blue-Marlin, 2016-08-18:
For the laser people we got a trick.
Use the e-direction-pin and modify:
planner.cpp ~ line 647
```diff
  #else
    if (dx < 0) SBI(db, X_AXIS);
    if (dy < 0) SBI(db, Y_AXIS);
    if (dz < 0) SBI(db, Z_AXIS);
  #endif
-  if (de < 0) SBI(db, E_AXIS);
+  if (de <= 0) SBI(db, E_AXIS);
  block->direction_bits = db;

  block->active_extruder = extruder;
```
This will include 0 into the negative direction, so any positive extrusion will have one pin state and 0 and negative directions (retracts) the other.

timmit99, 2016-08-18:
I am not seeing that line "if (de < 0) SBI(db, E_AXIS);" anywhere in my planner.cpp. or in ANY of the marlin files. I'm currently running Marlin 1.0.2 .

zenmetsu, 2016-08-18:
I just checked the 1.0.x branch here on github, it is present, but is line 674 not 647.

And I'm an idiot for not thinking about using the e direction pin to do this... lol.

Blue-Marlin, 2016-08-18:
```
#else
  if ((target[X_AXIS]-position[X_AXIS]) + (target[Y_AXIS]-position[Y_AXIS]) < 0)
  {
    block->direction_bits |= (1<<X_AXIS); 
  }
  if ((target[X_AXIS]-position[X_AXIS]) - (target[Y_AXIS]-position[Y_AXIS]) < 0)
  {
    block->direction_bits |= (1<<Y_AXIS); 
  }
#endif
  if (target[Z_AXIS] < position[Z_AXIS])
  {
    block->direction_bits |= (1<<Z_AXIS); 
  }
-  if (target[E_AXIS] < position[E_AXIS])
+  if (target[E_AXIS] <= position[E_AXIS])
  {
    block->direction_bits |= (1<<E_AXIS); 
  }

  block->active_extruder = extruder;
```

timmit99, 2016-08-18:
ok, i see that on the RC version but not in my version. I'll test it on a seperrate ramps board for now untill we can track down what will make it work on my 1.0.2

timmit99, 2016-08-18:
oh well as i was typing that Blue-Marlin commented with it. Thank you a ton! i really hope this works and would make my job so simple!

timmit99, 2016-08-18:
It seems to not do anything. If the extruder is moving forward the direction pin has 5v on it, backwards is 0v. However if it is stopped it seems to retain the previous state (Forward -> Stop) pin stays at 5v

zenmetsu, 2016-08-18:
Configuration.h: `#define DISABLE_E false // For all extruders`

Does changing that to `true` help at all?

timmit99, 2016-08-19:
Setting it to "true" works for manual moves (it disable sit after it stops moving) but if i do a quick "print" the motor stays enabled during travel moves. I e to make a python script that scans through the generated g-code files and finds single or groups of lines that have no "E" term and wraps them in a turn pin on/off lines. It generates the g-code properly but it seems to not work when i run the print.the pin doesn't seem to toggle even though i can toggle it manually with the M42 command.

thinkyhead, 2016-08-21:
@timmit99 Unfortunately for your slicer plan, G1 isn't executed right away, and Marlin continues to the next command as soon as the G1 move has been added to the queue. Meanwhile, every other command is executed as soon as it's read. The fan speed is one exception. When fan speed is changed in gcode, it goes into the planner queue and is changed in sync with the block. You could use a similar approach to add support for a pin-toggle that goes into the planner block queue.

thinkyhead, 2016-08-21:
Actually, there's a feature you could totally supplant to get what you want: `BARICUDA`. It's a paste extruder supported in the planner blocks. Simply replace (or duplicate) what's done for the `BARICUDA`, supply your own pins, and it should get you most of the way.

timmit99, 2016-10-05:
I was able to make a python script to postprocess the G-Code and look for lines where it transitions from travel to extrusion and vice versa. I have it inserting a fan-enable/disable G-code  but this means i cannot use my fan.
 If there was a  option to have a version of the M42 command that is changed in sync with the block like the fan does now. That would free up my fan pin as i would like to be bale to use it but it seems to be the best way to enable our custom extruder.

thinkyhead, 2016-10-06:
> It looks like you are needing M3, M4, and M5 support. These are absent from Marlin at the present time.

But soon to change. #4939

thinkyhead, 2016-10-06:
> If there was a option to have a version of the M42 command that is changed in sync with the block like the fan does now.

It would require extending the planner block to include more detailed information. This would add extra overhead to the planner and stepper functions for all moves, so it should be done as an optional feature. And perhaps it will be made up for by the fact that there would be no E stepper.

closed 2018-02-26 by thinkyhead