MarlinFirmware/Marlin #4646: Auto leveling only probes the front half of the bed (Delta)

robingantois, 2016-08-18:
I have a problem with auto bed leveling. It only probes the front half of the bed. 
Do you have any idea what I'm doing wrong?

zenmetsu, 2016-08-18:
Is your Configuration.h available anywhere?

Could you post the output of G29 V4 as well so we can see which points are being probed?  I have a suspicion that you are running into the same problem that I had.   I wrote a fix for it which is pending integration into the official codebase.

robingantois, 2016-08-18:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/425125/Configuration.txt)

[image]

zenmetsu, 2016-08-18:
The output shows that the probed points are centered on your build platform, but mainly in the center and nothing past 35mm in either axis is getting probed.

`#define MIN_PROBE_EDGE 50`   I think this defaults to 10.  I think it is preventing points from being probed which are closer together.

```
    #define DELTA_PROBABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)
    #define LEFT_PROBE_BED_POSITION -35
    #define RIGHT_PROBE_BED_POSITION 35
    #define FRONT_PROBE_BED_POSITION -35
    #define BACK_PROBE_BED_POSITION 35
```
These are normally set to `-DELTA_PRINTABLE_RADIUS` or `DELTA_PRINTABLE_RADIUS`.  This is why the probed points are being constrained to 35mm radius.

Were these changed for a specific reason?  If not, could you try looking at one of the example configurations for deltas and set the `XXXX_PROBE_BED_POSITION` values and `MIN_PROBE_EDGE` accordingly.

robingantois, 2016-08-18:
Thank you very much for your help.
if I make the adjustments as you say, he gives the following output:
"?Probe (F)ront position out of range"

zenmetsu, 2016-08-18:
`#define Y_PROBE_OFFSET_FROM_EXTRUDER 36   // Z probe to nozzle Y offset: -front +behind`

This is why.  Your extruder is 36mm from your nozzle.

Change this...
`#define DELTA_PROBABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)`
to
`#define DELTA_PROBABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 36)`

And give it another shot.

robingantois, 2016-08-18:
the same: "?Probe (F)ront position out of range"

zenmetsu, 2016-08-18:
So you have this now?

```
  #if ENABLED(AUTO_BED_LEVELING_GRID)

    // set the rectangle in which to probe
    #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 36)
    #define LEFT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
    #define RIGHT_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS
    #define FRONT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
    #define BACK_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS

    #define MIN_PROBE_EDGE 10 // The Z probe square sides can be no smaller than this.
```

That should work.  Maybe try bumping it to `#define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 37)`  If your probe's distance from the nozzle is larger than the difference between `DELTA_PROBEABLE_RADIUS` and `DELTA_PRINTABLE_RADIUS` it tends to generate that message.  In your case, your probe has no X offset, so it should just be equal to your Y offset in the `DELTA_PROBEABLE_RADIUS` calculation.  If this doesn't work by bumping it up to 37, I'll have to let one of the devs jump in.

I have a bit of a disagreement on how this part of the probing works because a person could have a small printing radius (due to a small heated bed), but an absolutely monstrous range of motion if they have a large delta.  Safe effector travel radius might be 150mm, but a small bed limits printing radius to 80mm.  In this case, an additional 30mm is removed from your probeable radius if your probe is 30mm offset from the nozzle, which means that although your effector could safely move to allow the probe to hit every point on your bed (80mm bed radius + 30mm offset probe = 110mm which is lower than the safe 150mm), the code will end up limiting you to a small 50mm radius of probe points.

robingantois, 2016-08-18:
hmmm, Still the same!

robingantois, 2016-08-18:
When I use :

    #define DELTA_PROBABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)
    #define LEFT_PROBE_BED_POSITION -35
    #define RIGHT_PROBE_BED_POSITION 35
    #define FRONT_PROBE_BED_POSITION -35
    #define BACK_PROBE_BED_POSITION 35

and i set:

#define Y_PROBE_OFFSET_FROM_EXTRUDER 0 // Z probe to nozzle Y offset: -front +behind

It is probing the bed completely!

???

thinkyhead, 2016-08-21:
> It is probing the bed completely!

That tends to work for all, but if your probe is 36mm to one side of the nozzle, then I expect your effector is getting very close to the towers on some of those probing moves. In the long run I'd like to add better logic to handle a probe offset, especially on Delta.

If your probe is truly 36mm directly behind the nozzle, then it can't reach the very front of the bed without really straining the mechanics. So my recommendation is to use something like this:
```cpp
#define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS)
#define LEFT_PROBE_BED_POSITION (-(DELTA_PROBEABLE_RADIUS) + 10)
#define RIGHT_PROBE_BED_POSITION (DELTA_PROBEABLE_RADIUS - 10)
#define FRONT_PROBE_BED_POSITION (-(DELTA_PROBEABLE_RADIUS) + 36 + 10)
#define BACK_PROBE_BED_POSITION (DELTA_PROBEABLE_RADIUS - 10)
```

Roxy-3D, 2016-08-21:
> In the long run I'd like to add better logic to handle probe offsets on Delta, but in the meantime…

Soon...  I'll be bring up UBL on a Delta.   It will handle this nicely....

zenmetsu, 2016-08-21:
Pardon my glaring ignorance.  UBL? Uber Bed Leveling? lol

Roxy-3D, 2016-08-22:
Unified Bed Leveling.  It will have the Best of each all available at the same time.

closed 2017-10-31 by thinkyhead