MarlinFirmware/Marlin #4647: UBL Z offset

OrbitAVI, 2016-08-18:
@Roxy-3D First I have to say thank you so much for UBL! I am printing so well with it. I've never been able to print multi-parts in a single print due to my warped bed. Now with UBL its working great! and is so much easier then manual mesh. 

OK so now for my question. I posted a similar one before, so please excuse me for opening a new one. In any case my z offset from extruder seem's to behave oddly.  My Z probe offset from extruder measured with a gage and fine tuned is -.9. I've been using that value for a while now with manual mesh. In any case with UBL that makes it much too close to the bed so I raised it a little, but I end up having to baby step it back down to -.9 once the print starts to get the first layer just right. I'm so confused why baby stepping it to the same original value would result in a different outcome than just setting it to -.9. 

Am I doing something wrong?

Roxy-3D, 2016-08-18:
> First I have to say thank you so much for UBL! I am printing so well with it. I've never been able to print multi-parts in a single print due to my warped bed. Now with UBL its working great! and is so much easier then manual mesh.

You are very welcome!   It is going to get a little bit of a speed kick next.  I need to do that to get it to run on Delta's.    

adamfilip also has seen this behavior.  I'm trying to get setup to make it happen and debug it.  Almost for sure there is a bug in the code causing this.  (UBL is 2000 lines of new code.  There are going to be a few issues to work through)

First...   Be sure you do a M502 and a M500 any time you change a Z_OFFSET_FROM_EXTRUDER number.  Otherwise, you will be loading the old value from EEPROM.   

Second...   Don't try to use a micrometer or ruler to decide how much adjustment is needed.   In theory, that should work.  But right now that is causing trouble.   I'm gearing up to find and fix this issue.  It could be caused by the Z-FADE-HEIGHT changing the numbers you enter as the nozzle goes up and down.  I don't know the cause yet.   But it has been seen by other people.     For right now, make 'most' of the change to your Configuration.h numbers, reflash the electronics, do a M502 and M500.     And see where you are.   See how much more you want to change the number.   Ideally, each change should be less than the previous change.     What I'm suggesting you do is "Don't try to fix the number in one attempt.  It is OK to zero in on the correct number.  It won't take more than 3 or 4 tries to get it exactly correct."

The problem I'm hunting down has to do with the Z-Offset  (Not the Z_OFFSET_FROM_EXTRUDER number).   If you get your Z_OFFSET_FROM_EXTRUDER number correct, I think the Z-Offset issue won't bother you and I'll have a fix for that soon enough.

OrbitAVI, 2016-08-18:
Excellent. I'll redo my z probe offset one more time in either case it's working great. Also would it be possible to get a STL for the verification print? I can't seem to get that right in G26. I'm using a volcano with larger nozzel and my settings are dialed in on my slicer.

kmbecker13, 2016-09-05:
@Roxy-3D when doing a G29 Z command, which offset is being set?  I've noticed that after doing a G29 P1 and then a G29 P2 B (which probes the center of my bed first) if i print immediately after i get great results (or close to it).  Once i do a G29 Z later on, things get weird and my offset changes - i'm assuming a forgot the M502 or M500 somewhere? 

Sorry to ask the same question, not sure if changing those numbers in the configuration.h file manually is different from using the G29 Z command.

Roxy-3D, 2016-09-05:
> Once i do a G29 Z later on, things get weird and my offset changes - i'm assuming a forgot the M502 or M500 somewhere?

I found a bug with respect to the Z_Offset.   The probe's Z_PROBE_OFFSET_FROM_EXTRUDER was being put into the UBL's State information (specifically the blm.state.z_offset variable. )   I have that fixed, but I'm not ready to upload the changes yet because I'm in the middle of merging in the higher speed mesh_buffer_line() routine.   In a couple of days, there will be a fixed version and you won't be having that problem.

As a work around, until the updated version shows up, you can do this:   
- make sure your Z_PROBE_OFFSET_FROM_EXTRUDER is exactly correct.  Use that number to adjust things.
- And set the z_offset to 0.0000

closed 2017-10-31 by thinkyhead