MarlinFirmware/Marlin #4648: Question: ABL and G1

apballard, 2016-08-18:
So I use ABL and it works well. When I print, as the extruder moves over the print bed, I see the z-axis compensating for the uneven surface.

So here is my question. If I, from the terminal, execute a G1, let's say from x=10,y=10 to x=290,y=190 using the command G1 X290 Y190, why does the z-axis not change? The stepper motors for the z- axis don't move.  Is this normal behavior when using ABL?

Roxy-3D, 2016-08-18:
> G1 X290 Y190

First, which ABL are you using?   But remember, in all but the latest UBL code, if you have done a G28 (or homed any axis), the ABL system is turned off.

apballard, 2016-08-18:
So I perform a normal G28/G29 then immediately after that I perform a G1 X290 Y190. So it's not that.  While printing, it's clear that it's working, but issuing the G1 command manually straight after the G28/G29 does not move the z-axis at all.

Um, I was not aware there were different ABL's?! What are they?

Roxy-3D, 2016-08-18:
Is it possible that moving to G1 X290 Y190 doesn't require any correction?   Your Bed Level Correction Matrix would help us understand that.

For Cartesian machines there is 3-Point, Grid-Based and Mesh bed leveling.   I'm working on the UBL (unified bed leveling) code that is off in a development branch.    The Delta's have a hybrid Mesh Grid scheme right now.

apballard, 2016-08-19:
Thanks Roxy. You 100% right. I was testing on the x-axis and there was no need for correction.  A long move on y-axis did produce the expected results.

Also thanks for clarifying about ABL. I'm using Grid-Based. In my mind I viewed the Auto part as the method it collected the offsets, not the fact that once collected it auto compensated for differences.

And more importantly, thanks for all your hard dev work!

closed 2016-08-19 by apballard