MarlinFirmware/Marlin #4653: Problem with BLTouch when VIKI2 is defined

bgort, 2016-08-19:
Recently I've added a BLTouch to my printer and it works beautifully, generally, but after adding a VIKI2 and defining `VIKI2` in the most current RCBugFix, recompiling (w/U8glib 1.19.1), and uploading, `G29` regularly fails.  Re-commenting `VIKI2` and reuploading restores 'normal'/working behavior.  The BLTouch and VIKI2 are connected and configured consistent with the instructions from the manufacturers except that the BLTouch is using pin 45 for the servo instead of pin 44, and it has its own dedicated 5V supply.  Both work normally separately, and I don't believe there's a hardware or other problem with either.

When `G29` fails, it seems as though the endstop trigger isn't detected, as the Z axis continues lower until the BLTouch goes into an error state (LED blinking).  When working normally, the axis stops and reverses direction after the BLTouch sense pin has traveled a certain distance (presumably until the hall effect sensor detects the motion of the pin, and triggers Z min).  I'm using a 3x3 ABL grid, and the failure randomly occurs at any/all probe points (I tested quite a few times).

I've recorded two nozzle-cam videos:  one showing normal `G28` -> `G29` behavior with `VIKI2` undefined, and the other showing the failure when `VIKI2` is defined.

Normal:  https://www.youtube.com/watch?v=0hhQDTnslaI

Failure (occurs at ~45s):  https://www.youtube.com/watch?v=CO_2BeMpbSQ

Is it possible that the VIKI2 is simply consuming too many uC cycles and that's interfering with detection of the endstop trigger?  Compiling with -O3 doesn't help, but it seems plausible, at least.  

Thoughts?

thinkyhead, 2016-08-21:
Since it sort-of works, I'm thinking there's some kind of electrical interference, not a software issue _per se_. The VIKI2 is an SPI graphical display, so it adds both computational overhead and noise to the circuitry. The BLTouch has a hall effect sensor, and these are quite sensitive. See if you can isolate either the display or the sensor from magnetic fields, and whether that helps.

bgort, 2016-08-21:
Thank you for the reply.

Hmm.  I've yet to see the BLTouch fail when doing a `G28` (it only fails on `G29`), and I've tried the `G29` with the VIKI2 physically disconnected but support still compiled in and it still fails; plus, the display is physically remote from the hall sensor (which as I understand it is behind an ATTiny).  That's why I thought it was likely a firmware issue.  Maybe I'll try a series of 10 or 20 `G28`s and a `M48` and see if it fails that way; `G29` fails very nearly every time - it's worked twice out of 15-20 trials.

Not sure how I'd isolate it further - already has its own supply (it fails when using Rambo-supplied power, too) and it has to plug into the board eventually.  Suppose I could try a ferrite bead/choke on the line, or something.  Before I do that, though, I think I'll put the oscilloscope on the pins and see what I can learn that way.

bgort, 2016-08-22:
Have just done the following trials:

With VIKI2 support compiled in, but with it physically disconnected:
- 10x `G28`s (double probing, so 20 probes total):  10 successes (all 20 probes)
- 10x `G29`s (9 single probes each):  10 successes (all 90 probes)
- 2x `M48 P50 X145 Y131 E`:  
  - First attempt failed (with a bed strike) after 30 or so probes?  I walked away because I expected it'd be successful and got bored, so I'm not sure where it failed, exactly.
  - Second attempt successful, stddev 0.015484
  - Third attempt successful, stddev 0.029504

With VIKI2 support compiled in, and VIKI2 connected:
- 10x `G28`s (double probing, so 20 probes total):  10 successes (all 20 probes)
- 10x `G29`s (9 single probes each):  10 successes (all 90 probes)
- 2x `M48 P50 X145 Y131 E`:  
  - First attempt successful, stddev 0.036918
  - Second attempt successful, stddev 0.019842

My 'normal' configuration, without VIKI2 support:
- 1x `G28`s (double probing, so 20 probes total):  successful
- 1x `G29`s (9 single probes each):  successful
- 2x `M48 P50 X145 Y131 E`:
    - First attempt successful, stddev 0.021626
    - Second attempt successful, stddev 0.024296

It doesn't appear that I'm still having a problem, really.  Just the one failure, which is probably acceptable given the number of probes.  Nothing has changed on the printer physically - wiring the same, etc. - the only difference is that I've updated to the latest RCBugFix.  Perhaps there was some change made there.  Hard to say.  If I see this occur again, I'll use the oscilloscope to look for noise.  It was -very- repeatable previously (as in, very nearly every time).

The only thing concerning at this point is that my probe repeatability isn't as good as I've seen previously; standard deviation has been 6-7um before updating (earlier today), now 15-40um.  I'm not sure if this is a hardware or software issue, but I suppose it's possible it's either.  I'll keep playing with things.  Maybe the bed strike threw something off.

thinkyhead, 2016-08-22:
The display component of Marlin is well-isolated from things like the G29 function, so I'm glad to hear the issue is gone with `RCBugFix`. I can't explain why it wouldn't work with an earlier version, but perhaps some pin mappings changed along the way. I know we did have an issue with the MOSI pin being initialized incorrectly, and that was only recently patched. However, that only applied when using a MAX6675 thermocouple. So, again, a mystery.

closed 2016-08-22 by thinkyhead