MarlinFirmware/Marlin #4654: Z Probe_Offset not seeming to work in RC7

timeshell, 2016-08-19:
Ever since upgrading to RC7, the Z axis doesn't seem to compensate using the probe_offset anymore.  Using the new BLTouch definition.  I haven't tried to configure without it yet.  Any known reason why this would be happening?

Roxy-3D, 2016-08-19:
This has been seen by a number of people and in different configurations.   The problem is real.

But...   If you have an LCD Panel and enable Z-Baby-Stepping, the Z-Offset can be adjusted.  So the bulk of the Z-Offset logic is correct.  However, there is some place where it is wrong and causing trouble.

thinkyhead, 2016-08-19:
I've just merged some changes that may produce a better result. So download `RCBugFix` and try it at your first opportunity. Be sure to enable `DEBUG_LEVELING_FEATURE` and use `M111 S255` to enable full logging before testing. If you encounter problems, post the log output here so we can see what's going on.

timeshell, 2016-08-20:
Still not working.
[RC-Z-Homing-Offset-Issue.txt](https://github.com/MarlinFirmware/Marlin/files/428368/RC-Z-Homing-Offset-Issue.txt)

boelle, 2016-08-20:
@timeshell  "If you encounter problems, post the log output here so we can see what's going on."

thinkyhead, 2016-08-21:
> Anything to do with this line in Marlin_Main.cpp?

No. Not unless you're on the cutting edge:
```cpp
#if ENABLED(SWITCHING_EXTRUDER)
  . . .
```

thinkyhead, 2016-08-21:
@timeshell What do you mean by "compensating"? Your G28 log looks fine to me.

timeshell, 2016-08-21:
G28 was fine.  I was in error on that.   The G29 is the problem and I didn't get the log for that yet.

I set the offset in the configuration.h to -0.45.  But as after uploading it, the LCD showed it as -0.75. I had to removed that 0.3 and then if showed on the LCD as I had set it.  But the g29 still didn't work properly when I removed it on RC7, so yes that wasnt the original problem.

Sorrym, I know Im not always clear in the way I express myself.

timeshell, 2016-08-21:
I had also edited my post before you responded.  I realised I was wrong.  It doesn't say that anymore.

thinkyhead, 2016-08-21:
@timeshell We have several duplicates of this question concerning the Z position after `G29`, and are already trying to isolate the issue and provide an intelligent solution. First test the current `RCBugFix` to determine the extent of the issue. You may want to disable `EEPROM` to avoid confusion about what your actual probe offset is. After you've tried `RCBugFix` please test this branch, which proposes one possible solution: https://github.com/thinkyhead/Marlin/tree/rc_final_z_correction

timeshell, 2016-08-21:
Thank you.  I'll advise of my results on Tuesday.

rogueqd, 2016-08-22:
I just installed an IR z-probe yesterday and encountered an issue similar to this.
Chaging the Z_OFFSET_FROM_EXTRUDER had no effect, but changing the z-offset via the LCD had the desired effect.

rogueqd, 2016-08-22:
I'll also try the z_correction update when I can, but it will probably be a couple of days.

thinkyhead, 2016-08-22:
@rogueqd You don't just happen to have `EEPROM_SETTINGS` enabled, do you? If you do, you need to use `M502` to reload your defaults, or after you change the offset using the LCD or M851, you must save it to EEPROM using `M500`. If you don't do one of these, you will simply get the last value that was stored in EEPROM.

rogueqd, 2016-08-23:
I've changed the z-offset in EEPROM to 0.0 and saved it, just to remove it from any future testing.
I just tried the rc_final_z_correction firmware, it got some of the way through the G29 and the steppers started skipping. It was working fine only a few minutes before with older firmware (rc_bug_fix from a week or two ago). I tested twice, the skipping was different in each test but neither G29 completed successfully.

Roxy-3D, 2016-08-23:
Nothing has changed in the stepping motor code for a while now???

thinkyhead, 2016-08-23:
Which of your steppers is skipping? All of them, or just the Z?

rogueqd, 2016-08-23:
It's a delta. I didn't really pay attention to which individual motors, but the head seemed to loose is position along the X axis (not X tower). That's ok. It's not really relevant to this bug, just prevents me from testing at the moment. The acceleration values haven't changed, I just copied over the same config file I've used for all my 1.10 testing, except I had to update the name of the prevent_cold_extrusion define.

thinkyhead, 2016-08-23:
If you enter some "extreme" value in `M851`, such as "-50" do you see any effect in the result?

thinkyhead, 2016-08-24:
> seemed to lose its position along the X axis

Hmm. That can't just be related to a single stepper, so you're implying a fundamental flaw in the tracking of position.

Frostyybeer, 2016-08-25:
I am having a similar issue where my z_probe_offset would not compensate after sending a G28 command. I tried both the RCBugFix and the rc_final_z_correction branches without any success. If you could please take a look at my [log.txt](https://github.com/MarlinFirmware/Marlin/files/436180/log.txt) and my firmware setup ([Marlin-RC7_BugFix.zip](https://github.com/MarlinFirmware/Marlin/files/436183/Marlin-RC7_BugFix.zip)) that would be much appreciated. 

Thank you in advance for your time.

- Nate

rogueqd, 2016-08-25:
Ok, so I found a mechanical fault with my printer this morning. Please disregard my earlier comments about skipping.

cruzpablo, 2016-08-26:
I have same problem. Any solution ? After write marlin-rc firmware working fine 5 days, and return this problem

thinkyhead, 2016-08-26:
@Frostyybeer Is your probe connected to the same pin as your z min endstop? If so, you should use `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` instead of `Z_MIN_PROBE_ENDSTOP`, because with the latter option Marlin assumes you are homing Z with your endstop, and not with your probe. You can see this at line 1568 in `Marlin_main.cpp`:
```diff
    #if HAS_BED_PROBE && Z_HOME_DIR < 0 && DISABLED(Z_MIN_PROBE_ENDSTOP)
```

thinkyhead, 2016-08-26:
@AnHardt @Roxy-3D I'm not fond of this "side-effect," but at least it is documented.

AnHardt, 2016-08-27:
@thinkyhead 
You already know - i'm not a friend of the current naming. Having discussed this for over a year now, i have no further helpful contributions about that.

If we home with a z-min-endstop - that's not a probe - there is no probe-offset - that's logical.
The logical way out, are endstop offsets - also for Cartesian systems. 

Q: Why is z-probe offstet different between G28 and G29?
A: Because it's designed that way.
Assuming you have a system, able to do bed-leveling, there is no reason not to use bed leveling.
Why not simply using the 'd' result? Because 'd' is the average level for the measured points. At that level we are scratching the highest parts of the bed. Using a z-probe offset, including the d-correction, we have to find by experiment is not more inconvenient, then if the d-correction would not be included.
(As far as i remember, that was state of the discussions a ~9-12 month ago when a PR about this was rejected (ether here or in Dev))

Frostyybeer, 2016-08-27:
@thinkyhead I am using a separate z probe for porbing (assigned it to pin 57 in pin.h) and z-min endstop for homing. I test the signals with M119 and all endstops are working correctly as well as the inductive probe. Could it be that I have to define/undefine something else in order for my configuration to work properly? I am slowly loosing my hair over this issue... Thank you for your reply.

- Nate

thinkyhead, 2016-08-27:
@Frostyybeer You wrote:
>  my z_probe_offset would not compensate after sending a G28 command.

If you're using your Z min endstop to home, then no offset should be applied with `G28`.

Frostyybeer, 2016-08-27:
@thinkyhead So if my nozzle is too high above the print bed when print start (after bed-leveling with G29) what could be causing the problem?? I have tried adjusting the z-probe offset in the firmware but my nozzle still stay at the same height.

- Nate

thinkyhead, 2016-08-27:
What does `M503 S0` say? Have you set an offset using `M206` by any chance?

Frostyybeer, 2016-08-27:
@thinkyhead Let say I have more that 10 printers, is there a way to set it permanently in the firmware without having to put M206 at the start of every print?

thinkyhead, 2016-08-27:
`M206` is irrelevant to your issue. You shouldn't need it.

Frostyybeer, 2016-08-27:
In simplify3D for the Starting script I only have G28; home all axis follow by G29; auto bed leveling. My first layer height is set to 130% of 0.3mm (= 0.39mm). The LCD screen seem to be showing the correct value of first layer z height when printing. However, the actual height is completely off by more than 2.0mm

- Nate

thinkyhead, 2016-08-27:
So, what is the result if G29 is left off?

Frostyybeer, 2016-08-27:
When printing without G29 the actual height seem to be very close to the 0.39mm.

thinkyhead, 2016-08-27:
Can you post a log of the `G29` output with `DEBUG_LEVELING_FEATURE` enabled, and after `M111 S255`? I'd like to see whether your M851 offset is actually accurate, or if it could be off by ~2mm.

Frostyybeer, 2016-08-27:
[Debug Leveling Feature log output.txt](https://github.com/MarlinFirmware/Marlin/files/440530/Debug.Leveling.Feature.log.output.txt)

Roxy-3D, 2016-08-27:
> Why not simply using the 'd' result? Because 'd' is the average level for the measured points. 

There is one issue with just using d.   Remember...  The plane that was just calculated and described by ax+by+cz=d is offset from the nozzle.   The Probe is offset by X_OFFSET_FROM_EXTRUDER and Y_OFFSET_FROM_EXTRUDER.   If the plane is perfectly level, it won't matter.  But with any tilt at all, as those two numbers get bigger, they start factoring in an extra Z correction number.

> At that level we are scratching the highest parts of the bed. Using a z-probe offset, including the d-correction, we have to find by experiment is not more inconvenient, then if the d-correction would not be included.

With the ABL correction turned on, if we are scratching the highest parts of the bed I think you can conclude that the bed is warped by more than the first layer height.   Right?

AnHardt, 2016-08-27:
> There is one issue with just using d. Remember... The plane that was just calculated and described by ax+by+cz=d is offset from the nozzle. The Probe is offset by X_OFFSET_FROM_EXTRUDER and Y_OFFSET_FROM_EXTRUDER. If the plane is perfectly level, it won't matter. But with any tilt at all, as those two numbers get bigger, they start factoring in an extra Z correction number. The reason is we care about the nozzle's height, and not the probe.

Complete crap! See: https://github.com/MarlinFirmware/Marlin/pull/4670#issuecomment-242889972

> With the ABL correction turned on, if we are scratching the highest parts of the bed I think you can conclude that the bed is warped by more than the first layer height. Right?

Yes. If it would still scratch a 'first layer height' the bed would be really bad warped. But that was not my point. My point was, if we only add 'd', half of the bed will be scratched at zero level - so 'd' is not the right correction because it's to deep.

Roxy-3D, 2016-08-27:
>> There is one issue with just using d. Remember... The plane that was just calculated and described by ax+by+cz=d is offset from the nozzle. The Probe is offset by X_OFFSET_FROM_EXTRUDER and Y_OFFSET_FROM_EXTRUDER. If the plane is perfectly level, it won't matter. But with any tilt at all, as those two numbers get bigger, they start factoring in an extra Z correction number. The reason is we care about the nozzle's height, and not the probe.
> Complete crap! See: #4670 (comment)

Wow!   I think I'll just stay out of any discussion where you are 'contributing'.

AnHardt, 2016-08-27:
@Roxy-3D 
I would not be so harsh if i hadn't wrote the paragraph about x/y-offsets in https://github.com/MarlinFirmware/Marlin/pull/4670#issuecomment-242889972 extra for you.

Roxy-3D, 2016-08-27:
I read that twice.   Even if you are correct, people that are trying hard to understand a difficult topic deserve some patients.  

If I remember correctly, the original ABL probing did not compensate for the Nozzle's offset.   In the original ABL, it was very important to add in a small amount of correction for how much the plane shifted its Z Position moving from the probe to nozzle.      It maybe that because the probed positions are now 'corrected' to operate in the Nozzle's coordinate system that doesn't need to be done.

thinkyhead, 2016-08-28:
We all are surely a little Aspergers. Some a bit more than others.

thinkyhead, 2016-08-28:
@Frostyybeer Am I correct that your `Z_PROBE_OFFSET_FROM_EXTRUDER` (aka `M851`) is set to -4.3 ?

If you've been testing with `RCBugFix` (and I hope you have) then please re-download and try again. I accidentally left in a redundant adjustment in the homing of Z when homing with a probe.

And then, please test one or both of my branches, which are attempting to come up with solutions.

This one adjusts Z based on the homing position:
https://github.com/thinkyhead/Marlin/tree/rc_fix_leveling_maths

And this one adjusts Z based on the final probe position:
https://github.com/thinkyhead/Marlin/tree/rc_final_z_correction

Frostyybeer, 2016-08-28:
@thinkyhead Correct, I set the Z_PROBE_OFFSET_FROM_EXTRUDER in the original **RCBugFix** firmware to -4.3 just to see if that would do anything to adjust my nozzle height after ABL. (It does not)

When trying the **rc_fix_leveling_maths** after ABL my nozzle is crashing to my print bed (differ than previous issue where the nozzle is too far above the print bed). However, when trying to adjust the Z_PROBE_OFFSET_FROM_EXTRUDER the same problem occur where my nozzle still crash to the print bed regardless of whatever number I put. 

As for the **rc_final_z_correction**, I got this error when flashing firmware.
[Error.txt](https://github.com/MarlinFirmware/Marlin/files/440980/Error.txt)

- Nate

Blue-Marlin, 2016-08-28:
```
// Z Probe to nozzle (X,Y) offset, relative to (0, 0).
// X and Y offsets must be integers. !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
//
// In the following example the X and Y offsets are both positive:
// #define X_PROBE_OFFSET_FROM_EXTRUDER 10
// #define Y_PROBE_OFFSET_FROM_EXTRUDER 10
//
//    +-- BACK ---+
//    |           |
//  L |    (+) P  | R <-- probe (20,20)
//  E |           | I
//  F | (-) N (+) | G <-- nozzle (10,10)
//  T |           | H
//    |    (-)    | T
//    |           |
//    O-- FRONT --+
//  (0,0)
#define X_PROBE_OFFSET_FROM_EXTRUDER 0  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 50  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.21   // Z offset: -below +above  [the nozzle]
```

Frostyybeer, 2016-08-28:
@Blue-Marlin I should have read that "// X and Y offsets must be integers." 
Thank you!

- Nate

Frostyybeer, 2016-08-28:
@thinkyhead 

The [**rc_z_final_z_correction**](https://github.com/thinkyhead/Marlin/tree/rc_final_z_correction) **`solved`** my problem perfectly. The ABL-(Auto Bed Leveling) is now working like it supposed to.

I can adjust the Z_PORBE_OFFSET_FROM_EXTRUDER in the firmware and my nozzle will react according to the value I input. For example, by **decreasing** the number from -1.0mm to -2.0mm will put my nozzle **closer** to my print bed by 1.0mm after G29.

Cheers!
- Nate

thinkyhead, 2016-08-28:
I'm gratified to hear that the **rc_final_z_correction** branch worked. Unfortunately we're most likely not going to be able to use it, because it is highly idealized for a certain kind of setup. I looked more deeply into bed leveling issues for [rc_fix_leveling_maths](https://github.com/thinkyhead/Marlin/tree/rc_fix_leveling_maths) and that is probably going to end up being merged if it passes muster. So you should try it and let us know if it also produces better results.

timeshell, 2016-08-30:
I'm sorry I haven't been responding.  Waiting for a part for my printer.  I'll be following up soon too.

thinkyhead, 2016-08-30:
@timeshell That's okay. I'm looking at yet another refinement where we may use the probed points along with the probe Z offset to perform a height correction, and also use the center of the probe region as the fulcrum for the tilted plane. Combining these two method should lead to the smallest error and the most accurate distance to the bed, presuming that the `M851` offset is set accurately and the probe is reliable.

Frostyybeer, 2016-08-31:
@thinkyhead I will keep on testing the RCBugfix and rc_fix_leveling_maths branches once I have my 3Dprinter available again (just recently moved back to college.)

boelle, 2016-09-10:
@Frostyybeer Got your printer back or will it be a while yet?
@timeshell had time to revisit this subject?

timeshell, 2016-09-10:
Not directly, but have discovered a couple things working with a client.  I have a couple things to test which might narrow it down.  I'll get to it in the next few days.

timeshell, 2016-09-11:
Ok.  Here is where I am at.  I managed to get the offset to work in RC7.

I have two copies of the RC7.  One with a config that doesn't work and one with a config that does.  It seems to be specifically related to something happening within the configuration.h file.  I am going to compare these two configurations and see if I can find the culprit.

timeshell, 2016-09-12:
In RC7 Marlin_main.cpp line 3059-3061:
            destination[X_AXIS] = round(Z_SAFE_HOMING_X_POINT - (X_PROBE_OFFSET_FROM_EXTRUDER));
            destination[Y_AXIS] = round(Z_SAFE_HOMING_Y_POINT - (Y_PROBE_OFFSET_FROM_EXTRUDER));
            destination[Z_AXIS] = current_position[Z_AXIS]; // Z is already at the right height

Why is the offset not being applied to Z with safe homing enabled?  It says Z is already at the right height but I'm not so sure that it is.  And the reason I ask is Z_SAFE_HOMING is enabled in the one that is not working and is not enabled in the one that is.

thinkyhead, 2016-09-12:
What offset did you have in mind? When Z safe homing is done, it happens after homing X and Y. If there's a setting for `Z_HOMING_HEIGHT` that raise occurs _before_ homing X or Y. So by the time the blocking move to the safe homing XY occurs, the Z axis will be off the bed.

thinkyhead, 2016-09-12:
I'm actually patching Z safe homing a little at the moment. It can be used when there isn't a probe, so it needs a tweak in consideration of that.

timeshell, 2016-09-12:
The extruder offset?  Same as the X/Y in that block of code?  The Z_SAFE_HOMING was the only significant difference between the two configurations.  I can email them to you if you'd like to compare.

timeshell, 2016-09-12:
Ok.  Here's a question.  If that code is assuming that the user isn't using a probe, why are the probe offsets being applied to X and Y then?  Just curious now.

thinkyhead, 2016-09-12:
The code assumes the user _is using a probe_.

[5 more replies]

closed 2016-11-02 by Roxy-3D