MarlinFirmware/Marlin #4657: First print and Resend: errors

ruggb, 2016-08-20:
This is the first print I am running with RC7-62fd176 and I am getting a lot of resend errors. There are no other error types. I have NEVER had this and I am running at 30mm/sec = VERY slow. Baud rate is 250000, CoreXY. Same config options as RC6. It happens at certain spots in the print. Long straight runs are fine. Lots of slow movement is not happy. Maybe this is better in that the print is fine because it is not missing steps. So why is it resending? Maybe if I print faster it will run better.

RC6 ran with no errors at this speed. I had to get it to at least 50mm/sec, then I got errors that indicated missed steps - like wrong step or wrong response - I forget exactly. Sometimes the print was messed up - obviously missed steps. Are we going sideways?

Roxy-3D, 2016-08-20:
Can we have some more information about your hardware?  On both the computer side and the printer side?   What Host Program are you using?   

We are pretty far into RC-7 and this isn't a common issue.  We are going to need some help to figure out what is sick with RC-7 on your printer.

ruggb, 2016-08-20:
I am printing a slightly different version of the previous print with different Slic3r settings and I have not rcvd any errors yet. I believe the previous print was not sliced with the config that I selected - I have no idea why and no idea what config it used. Can you tell me what might cause resend errors relative to slicer settings so I can see if I can nail it down? My system is;
CoreXY printer
Repetier 1.6.2
Slic3r 1.2.9 stable
Arduino 1.6.9
Mega 2560
Ramps 1.4
DRV8825 drivers
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Automation Technology Inc. Stepper KL17H248-15-4A
12V 40A PSU

thinkyhead, 2016-08-21:
Either the host was sending odd bytes or the USB/Serial connection was being flaky. Try using a lower baud rate such as 115200 or 57600 and see if you still get errors.

ruggb, 2016-08-21:
Whatever changed was relative to the config of the slicer so it has nothing to do with the serial interface or baud rate. It stopped when it sliced it per the config I selected. The changes to the print were not significant. Since I have no idea what the config was (which is another issue) I guess I will close this till it reoccures.

closed 2016-08-21 by ruggb