MarlinFirmware/Marlin #4662: Auto bed leveling issues

markovicb85, 2016-08-20:
Hi, first of all, you boys doing a excellent job. 
I need help with auto bed leveling calibration. When it was manually calibrated, printer works great, the print quality of first layer is excellent. The trouble starting after finishing ABL (G28 then G29), the printer front part is high (maybe 0.5mm), and back part is to low (nozzle relies on bed). I'm talking about Y direction.
Maybe this going to be more helpful:
 
21:02:45.030 : G29 Auto Bed Leveling
21:02:53.817 : Bed X: 23.000 Y: -42.000 Z: 1.890
21:02:54.630 : Bed X: 10.000 Y: -42.000 Z: 1.730
21:02:55.396 : Bed X: -3.000 Y: -42.000 Z: 1.710
21:02:56.161 : Bed X: -16.000 Y: -42.000 Z: 1.710
21:02:56.896 : Bed X: -29.000 Y: -42.000 Z: 1.800
21:02:57.717 : Bed X: -42.000 Y: -29.000 Z: 1.820
21:02:58.483 : Bed X: -29.000 Y: -29.000 Z: 1.710
21:02:59.233 : Bed X: -16.000 Y: -29.000 Z: 1.630
21:02:59.998 : Bed X: -3.000 Y: -29.000 Z: 1.630
21:03:00.764 : Bed X: 10.000 Y: -29.000 Z: 1.670
21:03:01.561 : Bed X: 23.000 Y: -29.000 Z: 1.720
21:03:02.405 : Bed X: 36.000 Y: -29.000 Z: 1.740
21:03:03.373 : Bed X: 49.000 Y: -16.000 Z: 1.600
21:03:04.311 : Bed X: 36.000 Y: -16.000 Z: 1.500
21:03:05.155 : Bed X: 23.000 Y: -16.000 Z: 1.520
21:03:05.952 : Bed X: 10.000 Y: -16.000 Z: 1.490
21:03:06.733 : Bed X: -3.000 Y: -16.000 Z: 1.440
21:03:07.498 : Bed X: -16.000 Y: -16.000 Z: 1.430
21:03:08.248 : Bed X: -29.000 Y: -16.000 Z: 1.480
21:03:08.990 : Bed X: -42.000 Y: -16.000 Z: 1.560
21:03:09.769 : Bed X: -42.000 Y: -3.000 Z: 1.410
21:03:10.527 : Bed X: -29.000 Y: -3.000 Z: 1.380
21:03:11.277 : Bed X: -16.000 Y: -3.000 Z: 1.410
21:03:12.043 : Bed X: -3.000 Y: -3.000 Z: 1.470
21:03:12.822 : Bed X: 10.000 Y: -3.000 Z: 1.540
21:03:13.617 : Bed X: 23.000 Y: -3.000 Z: 1.590
21:03:14.484 : Bed X: 36.000 Y: -3.000 Z: 1.570
21:03:15.409 : Bed X: 49.000 Y: -3.000 Z: 1.490
21:03:16.229 : Bed X: 49.000 Y: 10.000 Z: 1.300
21:03:17.171 : Bed X: 36.000 Y: 10.000 Z: 1.320
21:03:18.032 : Bed X: 23.000 Y: 10.000 Z: 1.370
21:03:18.850 : Bed X: 10.000 Y: 10.000 Z: 1.360
21:03:19.658 : Bed X: -3.000 Y: 10.000 Z: 1.320
21:03:20.435 : Bed X: -16.000 Y: 10.000 Z: 1.260
21:03:21.201 : Bed X: -29.000 Y: 10.000 Z: 1.220
21:03:21.963 : Bed X: -42.000 Y: 10.000 Z: 1.210
21:03:22.728 : Bed X: -42.000 Y: 23.000 Z: 1.180
21:03:23.496 : Bed X: -29.000 Y: 23.000 Z: 1.190
21:03:24.265 : Bed X: -16.000 Y: 23.000 Z: 1.240
21:03:25.032 : Bed X: -3.000 Y: 23.000 Z: 1.320
21:03:25.838 : Bed X: 10.000 Y: 23.000 Z: 1.350
21:03:26.663 : Bed X: 23.000 Y: 23.000 Z: 1.330
21:03:27.571 : Bed X: 36.000 Y: 23.000 Z: 1.270
21:03:28.562 : Bed X: 49.000 Y: 23.000 Z: 1.170
21:03:29.425 : Bed X: 36.000 Y: 36.000 Z: 1.140
21:03:30.350 : Bed X: 23.000 Y: 36.000 Z: 1.180
21:03:31.198 : Bed X: 10.000 Y: 36.000 Z: 1.210
21:03:32.028 : Bed X: -3.000 Y: 36.000 Z: 1.190
21:03:32.813 : Bed X: -16.000 Y: 36.000 Z: 1.160
21:03:33.596 : Bed X: -29.000 Y: 36.000 Z: 1.130
21:03:34.490 : Bed X: -16.000 Y: 49.000 Z: 1.240
21:03:35.288 : Bed X: -3.000 Y: 49.000 Z: 1.290
21:03:36.129 : Bed X: 10.000 Y: 49.000 Z: 1.330
21:03:37.012 : Bed X: 23.000 Y: 49.000 Z: 1.340
21:03:37.012 : 0.27 -0.03 -0.21 -0.21 -0.21 -0.21 -0.17 0.11 0.24
21:03:37.012 : 0.16 -0.01 -0.20 -0.29 -0.29 -0.27 -0.11 -0.02 -0.04
21:03:37.012 : -0.16 -0.18 -0.29 -0.37 -0.37 -0.33 -0.28 -0.26 -0.29
21:03:37.028 : -0.36 -0.44 -0.52 -0.57 -0.56 -0.51 -0.48 -0.50 -0.40
21:03:37.028 : -0.56 -0.59 -0.62 -0.59 -0.53 -0.46 -0.41 -0.43 -0.51
21:03:37.028 : -0.80 -0.79 -0.78 -0.74 -0.68 -0.64 -0.63 -0.68 -0.70
21:03:37.028 : -0.96 -0.82 -0.81 -0.76 -0.68 -0.65 -0.67 -0.73 -0.83
21:03:37.028 : -0.89 -0.88 -0.87 -0.84 -0.81 -0.79 -0.82 -0.86 -0.90
21:03:37.028 : -0.95 -0.98 -0.93 -0.76 -0.71 -0.67 -0.66 -0.99 -1.05
21:03:38.359 : X:53.00 Y:65.00 Z:51.34 E:0.00 Count X: 14568 Y:22622 Z:25815

My printer is delta kossel mini, and firmware version is 1.1.0 RC 6.  The bed is made of glass.

markovicb85, 2016-08-23:
Any ideas...?

Blue-Marlin, 2016-08-23:
As long as you are referring to RC6 it will be hard to find one who wants to have a look into it. Please don't expect us to solve every problem more then once. Your chances increase dramatically if you have tested with RCBugFix.

markovicb85, 2016-08-23:
Sorry I'm new on this forum, and also new in programming and 3d printing. Before posted this issue, I tried to find solution over internet but without luck.
I thought RC6 is stable version (don't know way), but if you recommended RCBugFix I'll try.

Roxy-3D, 2016-08-23:
> I thought RC6 is stable version (don't know way), but if you recommended RCBugFix I'll try.

What we are doing is a little bit out of the ordinary.  RC stands for Release Candidate.  It is not a stable version in that sense.  We are trying to see if we find any major bugs in the Release Candidate, and if we do, the bugs get fixed.   We are on RC-7 and we are pretty sure there will be a RC-8.  RCBugFix is the most 'correct' version of the RC-7 that we have.   

So even though the code is changing...  RCBugFix is the most trusted and 'correct' version of the code right now.  That will also be true if we start a RC-8 version.

thinkyhead, 2016-08-24:
> don't expect us to solve every problem
> Sorry I'm new on this forum, and also new in programming and 3d printing.

Of course. I'm sorry that some around here are so quick to be condescending. As Roxy explains, we're putting out these release candidates periodically as we reach a point where public testing is needed, but the very latest code is in `RCBugFix` and so we ask everyone who reports an issue to test it so we can see if we already fixed the issue.

thinkyhead, 2016-08-24:
Anyway, once you get `RCBugFix`, if you still experience problems, there's a procedure that you can follow to help us collect data.  First enable `DEBUG_LEVELING_FEATURE` and flash with that enabled. Before you use `G28` and `G29` execute `M111 S255` to turn on all logging. Finally, when you do `G29` do it as `G29 V4` to turn on the most verbose output.

Good luck! I look forward to seeing your results.

markovicb85, 2016-08-24:
OK, thanks for you advice. I'm uploaded  RCBugFix. Nothing changed for me, I still have the same issues (manual level is perfect, ABL nozzle relies on back side of bed). 
I noticed when using M666 Z-0.4 M500 to save, end M501 to show new value, it's change in log window, but hardware is unchanged. 
Tanks for your's support.

markovicb85, 2016-08-24:
I executed M111 S255, then G29 V4 here is result:

00:09:09.640 : echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
00:09:11.452 : echo:N22 M105*23
00:09:14.521 : echo:N23 M105*22
00:09:17.583 : echo:N24 M105*17
00:09:20.657 : echo:N25 M105*16
00:09:20.814 : echo:N26 G29 V4*100
00:09:20.814 : >>> gcode_G29
00:09:20.814 : current_position=(0.00, 0.00, 163.62) :
00:09:20.814 : G29 Auto Bed Leveling
00:09:20.814 : reset_bed_level
00:09:20.814 : current_position=(0.00, 0.00, 163.62) : setup_for_endstop_or_probe_move
00:09:20.814 : current_position=(0.00, 0.00, 163.62) : set_probe_deployed
00:09:20.814 : deploy: 1
00:09:20.814 : do_probe_raise(50.00)
00:09:20.829 : >>> do_blocking_move_to(0.00, 0.00, 163.62)
00:09:20.829 : destination=(0.00, 0.00, 163.62) : set_destination_to_current
00:09:20.829 : current_position=(0.00, 0.00, 163.62) : xy move
00:09:20.829 : <<< do_blocking_move_to
00:09:20.829 : >>> probe_pt(23.00, -42.00, no stow)
00:09:20.829 : current_position=(0.00, 0.00, 163.62) :
00:09:20.829 : do_probe_raise(5.00)
00:09:20.845 : > do_blocking_move_to_xy(53.00, -27.00)
00:09:20.845 : >>> do_blocking_move_to(53.00, -27.00, 163.62)
00:09:20.845 : destination=(0.00, 0.00, 163.62) : set_destination_to_current
00:09:21.782 : current_position=(53.00, -27.00, 163.62) : xy move
00:09:21.782 : <<< do_blocking_move_to
00:09:21.845 : > current_position=(53.00, -27.00, 163.62) : set_probe_deployed
00:09:21.845 : deploy: 1
00:09:21.845 : current_position=(53.00, -27.00, 163.62) : >>> run_z_probe
00:09:21.845 : >>> do_blocking_move_to(53.00, -27.00, 4.80)
00:09:21.845 : destination=(53.00, -27.00, 163.62) : set_destination_to_current
00:09:21.845 : current_position=(53.00, -27.00, 163.62) : xy move
00:09:21.845 : destination=(53.00, -27.00, 4.80) : prepare_move_to_destination_raw
00:09:21.845 : current_position=(53.00, -27.00, 4.80) : z lower move
00:09:21.845 : <<< do_blocking_move_to
00:09:25.311 : current_position=(53.00, -27.00, 4.80) : >>> do_probe_move
00:09:25.311 : >>> do_blocking_move_to(53.00, -27.00, -10.20)
00:09:25.311 : destination=(53.00, -27.00, 4.80) : set_destination_to_current
00:09:25.326 : current_position=(53.00, -27.00, 4.80) : xy move
00:09:25.326 : destination=(53.00, -27.00, -10.20) : prepare_move_to_destination_raw
00:09:25.326 : current_position=(53.00, -27.00, -10.20) : z lower move
00:09:25.326 : <<< do_blocking_move_to
00:09:25.467 : current_position=(53.00, -27.00, 1.74) : sync_plan_position_delta
00:09:25.467 : current_position=(53.00, -27.00, 1.74) : <<< do_probe_move
00:09:25.467 : current_position=(53.00, -27.00, 1.74) : <<< run_z_probe
00:09:25.467 : > do_probe_raise
00:09:25.467 : do_probe_raise(5.00)
00:09:25.467 : >>> do_blocking_move_to(53.00, -27.00, 4.80)
00:09:25.467 : destination=(53.00, -27.00, 1.74) : set_destination_to_current
00:09:25.467 : destination=(53.00, -27.00, 4.80) : prepare_move_to_destination_raw
00:09:25.483 : current_position=(53.00, -27.00, 4.80) : z raise move
00:09:25.483 : current_position=(53.00, -27.00, 4.80) : xy move
00:09:25.483 : <<< do_blocking_move_to
00:09:25.576 : Bed X: 23.000 Y: -42.000 Z: 1.739
00:09:25.576 : <<< probe_pt
00:09:25.576 : >>> probe_pt(10.00, -42.00, no stow)
00:09:25.576 : current_position=(53.00, -27.00, 4.80) :
00:09:25.576 : do_probe_raise(5.00)
00:09:25.576 : > do_blocking_move_to_xy(40.00, -27.00)
00:09:25.576 : >>> do_blocking_move_to(40.00, -27.00, 4.80)
00:09:25.576 : destination=(53.00, -27.00, 4.80) : set_destination_to_current
00:09:25.795 : current_position=(40.00, -27.00, 4.80) : xy move
00:09:25.795 : <<< do_blocking_move_to
00:09:25.842 : > current_position=(40.00, -27.00, 4.80) : set_probe_deployed
00:09:25.842 : deploy: 1
00:09:25.842 : current_position=(40.00, -27.00, 4.80) : >>> run_z_probe
00:09:25.842 : >>> do_blocking_move_to(40.00, -27.00, 4.80)
00:09:25.842 : destination=(40.00, -27.00, 4.80) : set_destination_to_current
[... 1843 lines cut ...]
00:09:56.318 : deploy: 0
00:09:56.318 : do_probe_raise(50.00)
00:09:56.318 : >>> do_blocking_move_to(53.00, 64.00, 49.80)
00:09:56.334 : destination=(53.00, 64.00, 4.80) : set_destination_to_current
00:09:56.334 : destination=(53.00, 64.00, 49.80) : prepare_move_to_destination_raw
00:09:56.334 : current_position=(53.00, 64.00, 49.80) : z raise move
00:09:56.334 : current_position=(53.00, 64.00, 49.80) : xy move
00:09:56.334 : <<< do_blocking_move_to
00:09:57.515 : >>> do_blocking_move_to(53.00, 64.00, 49.80)
00:09:57.530 : destination=(53.00, 64.00, 49.80) : set_destination_to_current
00:09:57.530 : current_position=(53.00, 64.00, 49.80) : xy move
00:09:57.530 : <<< do_blocking_move_to
00:09:57.530 : current_position=(53.00, 64.00, 49.80) : clean_up_after_endstop_or_probe_move
00:09:57.530 : current_position=(53.00, 64.00, 49.80) : > probing complete
00:09:57.530 : 4.39 3.85 3.35 3.28 3.21 3.18 3.27 3.82 4.38
00:09:57.546 : 4.21 4.02 3.71 3.42 3.39 3.42 3.74 4.06 4.04
00:09:57.546 : 3.82 3.77 3.64 3.56 3.57 3.65 3.74 3.73 3.62
00:09:57.546 : 3.60 3.47 3.33 3.27 3.27 3.35 3.42 3.43 3.76
00:09:57.546 : 3.38 3.34 3.30 3.28 3.35 3.50 3.66 3.81 3.89
00:09:57.546 : 3.14 3.10 3.06 3.05 3.11 3.23 3.39 3.49 3.60
00:09:57.546 : 2.91 3.11 3.09 3.10 3.14 3.23 3.34 3.42 3.35
00:09:57.546 : 3.16 3.12 2.97 2.97 2.99 3.03 3.01 3.03 3.10
00:09:57.546 : 3.15 2.85 2.85 3.11 3.14 3.21 3.31 2.78 2.72
00:09:57.562 : <<< gcode_G29
00:09:57.562 : X:53.00 Y:64.00 Z:49.80 E:0.00 Count X: 14564 Y:22559 Z:25655

thinkyhead, 2016-08-27:
I have created a branch to test possible solutions to current bed leveling issues. Please test with this branch and see if it helps. You can just do the same G29 and post the new log.

https://github.com/thinkyhead/Marlin/tree/rc_fix_leveling_maths

markovicb85, 2016-08-30:
Ok, thanks. First, I took the zip file, replaced Configuration.h and Configuration_adv.h from example_configuration/delta to Marlin folder. After that, changed settings and upload file. In repetier I did the first M502, M500, G28 and finaly G29.
The result is same. Near z axis is low should be +0.6 mm, in the middle is +0.3 mm, and front part of printer (near x and y axis) is high, should be -0.2 mm. 
Here is the log:   

23:39:24.247 : G29 Auto Bed Leveling
23:40:02.356 : 0.59 0.01 -0.19 -0.22 -0.25 -0.20 -0.13 0.26 0.66
23:40:02.356 : 0.38 0.24 -0.08 -0.20 -0.20 -0.14 0.09 0.31 0.27
23:40:02.356 : 0.09 -0.01 -0.11 -0.18 -0.15 -0.09 -0.02 -0.01 -0.22
23:40:02.372 : -0.13 -0.26 -0.40 -0.46 -0.43 -0.36 -0.29 -0.24 0.03
23:40:02.372 : -0.35 -0.39 -0.43 -0.44 -0.35 -0.20 0.01 0.21 0.27
23:40:02.372 : -0.60 -0.65 -0.70 -0.69 -0.60 -0.47 -0.22 0.01 0.13
23:40:02.372 : -0.85 -0.68 -0.68 -0.66 -0.61 -0.52 -0.38 -0.26 -0.29
23:40:02.372 : -0.66 -0.71 -0.78 -0.76 -0.75 -0.74 -0.72 -0.69 -0.66
23:40:02.372 : -0.74 -0.89 -0.88 -0.66 -0.65 -0.62 -0.62 -0.93 -1.03
23:40:02.372 : X:53.00 Y:64.00 Z:50.00 E:0.00 Count X: 14564 Y:22559 Z:25655

thinkyhead, 2016-08-30:
Mmm, points. What do you make of them?

Roxy-3D, 2016-08-30:
That bed has significant bumps in it.   It is going to be difficult to get 1st layer adhesion across the entire bed using a grid based ABL.

thinkyhead, 2016-08-31:
@Roxy-3D Any chance we may be able to detect and flag this kind of problematic grid? By the way, I've posted on another issue concerning refinements to `G29` to get a better average height, and possibly a method whereby we can always use the center of the bed as the fulcrum with grid-based ABL. We should do a Hangout soon to look over the solution and assess it. I'll try to do an implementation of the idea in the next day or two also.

@markovicb85 Enable the `Z_MIN_PROBE_REPEATABILITY_TEST` and try `M48` to see how reliable your probe is.

Roxy-3D, 2016-08-31:
@thinkyhead   
> Any chance we may be able to detect and flag this kind of problematic grid? 

Yes!  This isn't fully thought out.  But at the first level, this would work:   Once the bed plane is calculated, you would go through each probed point in the list.  It had better be within +/- 1/2 of a layer height.  If it isn't, that part of the bed is flawed.

It maybe that those points should be thrown out of the calculation and you start over with a new, slightly modified plane.   I don't know.  But for sure, bumps and dips in the bed can be detected prior to the person trying to put down a 1st layer.

boelle, 2016-09-10:
@thinkyhead @Roxy-3D  any news on this one?

thinkyhead, 2016-09-10:
You'll be the first to know when there is @boelle. You are intrinsic to everything we do.

boelle, 2016-09-10:
hehe, just surfing the bugs so none of them are left to collect dust for to long (14 days.....)

thinkyhead, 2016-09-10:
Need to leave this link on every related thread so we'll remember to apply it…

https://github.com/MarlinFirmware/Marlin/issues/4708#issuecomment-243389571

First, take note that the probe offset doesn't matter in itself to measuring the tilt of the bed. The points are "thrown out" once the tilt angle of the bed is determined.

However, the probe offset is useful to know, at any point, what the distance to the nozzle is. And therefore you can consider any point that you've probed, or any point that you can interpolate from the matrix, as being a valid reference point to determine the height of the bed.

Let's take a concrete example. With a probe offset of -10 you probe the bed and get the following 4 points:
```
1. X:  0 Y:  0 Z:9.95
2. X:  0 Y:200 Z:9.90
3. X:200 Y:200 Z:9.85
4. X:200 Y:  0 Z:9.90
```
Of course we also have a home position from G28 to an endstop that simply applies over the entire bed, making the assumption that the bed is flat in relation to the XY axes. In terms of the distance from the probe to the nozzle, it is idealized, and therefore you can take it as 10.00 across the entire bed.

Now, we could take the home Z at any point, but generally it applies to wherever you calibrated your endstop. For a min endstop, usually you use a screw to adjust until G28 brings the nozzle down nice and close to the bed. And you usually check a few points across the bed to ensure it's as close to level as you can get it. But let's assume, for the moment, that you haven't done that in a while, and so it's fine at the home position — probably 0, 0 — or maybe 200, 0 for an X-max-endstop, but elsewhere it's off by an indeterminate distance. So you'd like to have the probe process find out where the bed really is.

So we have these points above. They show that the bed is tilted diagonally. At the front-left the bed is lower, and at the back-right the bed is higher. So the bed leveling matrix will produce a tilted plane reflecting that.

Now, let's say you're using a docking sled probe, or maybe a self-stowing allen key. So at the end of the leveling procedure, but before applying the bed level matrix, you have the carriage move off to the corner and drop off or push the probe to put it away. And you leave XYZ there.

Now, you want to turn on the matrix. But as soon as you do, the Z position will no longer be correct. If it's near the high corner, it will have too high a value. It's actually closer to the bed at that point. So what we do is take the Z position and apply the inverse matrix to it. In the process we also adjust XY a tiny bit, because that's how planar leveling works. However, we haven't yet figured out our reference point, so we can't actually apply the Z adjustment based on the tilted plane just yet.

Remember, we need a valid reference point to make the determination of how much the bed is raised or lowered at the current Z, and we haven't yet decided where the fulcrum of the plane lies.

So how do we proceed to get the "most correct" Z for the nozzle, where it is now? Well, let's take another look at those probed points.

Obviously, all across the bed the Z position of the nozzle is not 10.0. Everywhere we probed, the nozzle was a little lower than 10.0. We have to ask ourselves: Was the Z=0 that G28 gave us wrong? Or is the probe offset wrong? How can we determine which is true, and does it matter?

Then: What about the case where leveling is turned off? Shouldn't we make sure we have the best Z with `G28` alone first? It's less vital if you plan to always do G29, but if in the future it becomes possible to save the result of G29, then we're going to need a reliable G28 to give us a good Z to work from.

Going back to correcting Z…

If we assume that the probe offset is 100% correct, then clearly our Z=0 isn't good. Taking the 10.0 offset as gospel, the probe hit the bed too "late." Whatever Z=0 we have from homing, it's clearly above the bed. So we simply have to adjust the current Z upward by some amount. How much?

Since every probed point was under 10.0 we could try adjusting by the average difference. That corrective makes some of the probed Zs over 10.0 and some under 10.0. So let's start there.
```
10 - (9.95+9.90+9.85+9.90) ÷ 4 = 0.1
```
Adding that to our current Z —let's say it's "15.0"— makes it now 15.1. With that adjustment the nozzle distances to the bed now look like this:
```
9.95 + 0.1 = 10.05 (LF)
9.90 + 0.1 = 10.00 (LB)
9.85 + 0.1 =  9.95 (RB)
9.90 + 0.1 = 10.00 (RF)
```
This correction to Z applies perfectly if the nozzle were in the _middle_ of the bed. Obviously, right? The mean of all Zs probed on a flat but tilted plane must be in the dead center.

So now we have our fulcrum. Now we just need to get the distance from center to the nozzle's current XY, apply the bed tilt compensation for that XY distance, and presumably we will have the correct Z.

See how simple it all could be?

Mugga6315, 2016-09-13:
Hey guys, I got the same issue with the RC7 release like markovicb85. I then downloaded the rcbugfix release, but there is a new bug. When I'm doing the G29 (using 9 grid), the first 2 rows of measurement are working without a problem. But as soon as the 3. row starts, the x-axis drives from the left the right, hitting the frame. I can only power down or do an emergency stop at that point.

thinkyhead, 2016-09-13:
@Mugga6315 Sorry about that. I've been doing a lot of patches in the last couple days and probably mistyped something. Can you post your configuration so I can test with exactly your settings? Thanks!

Mugga6315, 2016-09-13:
@thinkyhead 
Never mind, you guys doing a great job with marlin. Like I've written in the other issue, the new bug was fixed with your changes. But the issue markovicb85 described stays for me, even with the newest bugfix release.

thinkyhead, 2016-09-13:
@Mugga6315 It might help, but first test the branch I linked to on that issue.

Mugga6315, 2016-09-13:
You mean this branch:
https://github.com/thinkyhead/Marlin/tree/rc_fix_leveling_maths

If so, the link is not working for me.

boelle, 2016-09-20:
@Mugga6315 did you test the branch linked above?

Mugga6315, 2016-09-20:
@boelle 
Jup, tested it. But behaviour stays the same, maybe a little bit better. But I also think that this is maybe relateted to the fact that my probe is approx. 90mm away from the nozzle (in Y-Axis). I'm not quite sure if the big distance is maybe causing the problem, because the x-axis with no offset to the nozzle is almost perfect.

thinkyhead, 2016-09-21:
@Mugga6315 Apologies. You should have been referred to the `RCBugFix` branch, not the older branch linked above. Have you also tested with `RCBugFix` and found that the inaccuracy remains?

Mugga6315, 2016-09-24:
@thinkyhead 
Tested now the latest RCBugFix branch. But issue still remains. But its a little bit better than the RC7 release, but almost the same to the "rc_final_z_correction" branch.

thinkyhead, 2016-09-24:
Can you post a log so we can see what's up? Enable `DEBUG_LEVELING_FEATURE` then use `M111 S255` to enable full logging. Do a `G28` followed by `G29 V4` and then post the log output here. A .txt file is preferable. From that we will be able to glean the nature of the issue.

Mugga6315, 2016-09-25:
@thinkyhead 
Sure, no problem:
[2016-09-25_logfile.txt](https://github.com/MarlinFirmware/Marlin/files/491578/2016-09-25_logfile.txt)

thinkyhead, 2016-09-26:
Considering that you're only able to probe half of the bed in the Y direction, I'm sure that's not helping. And it's tilting the matrix based on the home position, at the center [100,100], so any inaccuracy in the probing is going to be exaggerated towards the front of the bed.

Meanwhile, let's look at the probe heights:
```
0.723 0.918 0.818 0.698 0.417 0.580 0.675 0.705 0.738
0.725 0.935 0.835 0.657 0.377 0.452 0.660 0.713 0.732
0.675 0.907 0.795 0.515 0.352 0.352 0.568 0.670 0.693
0.620 0.863 0.705 0.405 0.332 0.320 0.413 0.623 0.663
0.585 0.835 0.573 0.373 0.320 0.302 0.305 0.538 0.635
0.565 0.777 0.467 0.367 0.322 0.300 0.282 0.405 0.598
0.513 0.665 0.445 0.365 0.332 0.302 0.275 0.310 0.523
0.377 0.553 0.442 0.370 0.322 0.300 0.282 0.290 0.400
0.350 0.568 0.467 0.395 0.350 0.317 0.297 0.307 0.365
```
The height of the probe varies widely, from 0.275 all the way up to 0.935. And those very high points in the back-left lie right next to points that are quite a bit lower. I can't imagine the bed is really that uneven. You should use `M48` to test the accuracy of your probe and make sure it's got good repeatability.

Maybe the probe itself is accurate, but the gantry is moving up and down. Or the bed is being pushed around, or something else mechanical. Tighten everything that you can.

The problem is, if you measure a lumpy bed like that and then apply a single plane (as the current leveling system does) then the tilted plane is basically useless.

I suggest you also try the `debug_leveling` branch using Roxy's "unified bed leveling" and try the mesh-based leveling. I'm working on making mesh-based leveling available for all users in `RCBugFix` but at the moment it's limited only to users of DELTA machines. Actually, I will try to make that change tonight. If your bed or other mechanicals are really moving the nozzle up and down that much over the bed, then mesh (actually "nonlinear grid") bed leveling will help.

Roxy-3D, 2016-09-26:
> I suggest you also try the debug_leveling branch using Roxy's "unified bed leveling" and try the mesh-based leveling. 

It is here:  https://github.com/MarlinFirmware/Marlin/tree/devel-ubl  

Look at the read me to get a feel for how it works.   You can glean the 'Quick Start' instructions here:   https://github.com/MarlinFirmware/Marlin/issues/4884

The UBL System can accommodate a bed with peaks and valleys like you have.   It won't even blink at it.

thinkyhead, 2016-09-26:
Once #4899 is merged, you can also try `AUTO_BED_LEVELING_BILINEAR` – the type of leveling that up to no Marlin has only made available for `DELTA`. As with Roxy's UBL it's made to handle non-flat beds, and should do a good job of compensating in those cases. If you do test it, be sure to enable the `DEBUG_LEVELING_FEATURE` and enable logging as before. The final Z adjustment works differently from planar leveling. Simpler.

Mugga6315, 2016-09-26:
Ok, that is quite strange. Because the bed should be pretty flat (glass bed). There is painters-tape on the glass, but dont think that this would cause that high offsets. At the moment I leveled the bed manually with sheet of paper and only using G28, which works for me pretty good. So maybe there is really an issue with the probe.

I think I will wait with further tests, until I get my new ir based probe, instead of the capacitive probe I'm using at the moment. I should then also be able to place the sensor much nearer to the nozzle.

But I will definitely test the probe accuracy with the M84 command.

Mugga6315, 2016-09-26:
I tried the M48 (I think M84 was just a mistake), but nothing happends. I also tried with arguments like in the wiki "M48 P4 X100 Y100 V2 E L2", but no action at all.

Roxy-3D, 2016-09-26:
> Ok, that is quite strange. Because the bed should be pretty flat (glass bed). 

Admitably, I'm deliberately using a bad piece of glass just to cause problems for the UBL System to deal with.  But this piece of glass does look flat to the eye:
```cpp
(0,9)                                                                                               (10,9)
(0,180)                                                                                             (200,180)
-0.21000  -0.20000  -0.19500  -0.22000  -0.14000  -0.22500  -0.31500  -0.30000  -0.41000  -0.43500  -0.46500

-0.20000  -0.20500  -0.20000  -0.21000  -0.33000  -0.24000  -0.35000  -0.29000  -0.48500  -0.52500  -0.45000

-0.23500  -0.23000  -0.17500  -0.29000  -0.29000  -0.31500  -0.17000  -0.39000  -0.40500  -0.59000  -0.41000

-0.22500  -0.22000  -0.17000  -0.16500  -0.21000  -0.27500  -0.23000  -0.35000  -0.34500  -0.39500  -0.45000

-0.26500  -0.08500  -0.15000 [-0.26000] -0.19500  -0.21500  -0.31000  -0.24000  -0.31500  -0.41500  -0.44500

-0.37500  -0.43000  -0.12500  -0.30500  -0.24500  -0.32000  -0.38500  -0.32000  -0.34500  -0.51500  -0.47000

-0.40000  -0.47500  -0.35000  -0.23000  -0.28000  -0.30000  -0.37500  -0.48500  -0.45000  -0.60000  -0.51000

-0.53500  -0.50500  -0.41000  -0.47500  -0.31000  -0.28500  -0.36000  -0.32000  -0.39000  -0.33000  -0.47500

-0.73000  -0.79000  -0.57500  -0.44500  -0.44500  -0.43500  -0.45000  -0.49000  -0.33500  -0.46000  -0.51000

-0.68500  -0.75500  -0.74000  -0.63000  -0.53000  -0.51000  -0.61000  -0.37500  -0.33500  -0.45000  -0.50500
(0,0)                                                                                               (200,0)
(0,0)                                                                                               (10,0)
```

> I tried the M48 (I think M84 was just a mistake), but nothing happends. I also tried with arguments like in the wiki "M48 P4 X100 Y100 V2 E L2", but no action at all.

You do need to turn on the M48 Z-Probe Repeatability test.  It is off by default:
```cpp
// Enable Z Probe Repeatability test to see how accurate your probe is
#define Z_MIN_PROBE_REPEATABILITY_TEST
```

boelle, 2016-11-02:
@markovicb85 is your issue still present in the latest RCBugfix?

closed 2016-11-02 by Roxy-3D