MarlinFirmware/Marlin #4672: u8glib

takachinss1001jp, 2016-08-21:
Dear Sir;

   I found u8glib in 1.0x source.
  Marlin Licence is GPL v3, u8glib is new BSD Licence.
  I really confused about diffrent Licences in same  Repository.
 Plase let me know what is reason,that you released as above?

 K.Haitaka<takachinss1001jp@yahoo.co.jp>

howard0su, 2016-08-21:
this should be fine. Please read https://en.wikipedia.org/wiki/License_compatibility

thinkyhead, 2016-08-21:
@takachinss1001jp I answered this already here, when you posted this question the first time: https://github.com/MarlinFirmware/Marlin/issues/4671#issuecomment-241261255 As Howard points out, Google contains all the answers you seek.

closed 2016-08-21 by thinkyhead

takachinss1001jp, 2016-08-22:
Thank your prompt reply to my small question,  howard0su and thinkyhead.
 I had been learned wiki and Official GPL. There might be compatibility between new-BSD Licence
and GOL v3.

thinkyhead, 2016-08-22:
> Yeah, There might be compatibility between new-BSD License and GPL v3.

There is no incompatibility between the BSD license applied to u8glib. And Marlin is not explicit about its GPL license. Let us assume it is GPL v2, not GPL v3.

> not only fonts but also u8glib

Don't worry about u8glib. The BSD license is permissive. GPL is only incompatible with licenses that apply new restrictions.

Please give us more details on the font(s) that you are concerned about. Here is the source of our Katakana: http://www.thingiverse.com/thing:664397