MarlinFirmware/Marlin #4674: Z axis issue when X and/or Y endstop are triggered.

tnw513, 2016-08-21:
Printer: BQ Prusa i3 Hephestos
Marlin version: RC6, RCBugFix (08/21/2016 morning) and "rc_final_z_correction" branch

After send G28 **and** G29 if I send G1 Z50 (at this point Z is lower than this, it's around 24) Z axis doesn't move but the position is update at 50. 
This is dangerous because even with min software endstop enabled the nozzle will smash on the print bed if after G1 X50 there is G1 Z0 (but also G1 Z20) in a gcode file.

Step to reproduce:
```
G28
G29
G1 X0 Y0
G1 X50 (greater than the current value)
```

Serial response:
```
>>> G1 X0 Y0
SENDING:G1 X0 Y0
echo:endstops hit:  Y:0.00
>>> G1 Z50
SENDING:G1 Z50
echo:endstops hit:  Y:0.00
```

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/428912/Configuration.txt)

thinkyhead, 2016-08-21:
Try disabling `ENDSTOPS_ALWAYS_ON_DEFAULT`.

thinkyhead, 2016-08-21:
Thing is, when the stepper-planner aborts on endstop hit it doesn't re-orient the true position. To manage that it would need to set a flag so that the main `loop()` can handle that. And I believe it only kills the current block. This is an area of the code that hasn't been examined or refined in a very long time.

thinkyhead, 2016-08-21:
Here's a change you can try out. In `endstops.cpp`…
```diff
- volatile char Endstops::endstop_hit_bits; // use X_MIN, Y_MIN, Z_MIN and Z_MIN_PROBE as BIT value
+ volatile uint16_t Endstops::endstop_hit_bits; // use X_MIN, Y_MIN, Z_MIN and Z_MIN_PROBE as BIT value
```

tnw513, 2016-08-22:
@thinkyhead  I'll try disabling ENDSTOPS_ALWAYS_ON_DEFAULT. 

However it will be better that if the stepper abort moves also the position update will abort.
But I don't know how this code works. Probably my observation is totally wrong :)

> When you leave endstops on, the stepper will abort moves on endstop hits.

Even if X or Y endstop is triggered and I try to move Z axis? One axis endstop influence the movement of other axes?

Please note that if I operate from LCD all kind of moves are permitted.

thinkyhead, 2016-08-22:
> if the stepper abort moves also the position update will abort

That's not how it works, so this is a non-trivial operation. There are three copies of the current position to track in Marlin:

- The `current_position[]` contains the "high level" (or "logical," or "ideal") XYZE position expressed in millimeters. This position is set by movement commands such as G1 when a move is first added to the planner queue. This position **only** represents the actual position of the XYZE axes when the carriage is standing still.

- The `planner.position[]` array contains the last XYZE (or ABCE for `DELTA` and `SCARA`) destination position sent to the planner, and it is expressed not in millimeters but in stepper steps. Bed leveling is applied before these positions are set, and for CoreXY, CoreXZ, and CoreYZ the planner also applies core kinematics before it sets `planner.position`. As with `current_position`, `planner.position` only corresponds to the axes' current positions when the carriage is standing still.

- The `stepper.count[]` array represents the current **actual** position of the XYZE (or ABCE for `DELTA` or `SCARA`) steppers, and it is the **only** array that keeps pace with the steppers. In fact this array is used directly in the movement of steppers in the movement ISR.

So if you interrupt the steppers in their motion it entirely invalidates the `current_position` and the `planner.position`, and the only way to repair those positions is to work backward from the current stepper positions. Due to the application of bed leveling compensation, `DELTA`/`SCARA` kinematics, and Core translation, working backward requires extra work. It necessitates that every one of these operations be reversible. And in fact, currently there are flaws that still need to be addressed in that area.

However, the hooks are in place. In `Marlin_main.cpp` we have `set_current_from_steppers_for_axis()` which uses one of three strategies depending on whether it's a `DELTA`/`SCARA`, or if bed leveling is enabled. (It is still flawed and needing work.) The `Stepper::get_axis_position_mm()` works perfectly and returns the true ABCE (i.e., XYZE on Cartesian) positions, but these will be the adjusted values that follow after applying bed leveling.

The work that still needs to be done:
- For `DELTA`/`SCARA`, reverse the bed leveling after obtaining the adjusted XYZ
- For `MESH_BED_LEVELING` un-correct the Z after obtaining XYZ
- Take a fresh look at `Planner::adjusted_position` and fix up

> Even if X or Y endstop is triggered and I try to move Z axis? 

To quote myself from above: "the intention is only to stop a move when it is actually heading in the direction of a particular endstop."

Roxy-3D, 2016-08-22:
> The work that still needs to be done:
> - For MESH_BED_LEVELING un-correct the Z after obtaining XYZ

This is easy to do if you know the movement was a primarily (X,Y) move.  But if a layer change gets interrupted, you can't just look at the Mesh (and Fade) (and Z-Offset) correction amounts.   In the case where a layer change is interrupted, I'm thinking we need more information.  Maybe we need to re-home?

Blue-Marlin, 2016-08-22:
If a matrix based ABL (3-point, grid) is active there is no pure z-move. Always also x and y is involved.

The root problem is that the legal min/max-positions are defined with triggered endstops. If we'd back up from the hardware endstops before we set the software endstops ... .
At least for x, y and z-max that should not be a problem.

thinkyhead, 2016-08-23:
> This is easy to do if you know the movement was a primarily (X,Y) move. But if a layer change gets interrupted, you can't just look at the Mesh (and Fade) (and Z-Offset) correction amounts. In the case where a layer change is interrupted, I'm thinking we need more information.

I'm not sure this is so. The idea is to go from the current stepper positions — that is, the presumed physical location of the axes — and gain the position that you would put into G1 to obtain that physical position. In the case of MBL you merely obtain the current XYZ from the steppers. The XY needs no re-orientation, because MBL doesn't shift those axes. All you need to obtain, then, is how much MBL shifts Z at the XY position and subtract that from the physical Z position. This works just fine regardless of where Z is, and regardless of whether it's half-way through a move.

> Maybe we need to re-home?

Re-homing works in all cases, but the point of re-establishing the current position after a quick-stop is to get a closest approximation of the current position. I understand it isn't entirely vital, but it came from a request and at the time it seemed like a good idea to be able to glean the logical position from the steppers.

closed 2016-12-23 by tnw513