MarlinFirmware/Marlin #4676: Feature request: add ability to use radius format of G02/G03 commands (CNC)

Bob-the-Kuhn, 2016-08-21:
My CAM software sometimes generates the radius version of the G02/G03 command.   It would be great if Marlin could handle the radius format.

I've written an Excel macro to do the conversion.  I've attached it incase it's usefull to the coders.

The macro only handles positive values of the radius.  Negative values of the radius requires messing with the X and Y parts of the command.
[radius portion of Excel macro.txt](https://github.com/MarlinFirmware/Marlin/files/428999/radius.portion.of.Excel.macro.txt)

Bob-the-Kuhn, 2016-08-22:
I've added support for negative radius.

Turns out you don't need to mess with the X and Y values when doing negative radius. 

[radius portion of Excel macro (with neg radius support).txt](https://github.com/MarlinFirmware/Marlin/files/429225/radius.portion.of.Excel.macro.with.neg.radius.support.txt)

Bob-the-Kuhn, 2016-08-22:
Didn't mean to close this request.
Just another example of shooting myself in the foot ...

Roxy-3D, 2016-08-22:
Bob, do you know the C programming language?   Can you put your concept into C for us?

I don't understand what a negative radius even means.   That seems kind of like an imaginary number.   Why is this useful (from a mathematical perspective)?

Bob-the-Kuhn, 2016-08-22:
Sorry – I knew a little C about 15 years ago.

 

Here’s a link that shows an example of a command with a negative radius.    It’s towards the bottom of the page & is titled “R Method Example 2:”

       http://www.manufacturinget.org/2011/12/cnc-g-code-g02-and-g03/

 

I only included the negative radius so I can call my Excel macro complete.

 

I’d be really surprised if any of the well known tools would generate a command with a negative radius.

 

   Bob

Roxy-3D, 2016-08-22:
Do CNC machines use negative radius on their GCode commands?

Bob-the-Kuhn, 2016-08-22:
I doubt it but … I’m just a beginner at CNC

Roxy-3D, 2016-08-22:
>> Do CNC machines use negative radius on their GCode commands?
> I doubt it but … I’m just a beginner at CNC

Bob, we have a constant challenge to keep the feature set of Marlin up to date and debugged for the items that user really need.  We can't be adding features that are not going to be used by a significant number of those users.

The GitHub system handles this situation very well.   What you do is you fork the code base, and add your features.   Your customized version of Merlin that does the things you need it to do can exist in parallel to the main branch of Merlin.

Bob-the-Kuhn, 2016-08-22:
I agree that the negative radius is not worth the effort by itself.  It’s just easy to include if positive radius is implemented.

 

The positive radius commands are generated by the SketchUcam plugin for the SketchUp program.  These are used a lot by the DIY CNC community.  

 

I expect that the user base for Marlin with the radius feature are those of us that want to use one machine to do both CNC and 3D printing.

 

   Bob

 

PS – The second file I attached does both positive and negative radius.  There’s only a few added statements from the first file to implement the negative radius.

Bob-the-Kuhn, 2016-08-23:
I've done my best to convert the conversion routine to C.  

One file has the C functions and the Excel syntax.  That one I could test.  It it the conversions correctly on a large gcode program.  The C functions were mimiced by Excel functions that I created for the test.

The other file has the C functions and the C syntax.   I have no way of testing it so it needs to be looked at carefully before being used.

    Bob
[radius portion of Excel macro (with neg radius support) using C functions and syntax.txt](https://github.com/MarlinFirmware/Marlin/files/431649/radius.portion.of.Excel.macro.with.neg.radius.support.using.C.functions.and.syntax.txt)
[radius portion of Excel macro (with neg radius support) using C functions.txt](https://github.com/MarlinFirmware/Marlin/files/431648/radius.portion.of.Excel.macro.with.neg.radius.support.using.C.functions.txt)

Bob-the-Kuhn, 2016-08-23:
I've found the GRBL code to do the conversion.  

They use vector calculations.  Seems like a lot more work than the method I found but I expect the real test is how fast they execute.

Attached are two files that are portions of the  **gcode.c** file in the main grbl branch.  This is where the radius conversion code is.

- grbl_gcode.c_radius conversion.docx - straight copy of the conversion section.  It has a very good explanation of how they did the conversion.  Best viewed in landscape mode with 12" width paper selected.

- grbl_gcode.c_radius conversion most comments deleted.docx  - it was hard for me to see how much computing was being done so I stripped out 99% of the comments.

  Bob

[grbl_gcode.c_radius conversion most comments deleted.docx](https://github.com/MarlinFirmware/Marlin/files/431717/grbl_gcode.c_radius.conversion.most.comments.deleted.docx)
[grbl_gcode.c_radius conversion.docx](https://github.com/MarlinFirmware/Marlin/files/431718/grbl_gcode.c_radius.conversion.docx)

thinkyhead, 2016-08-23:
Marlin already implements `G2` and `G3` codes that trace an arc. `G2` is clockwise arc, `G3` is counter-clockwise arc. The arc begins at the current position. The `X` `Y` `Z` parameters specify the end-point of the arc. The `I` and `J` parameters that specify the pivot-point of the arc.

Marlin also implements `G5` which traces a Beziér curve.

Bob-the-Kuhn, 2016-08-23:
In the general case the G2 & G3 commands can use either I J K to specify the arc or a radius.  Marlin implements only the I J K method.

Bob-the-Kuhn, 2016-09-18:
I've created a pull request for this feature request.

thinkyhead, 2016-09-18:
Where? It seems to have been targeted someplace besides `RCBugFix`. Please target all pull requests to `RCBugFix`.

Roxy-3D, 2016-09-18:
> Marlin already implements G2 and G3 codes that trace an arc. G2 is clockwise arc, G3 is counter-clockwise arc. 

@thinkyhead   Have you heard anybody talking about making G2 and G3 work with Delta's ?    I wonder if it would be as easy as sending the intermediate steps down the same pathway as G1 does to get the work done?

Bob-the-Kuhn, 2016-09-18:
There’s two ways of specifying the arc.  Marlin currently supports the I J method.  This work would allow Marlin to also support the R (radius) method.

 

   Bob

thinkyhead, 2016-09-18:
@Roxy-3D G2 and G3 work just wonderfully with Delta.

thinkyhead, 2016-09-18:
The "R" method seems like an interesting challenge. The code needs to determine the point that is equidistant from the start and end points, taking the clockwise or counter-clockwise path between them.

Note that if the start and end points are the same, there's no way to determine (with just an "R" argument) what the center of the circle should be. It could be in any direction. For this reason, full circles are forbidden with the "R" method, and you must therefore specify an XY which is a different point from where you started.

thinkyhead, 2016-09-18:
If the code isn't already written, here's how it will need to go…
http://mathforum.org/library/drmath/view/53027.html

Bob-the-Kuhn, 2016-09-19:
The code is written & tested.

 

I think I’ve finally figured out how to set things up so I can do the pull.  But I’ve said that twice before ..

 

   Bob

 

http://mathforum.org/library/drmath/view/53027.html

thinkyhead, 2016-09-19:
Oops, I went and wrote it. Let's compare. #4840

thinkyhead, 2016-09-19:
Just looking over the formula, I think maybe `sx` needs to be negated for the correct 90° rotation.

Bob-the-Kuhn, 2016-09-19:
The way I’ve been shooting myself in the foot today … it may take a couple of hours before I can get my code officially into an RCBugFix pull request.

 

It’s already in a pull request in my section.  Here’s the link to it.

   https://github.com/Bob-the-Kuhn/Marlin-RCBugFix--Radius-/pull/1

 

 

  Bob

thinkyhead, 2016-09-19:
Thanks! But I think it's pretty well covered by #4840. I've got it checking the arguments for errors, calculating the arc offsets, and it simply uses the existing `plan_arc` function with no extra code needed there. Very straightforward.

Bob-the-Kuhn, 2016-09-19:
I think I finally got the pull request against RCBugFix created.  Please see if https://github.com/MarlinFirmware/Marlin/pull/4841 is what you want to see.

 

The good news is that the R method is moving forward.  

 

I looked at the  plan_arc function but was afraid of touching existing code.

closed 2016-09-20 by thinkyhead